espressif/esp-idf #9443: protocomm component should not use #if CONFIG_WIFI_PROV_BLE_FORCE_ENCRYPTION guard

AxelLin, 2022-07-27:
## Environment
- IDF version: v5.0-dev-4379-g36f49f361c

## Problem Description

People may implement a customized provision function that uses protocomm directly.
Having CONFIG_WIFI_PROV_BLE_FORCE_ENCRYPTION in protocom assumes wifi_provisioning
is the only user to use protocomm which is wrong.
The lower layer (protocomm) component should not use any #if/#ifdef guard from upper layer (wifi_provisioning) component.

The bug:
When I use esp_prov.py to test my own provision code it fails if setting "--sec_ver 0".
i.e. it fails if CONFIG_WIFI_PROV_BLE_FORCE_ENCRYPTION=y && "--sec_ver 0".
So I notice it's very strange that I need to change wifi_provisioning component setting even though I didn't use that component.

The suggestion:
The protocomm component should provide a config (maybe an argument) to allow setting enable encryption or not.

rahult-github, 2022-07-27:
Hi @AxelLin,

> Having CONFIG_WIFI_PROV_BLE_FORCE_ENCRYPTION in protocom assumes wifi_provisioning
is the only user to use protocomm which is wrong.

Isn't the option "CONFIG_WIFI_PROV_BLE_FORCE_ENCRYPTION "  configurable from menuconfig ? IDF is a generic code targeted for wide audience and user has the option to enable / disable features as per their use case . 

Thanks,

AxelLin, 2022-07-27:
@rahult-github 

It's in menuconfig:
Wi-Fi Provisioning Manager  --->
  [ ] Force Link Encryption during characteristic Read / Write

Why should I change this setting even though I'm sure I don't use "Wi-Fi Provisioning Manager".
So I disagree with your comment. You can make it better.

closed 2023-01-20 by espressif-bot