espressif/esp-idf #9444: spi_bus_free doesn't disable pins properly ie. leaks power on SPI pins

Yardie-, 2022-07-27:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-S3-WROOM-16|]
- IDF version :v4.4.1
- Build System:         [idf.py]
- Compiler version :xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     [Linux|]
- Using an IDE?: [No]
- Power Supply:     [USB]

## Problem Description
spi_bus_free leaves all SPI pins pulled up. 
This is enough to visibly illuminate an LED 
### Expected Behavior
Pins should be dissabled and floating
### Actual Behavior
As above
### Steps to reproduce
After setting up spi as per example in peripherals 
with 
spi_device_handle_t spi = NULL;

void spi_end()
{
  if(spi != NULL)
    spi_bus_remove_device(spi);
  spi_bus_free(EPD_HOST);
  spi = NULL;
}
I have an LED connected on each pin as a test device this is where I noticed the issue.

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp

void epd_spi_end()
{
  if(spi != NULL)
    spi_bus_remove_device(spi);
  spi_bus_free(EPD_HOST);
  spi = NULL;
// without the following 4 lines LEDs remain slightly iluminated. Thus a power leak.
  gpio_set(CS_PIN, 0);
  gpio_set(MOSI_PIN, 0);
 gpio_set(MISO_PIN, 0);
  gpio_set(SCK_PIN, 0);  
}

This is simple to reproduce and totally repeatable. It doesn't create errors.

kriegste, 2022-07-28:
Isn't pullup the default GPIO state?
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/gpio.html#_CPPv414gpio_reset_pin10gpio_num_t

Yardie-, 2022-07-29:
@kriegste  Thank you. That is true I missed that part in the documentation. That explains this completely.

closed 2022-07-29 by Yardie-