espressif/esp-idf #9445: Modbus Master RTU Resource release failure issue

Rbhagwat62, 2022-07-27:
## Environment

- Development Kit:      [Custom Board]
- Kit version (for WroverKit/PicoKit/DevKitC): [NA]
- Module or chip used:  [ESP32-WROVER-32 (ESP32-D0WDQ6 (revision 1))]
- IDF version (run ``git describe --tags`` to find it): 4.3.2
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [VS code]
- Power Supply:         [external 3.3V]

## Problem Description

I am working on the Modbus RTU  Master code. It uses polling logic to send the Multiple register read request to different address. Earlier code with Modbus RTU master and BLE works fine.

When added Wi-Fi and cloud functionality then Modbus works fine for few minutes till the Resource release failure and invalid response error occurs. After that Modbus response gets shifted with last request's response and issue continues till ESP reboots. The polling rate is 400ms. Currently I have enabled the external RAM on Wrover module. If I kept Modbus task priority to 23, in that case time to get first error increases between 30 min to 1 hour.

I referred to #5072 and #6134 issues and seem it is known issue. Then I added the 0002-bugfix-workaround-to-clear-Modbus-stack-misalignment.patch.log from #5738. Patch was not matching with the my IDF version but still I added the it manually. Now in updated code, after ESP_ERR_INVALID_RESPONSE error occurred for first time Modbus gets stuck in sending request and need to reset the ESP32

Please let me know how to solve the issue.

## Sdkconfig 

#Modbus configuration
#CONFIG_FMB_COMM_MODE_TCP_EN is not set
CONFIG_FMB_COMM_MODE_RTU_EN=y
#CONFIG_FMB_COMM_MODE_ASCII_EN is not set
CONFIG_FMB_MASTER_TIMEOUT_MS_RESPOND=300
CONFIG_FMB_MASTER_DELAY_MS_CONVERT=200
CONFIG_FMB_QUEUE_LENGTH=20
CONFIG_FMB_PORT_TASK_STACK_SIZE=4096
CONFIG_FMB_SERIAL_BUF_SIZE=256
CONFIG_FMB_PORT_TASK_PRIO=23
CONFIG_FMB_PORT_TASK_AFFINITY_NO_AFFINITY=y
#CONFIG_FMB_PORT_TASK_AFFINITY_CPU0 is not set
#CONFIG_FMB_PORT_TASK_AFFINITY_CPU1 is not set
CONFIG_FMB_PORT_TASK_AFFINITY=0x7FFFFFFF
CONFIG_FMB_CONTROLLER_SLAVE_ID_SUPPORT=y
CONFIG_FMB_CONTROLLER_SLAVE_ID=0x00112233
CONFIG_FMB_CONTROLLER_NOTIFY_TIMEOUT=20
CONFIG_FMB_CONTROLLER_NOTIFY_QUEUE_SIZE=20
CONFIG_FMB_CONTROLLER_STACK_SIZE=4096
CONFIG_FMB_EVENT_QUEUE_TIMEOUT=20
#CONFIG_FMB_TIMER_PORT_ENABLED is not set
CONFIG_FMB_TIMER_GROUP=1
CONFIG_FMB_TIMER_INDEX=0
CONFIG_FMB_MASTER_TIMER_GROUP=1
CONFIG_FMB_MASTER_TIMER_INDEX=0
CONFIG_FMB_TIMER_ISR_IN_IRAM=y
#end of Modbus configuration

#UART configuration
CONFIG_UART_ISR_IN_IRAM=y
#end of UART configuration

### Code for Modbus request and retry logic

```cpp
while(1)
{
 op_Err = mbc_master_send_request(&test_request,&mbData));
  if(op_Err == ESP_OK)
  {
  // Do data processing and update the test_request with next address
  }
  else
  { 
  //retry existing test_request two more time
  //if fail for two consecutive time then skip current address and update test_request with next address
  }
}

```

alisitsyn, 2022-09-29:
Hello @Rbhagwat62 ,

Sorry for the late response.

Could you consider to use the `esp-modbus` component instead of `freemodbus` component in v4.3? 
The information for migration to new component is [here](https://github.com/espressif/esp-idf/blob/master/docs/en/migration-guides/release-5.x/protocols.rst#esp-modbus)

Also please check the [issue#9354](https://github.com/espressif/esp-idf/issues/9354#issuecomment-1187012014). Also try to use [an updated code](https://github.com/espressif/esp-idf/issues/9354#issuecomment-1196450014) with the updated component.

This should help to solve your issue. Let me know if you need more information.

Thank you.

Rbhagwat62, 2022-10-03:
Hello @alisitsyn ,

Thanks for the Reply. 

Currently I am working on the non WiFi Modbus implementation. I have updated esp-modbus component. So far I am not facing any issue. 

I will add Wi-Fi and HTTP related code soon. I will update you once I finish with Wi-Fi part.

Thank you.

Rbhagwat62, 2022-10-14:
Hello @alisitsyn,

I have added the WiFi and cloud code and now I am facing the same issue as I described in problem description earlier.

For WiFi and cloud, I am using Third party component, which has WiFi, http client and some other tasks with priorities less than the Modbus Task.

Currently master polls slave device at 300ms poll rate, all work fine till first Invalid Response error occurred. Code resends the request to slave again, second attempt onward it doesn't give any error. But only first response after error is correct. All responses for next request are replaced with last request response. 

I tried using .yml file in main and also with the [ update code ](https://github.com/espressif/esp-idf/issues/9354#issuecomment-1196450014) component but got same result.

Please find the log file, I have logged the Modbus request response packet. Check logs timings
[modbus_issue.txt](https://github.com/espressif/esp-idf/files/9785413/modbus_issue.txt)

16:16:32.000 Invalid response error occurred.
16:16:32.417 Request resent
16:16:32.419 Correct response received
16:16:32.569 Next request sent
16:16:32.571 Resource release failure error occurred but  mbc_master_send_request returns ESP_OK
16:16:32.575 incorrect response. (It is response from previous request). From here onward all response are gets shift with last request response and never recovered.

Also find SdkConfig for Modbus component
#define CONFIG_FMB_COMM_MODE_RTU_EN 1
#define CONFIG_FMB_MASTER_TIMEOUT_MS_RESPOND 499
#define CONFIG_FMB_MASTER_DELAY_MS_CONVERT 200
#define CONFIG_FMB_QUEUE_LENGTH 20
#define CONFIG_FMB_PORT_TASK_STACK_SIZE 4096
#define CONFIG_FMB_SERIAL_BUF_SIZE 256
#define CONFIG_FMB_PORT_TASK_PRIO 10
#define CONFIG_FMB_PORT_TASK_AFFINITY_CPU0 1
#define CONFIG_FMB_PORT_TASK_AFFINITY 0x0
#define CONFIG_FMB_CONTROLLER_NOTIFY_TIMEOUT 20
#define CONFIG_FMB_CONTROLLER_NOTIFY_QUEUE_SIZE 20
#define CONFIG_FMB_CONTROLLER_STACK_SIZE 4096
#define CONFIG_FMB_EVENT_QUEUE_TIMEOUT 20
#define CONFIG_FMB_TIMER_GROUP 0
#define CONFIG_FMB_TIMER_INDEX 0
#define CONFIG_FMB_MASTER_TIMER_GROUP 0
#define CONFIG_FMB_MASTER_TIMER_INDEX 0
#define CONFIG_FMB_TIMER_ISR_IN_IRAM 1

Let me know anything else is required.

Thank You.

alisitsyn, 2022-10-14:
Hello @Rbhagwat62 ,

Thank you for your results. Unfortunately your logs are not very useful for me. The log verbosity set to debug in the sdkconfig can help with this but will increase the master/processing and slave response time significantly (needs to increase the response time in master). The only solution I can recommend in this case as the recovery procedure after error:

1. Once the error occurred, your application needs to delay for at least slave response time, reset input buffer and reset receive queue manually. The [function](https://github.com/alisitsyn/modbus_support/blob/modbus_support/fix_glitches_esp_modbus_master_rtu_ascii/modbus_master_rtu/components/espressif__esp-modbus/freemodbus/port/portserial_m.c#L151) declared with `extern` in your code to clear the buffer then retry request. Unfortunately the Modbus stack can not recognize that the data in the response is from previous transaction because all your requests have the same Modbus related fields (due to its standard).  I can not effectively clear the data in the stack or do the delay. You can implement this in your application code that performs communication.
2. Please make sure that the manifest file (*.yml in main) is not used and alternative version of the component (code from support repo) is compiled instead.

Rbhagwat62, 2022-10-14:
Hello @alisitsyn 

Thanks for the response.

1. I already tried adding delay of slave response time after getting error, in that case whenever register number changes in read request then first request gets invalid response, on retry next response receives correctly and all responses for next requests are response of the last request, pattern repeats. I will add the function code and will get back to you soon.
2. I have removed the .yml file from the main and using the component from the [update code ](https://github.com/espressif/esp-idf/issues/9354#issuecomment-1196450014).

I will modify the code as per your suggestion and get back to with more detail logs.

alisitsyn, 2022-10-17:
Hello @Rbhagwat62 ,

This happens when the slave response to the request delayed and stack starts new transaction but the response comes after start of new transaction or before it. Since the response can come any time there is no any reliable method to get correct data because Modbus protocol do not have the fields to decide if the frame is response to current or it is response to previous transaction (if the command and length is the same). The only way is to set maximum slave response time for stack and wait the time after timeout error then reset the buffer before start of new transaction.

I will wait the logs and it would be helpful to get some reproducible results. Thanks.

Rbhagwat62, 2022-10-17:
Hello @alisitsyn,

I am sorry, I am using schedular logic from one of the components from another project. In that freemodbus was added in the REQUIRES. So whatever errors I was getting were from older freemodbus component. 

When I pointed it to updated code freemodbus component, I have not seen any invalid response or resource release failure error. I am running the code from last 5 to 6 hours and it is working fine. I got one issue, but I don't think its related to Modbus.

For the time being I am closing this post. In case of any issue, I will reopen it.

Thanks for your time and support.

closed 2022-10-17 by Rbhagwat62

alisitsyn, 2022-10-17:
@Rbhagwat62 ,

Just to make sure. Is this correct that now the component from [this repo](https://github.com/alisitsyn/modbus_support/tree/modbus_support/fix_glitches_esp_modbus_master_rtu_ascii/modbus_master_rtu/components/espressif__esp-modbus) is used in your application and this works correctly now?

Rbhagwat62, 2022-10-17:
Hello @alisitsyn,

Yes, I am using the component from [this repo](https://github.com/alisitsyn/modbus_support/tree/modbus_support/fix_glitches_esp_modbus_master_rtu_ascii/modbus_master_rtu/components/espressif__esp-modbus). So far it is working fine with WiFi and http data posting and receiving. Other data calculation related code is yet to add. I will update you in case I get any issue related to Modbus.

Thanks