espressif/esp-idf #9451: AttributeError: module 'esptool' has no attribute '_main'

5abina, 2022-07-28:
- Module or chip used:  ESP32-PICO-V3
- IDF master-branch
- Operating System: Windows
- Using an IDE: Yes - VS Code

## Problem Description
Getting this error:

esptool._main()  # type: ignore
AttributeError: module 'esptool' has no attribute '_main'

 while trying to flash binary 

esptool.py -p COM18 -b 460800 --before default_reset --after hard_reset --chip esp32 write_flash --flash_mode dio --flash_freq 40m --flash_size 4MB 0x10000 pre_encrypted_ota.bin 0x1000 bootloader/bootloader.bin 0x8000 partition_table/partition-table.bin 0xd000 ota_data_initial.bin 

from Visual Studio Code

### Expected Behavior

Flash ESP32 chip.

### Actual Behavior

esptool._main()  # type: ignore
AttributeError: module 'esptool' has no attribute '_main'

### Steps to reproduce

1. Open any example from examples folder.
2. Build a binary.
3. Try to flash SoC.

radimkarnis, 2022-07-28:
Hello @5abina,

thanks for your report. This issue should be solved (at least in most of dev environments). Some Q's to help us understand your problem:

1) Are you in an active ESP-IDF environment?
2) Are you using `idf.py flash` or running `esptool.py` directly?
3) Have you tried reinstalling your IDF env? (there are reports that this helps)
4) If you download [esptool](https://github.com/espressif/esptool) and try flashing, does this issue persist?
5) What's the output of `where esptool.py` ?

Let me know, thank you!

5abina, 2022-07-28:
Hello, thank for reply.
It works with eclipse and running commands directly from Windows PowerShell. I will try reinstall VS Code and let you know.

Thanks.

5abina, 2022-08-02:
Hello.

Today I have completely reinstall VS Code and everything is working fine. I will close it.

Thank you.

closed 2022-08-02 by 5abina

radimkarnis, 2022-08-02:
I am glad you could resolve the issue, @5abina! Thanks for reporting.