espressif/esp-idf #9453: Support for bigger ESP-NOW messages

AmirHmZz, 2022-07-28:
I think it would be better to implement something like multipart message in esp-now. 250 bytes might be small for some purposes.

MaxwellAlan, 2022-08-01:
Hi @AmirHmZz 

Thanks for your feature request.

Could you provide more detail for this feature that you may used in what kind of scenarios ? And how about the Throughput/Latency/Power/Security etc that required?

AmirHmZz, 2022-08-01:
@MaxwellAlan Thanks for your quick replay! 
To be honest, I don't know where that _250 bytes_ message length limitation comes from exactly, but :

- If it's just a **high level software layer** limitation: I think it would be better to make `ESP-NOW` to be able to transfer bigger messages at a specific cost ( lower throughput / higher latency / more power usage / etc. ).
- If it's **hardware layer** or **low level protocol** limitation which prevents the maximum length to be increased: Maximum message length can still remain _250 bytes_ but it would better to have something like **multipart messages** in `esp-idf` level. 

In my own company I had to implement these multipart messages in many projects, I think If this feature gets added to `esp-idf` developers can take advantage of it in many scenarios : 

1. Sending OTA updates over `ESP-NOW`
2. Transferring configuration-like messages ( which is more than 250 bytes in my projects ) 
3. Transferring bigger messages in lower rate

MaxwellAlan, 2022-08-04:
Hi @AmirHmZz 

Have checked the length limitation of espnow is only software limitation, we will support large message for espnow.

AmirHmZz, 2022-08-04:
@MaxwellAlan Thanks a lot for your quick reply.
Just for my curiosity, I've got two questions:

1. Does `ESP-NOW` force messages to have a fixed size? I mean if it's possible, can `ESP-NOW` protocol let messgaes to have desired lengthes?

2. Does message size effect on other parameters like latency, power consumption, etc? 

3. How large `ESP-NOW` messages can get? 

4. Can `ESP-NOW` gets data of big messages as streaming buffers? I mean for a message with length of 128KB ( like reading it from a file or another stream ) we cannot have all data in RAM. Also in the receiver side, message should cannot be received and should be streamed or saved somewhere.

AmirHmZz, 2022-08-04:
I mean `esp-idf` have developed lots of features like MESH networks on top of WiFi but when it comes to `esp-now` we have limited features...

MaxwellAlan, 2022-08-08:
@AmirHmZz 

Since there are some reserved field in 802.11 Managment Frame, maybe we can try with these fields, but there maybe some confilcts when the new 802.11 spec update.
Or could you try with esp_wifi_80211_tx this API to send and sniffer mode to receive, both could work only base on WiFi stack and without TCP/IP stack.

AmirHmZz, 2022-08-17:
@ceribu accroding to 802.11, how many Vendor Specific IE could be present in a vendor specific content?

AmirHmZz, 2022-08-17:
@ceribu So theoretically, the maximum length of `ESP-NOW` messages cannot get bigger than **2304** bytes, am I right ?

AmirHmZz, 2022-12-18:
Any updates on this one?

AmirHmZz, 2023-02-11:
Any updates on this?

zhangyanjiaoesp, 2023-04-03:
@AmirHmZz We will consider increasing the length of ESP-NOW by adding more Vendor-Specific IE. But currently we have higher priority work to do, so this feature is delayed. Have you tried to use the `esp_wifi_80211_tx` API ?

AmirHmZz, 2023-04-06:
What is exactly `esp_wifi_80211_tx` API?

zhangyanjiaoesp, 2023-04-10:
@AmirHmZz You can refer to the doc about `esp_wifi_80211_tx`: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html?highlight=80211_tx#_CPPv417esp_wifi_80211_tx16wifi_interface_tPKvib

derkalaender, 2023-06-20:
I'm by no means an expert, but this is also how I understood the 802.11 standard.
The action frame body requires the action field, which in turn for the *vendor-specific* action frame is defined as Category, Organization Identifier, and then Vendor Specific Content. "The Vendor Specific Content field contains vendor-specific *field(s)*". However, these vendor-specific fields are not defined anywhere, so I assume it can be any custom data.

I think the vendor-specific *element* is used by mistake. The action frame body, after the action field, also specifies that "one or more vendor-specific *elements* are optionally present", however, **only if** the category is not vendor-specific [...] which it is in this case!

And I think you are aware of this, because the "Random Values" field you use is not defined as part of the action frame. So I think that already constitutes "vendor specific fields", otherwise ESP-NOW would not be follow the standard.

zhangyanjiaoesp, 2023-06-25:
@ceribus @derkalaender 
You are right,  any custom type of content could be used for the Vendor Specific Content field of the Vendor Specific Action Frame. And the ESP-NOW was designed to using Vendor Specific IEs.

When changing code to support for bigger ESP-NOW messages, we should consider the compatibility with the original ESP-NOW product. Adding more Vendor Specific IEs is one way, and using a new type of content is another way, we will try to find a suitable way to ensure compatibility and the needs of customers.

nicklasb, 2023-11-21:
Just found this thread, I have several use cases where I occasionally want to send a, say 30-100 kbytes low-res image without the overhead and power consumption of a full wifi session. There are lots of other cases. 
If the max length could be for example 1.5 Kb, (which seems possible according to the above api), that would probably make for a much more efficient transfer of data than 250 bytes. Also..I mean we are in the to 2GHz band among the gamerz and downloaderz, do we really need to be this shy? :-)
I would still have to implement multi-part data transfers, but I feel it would be a little bit less problematic get it performant. Ideally, I would want ESP-NOW to handle that behind the scenes.

zhangyanjiaoesp, 2024-03-04:
@nicklasb We will consider your scenario when modifying the code.

nicklasb, 2024-03-06:
> @nicklasb We will consider your scenario when modifying the code.

That's encouraging.  :-)

AmirHmZz, 2024-05-22:
Any updates or milestones about this feature request?

zhangyanjiaoesp, 2024-05-23:
@AmirHmZz We will solve this problem in a high priority.

zhangyanjiaoesp, 2024-05-29:
@AmirHmZz which IDF branch do you use? which chip do you use ? I have modify the code, and I can provide a wifi lib for you to test.

AmirHmZz, 2024-05-29:
@zhangyanjiaoesp Hello again.
I use `esp-idf v5.1.2` on `ESP32` chip. 
Thank you.

zhangyanjiaoesp, 2024-05-29:
@AmirHmZz 
Here is the wifi lib: 
[wifi_lib_esp32_v5.1.2.zip](https://github.com/espressif/esp-idf/files/15481310/wifi_lib_esp32_v5.1.2.zip)
wifi firmware version: 6979f8f
Currently, the max length is 1600, and the new ESP-NOW version which can support longer length is 2.0, the old ESP-NOW version is 1.0, when you are using ESP-NOW to send/receive, the devices should use the same version.

AmirHmZz, 2024-05-29:
Thank you. Do I need to change `ESP_NOW_MAX_DATA_LEN`?
 https://github.com/espressif/esp-idf/blob/003f3bb5dc7c8af8b71926b7a0118cfc503cab11/components/esp_wifi/include/esp_now.h#L52

zhangyanjiaoesp, 2024-05-29:
> Thank you. Do I need to change `ESP_NOW_MAX_DATA_LEN`?
> 
> https://github.com/espressif/esp-idf/blob/003f3bb5dc7c8af8b71926b7a0118cfc503cab11/components/esp_wifi/include/esp_now.h#L52

Yes, change it to 1600.

AmirHmZz, 2024-07-24:
Sorry for the delay, finally I could test it.
I've tried it on two `esp32` modules with `idf v5.1.2` with the [example](https://github.com/espressif/esp-idf/tree/master/examples/wifi/espnow) provided by esp-idf and the result was:
- When setting data length to 512 bytes both of chips were receiving and sending messages correctly.
- After setting  data length to 1024 or higher only 1-2 messages were being received after each reset.

Could you please try the [example](https://github.com/espressif/esp-idf/tree/master/examples/wifi/espnow) and reproduce it?

AmirHmZz, 2024-07-29:
@zhangyanjiaoesp Any updates on this?

zhangyanjiaoesp, 2024-07-31:
@AmirHmZz sorry, I will check it this week

zhangyanjiaoesp, 2024-07-31:
@AmirHmZz 
I have tested the espnow example locally, I can set the buffer length up to 1485. Here are the logs, I set the `CONFIG_ESPNOW_SEND_COUNT = 10`, so there only 10 unicast data in the logs. 

Because the static RX buffer size is 1600, excluding some headers, the actual data is less than 1600.

[image]

device1
```
I (659) wifi:wifi firmware version: 6979f8f
I (659) wifi:wifi certification version: v7.0
I (659) wifi:config NVS flash: enabled
I (659) wifi:config nano formating: disabled
I (669) wifi:Init data frame dynamic rx buffer num: 32
I (669) wifi:Init static rx mgmt buffer num: 5
I (669) wifi:Init management short buffer num: 32
I (679) wifi:Init dynamic tx buffer num: 32
I (679) wifi:Init static rx buffer size: 1600
I (689) wifi:Init static rx buffer num: 10
I (689) wifi:Init dynamic rx buffer num: 32
I (699) wifi_init: rx ba win: 6
I (699) wifi_init: tcpip mbox: 32
I (699) wifi_init: udp mbox: 6
I (709) wifi_init: tcp mbox: 6
I (709) wifi_init: tcp tx win: 5744
I (709) wifi_init: tcp rx win: 5744
I (719) wifi_init: tcp mss: 1440
I (719) wifi_init: WiFi IRAM OP enabled
I (729) wifi_init: WiFi RX IRAM OP enabled
I (729) phy_init: phy_version 4780,16b31a7,Sep 22 2023,20:42:16
I (819) wifi:mode : sta (64:b7:08:29:2b:a4)
I (819) wifi:enable tsf
I (819) ESPNOW: espnow [version: 2.0] init
I (829) main_task: Returned from app_main()
I (5829) espnow_example: Start sending broadcast data
I (5829) espnow_example: Receive 19th broadcast data from: 64:b7:08:29:31:f4, len: 1485
I (5829) espnow_example: Receive 20th broadcast data from: 64:b7:08:29:31:f4, len: 1485
I (5839) espnow_example: Receive 21th broadcast data from: 64:b7:08:29:31:f4, len: 1485
I (5849) espnow_example: Receive 22th broadcast data from: 64:b7:08:29:31:f4, len: 1485
I (5849) espnow_example: Receive 23th broadcast data from: 64:b7:08:29:31:f4, len: 1485
I (6859) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (6859) espnow_example: Receive 24th broadcast data from: 64:b7:08:29:31:f4, len: 1485
I (7869) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (7869) espnow_example: Receive 25th broadcast data from: 64:b7:08:29:31:f4, len: 1485
I (7869) espnow_example: Receive 0th unicast data from: 64:b7:08:29:31:f4, len: 1485
I (8469) espnow_example: Receive 1th unicast data from: 64:b7:08:29:31:f4, len: 1485
I (9479) espnow_example: Receive 2th unicast data from: 64:b7:08:29:31:f4, len: 1485
I (10489) espnow_example: Receive 3th unicast data from: 64:b7:08:29:31:f4, len: 1485
I (11519) espnow_example: Receive 4th unicast data from: 64:b7:08:29:31:f4, len: 1485
I (12529) espnow_example: Receive 5th unicast data from: 64:b7:08:29:31:f4, len: 1485
I (13539) espnow_example: Receive 6th unicast data from: 64:b7:08:29:31:f4, len: 1485
I (14549) espnow_example: Receive 7th unicast data from: 64:b7:08:29:31:f4, len: 1485
I (15569) espnow_example: Receive 8th unicast data from: 64:b7:08:29:31:f4, len: 1485
I (16579) espnow_example: Receive 9th unicast data from: 64:b7:08:29:31:f4, len: 1485
``` 
device2
```
I (659) wifi:wifi firmware version: 6979f8f
I (659) wifi:wifi certification version: v7.0
I (659) wifi:config NVS flash: enabled
I (659) wifi:config nano formating: disabled
I (669) wifi:Init data frame dynamic rx buffer num: 32
I (669) wifi:Init static rx mgmt buffer num: 5
I (669) wifi:Init management short buffer num: 32
I (679) wifi:Init dynamic tx buffer num: 32
I (679) wifi:Init static rx buffer size: 1600
I (689) wifi:Init static rx buffer num: 10
I (689) wifi:Init dynamic rx buffer num: 32
I (699) wifi_init: rx ba win: 6
I (699) wifi_init: tcpip mbox: 32
[... 30 lines cut ...]
I (24129) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (25149) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (26159) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (27169) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (28179) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (29189) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (30219) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (30219) espnow_example: Receive 0th broadcast data from: 64:b7:08:29:2b:a4, len: 1485
I (31239) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (31239) espnow_example: Receive 1th broadcast data from: 64:b7:08:29:2b:a4, len: 1485
I (31239) espnow_example: Start sending unicast data
I (31239) espnow_example: send data to 64:b7:08:29:2b:a4
I (32309) espnow_example: send data to 64:b7:08:29:2b:a4
I (32309) espnow_example: Receive 2th broadcast data from: 64:b7:08:29:2b:a4, len: 1485
I (33319) espnow_example: send data to 64:b7:08:29:2b:a4
I (34329) espnow_example: send data to 64:b7:08:29:2b:a4
I (35339) espnow_example: send data to 64:b7:08:29:2b:a4
I (36369) espnow_example: send data to 64:b7:08:29:2b:a4
I (37379) espnow_example: send data to 64:b7:08:29:2b:a4
I (38389) espnow_example: send data to 64:b7:08:29:2b:a4
I (39399) espnow_example: send data to 64:b7:08:29:2b:a4
I (40419) espnow_example: send data to 64:b7:08:29:2b:a4
I (40429) espnow_example: Send done
I (40439) ESPNOW: espnow [version: 2.0] deinit
```

AmirHmZz, 2024-07-31:
@zhangyanjiaoesp I've tested with two other boards and everything seems to work fine! Thanks.
Two questions:
1. When this change gets merged into the master?
2. Will this change be backported for idf `5.1` or `5.2`?

zhangyanjiaoesp, 2024-08-01:
> @zhangyanjiaoesp I've tested with two other boards and everything seems to work fine! Thanks. Two questions:
> 
> 1. When this change gets merged into the master?
> 2. Will this change be backported for idf `5.1` or `5.2`?

@AmirHmZz In general, new features will only be available on the master and will not backport to older versions. We will merged it into the  master ASAP.

AmirHmZz, 2024-08-01:
@zhangyanjiaoesp So, it will be available on esp-idf `5.3` or `5.4`?

zhangyanjiaoesp, 2024-08-01:
It will be available on esp-idf 5.4.

closed 2024-10-07 by espressif-bot