espressif/esp-idf #9459: Open source `espressif/esp-xtensaconfig-lib` repo?

noahp, 2022-07-29:
Hello!

Would it be possible to make the `espressif/esp-xtensaconfig-lib` GitHub repo public? I would like to build a custom copy of https://github.com/espressif/binutils-gdb .

Thanks!

igrr, 2022-08-02:
@noahp Thanks for the heads-up and apologies for not making the repository public in the first place. It should be available now: https://github.com/espressif/esp-xtensaconfig-lib.

Lapshin, 2022-08-02:
@noahp, `espressif/esp-xtensaconfig-lib` is used only for GDB builds (for now).

For **Xtensa** builds:
If you need to build any tool from https://github.com/espressif/binutils-gdb you need to apply overlay from https://github.com/espressif/xtensa-overlays . Just override files from [binutils](https://github.com/espressif/xtensa-overlays/tree/master/xtensa_esp32/binutils) (for binutils build) or [gdb](https://github.com/espressif/xtensa-overlays/tree/master/xtensa_esp32/gdb) (for gdb build) to binutils-gdb sources. Links are given for esp32, use another directory in [repo's root](https://github.com/espressif/xtensa-overlays/tree/master/).

If you need to build GDB you can use a [build script](https://github.com/espressif/binutils-gdb/blob/esp-gdb-11.2/build_esp_gdb.sh).

Lapshin, 2022-08-02:
> @noahp, `espressif/esp-xtensaconfig-lib` is used only for GDB builds (for now).
>
> For **Xtensa** builds: If you need to build any tool from https://github.com/espressif/binutils-gdb you need to apply [...]

And build it as usual after this

noahp, 2022-08-02:
Wonderful! :raised_hands: Thank you so much!

closed 2022-08-02 by noahp

noahp, 2022-08-16:
FYI the reason we were looking at this was to generate a build for aarch64-macOS hosts- it built the same as the other hosts, so if you end up adding that to your releases, it shouldn't be too bad (except you need an aarch64-macOS build machine, which is a big pain unfortunately).