espressif/esp-idf #9460: cache disabled panic crash

SunnyRuss, 2022-07-29:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:  
- // None
- Kit version 
- //
- Module or chip used:  
- // ESP32-WROVER-IE Rev 3
- IDF version (run ``git describe --tags`` to find it): 
- // v4.4.1
- Build System:   
- // idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 
- // xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:    
- // Windows
- (Windows only) environment type: 
- // ESP Command Prompt|
- Using an IDE?: 
- // No
- Power Supply:
- // External

## Problem Description

Crashes after enabling Wi-Fi

//Detailed problem description goes here.

Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed). <CR><LF>
<CR><LF>
Core  1 register dump:<CR><LF>
PC      : 0x40184b08  PS      : 0x00060034  A0      : 0x80084ed1  A1      : 0x3ffb15f0  <CR><LF>
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffb1fd4  A5      : 0x00002000  <CR><LF>
A6      : 0x3ffb5d20  A7      : 0x1300024c  A8      : 0x80084e04  A9      : 0x00001000  <CR><LF>
A10     : 0x3ffb1fd4  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x00000800  <CR><LF>
A14     : 0x00000001  A15     : 0x00060d23  SAR     : 0x00000004  EXCCAUSE: 0x00000007  <CR><LF>
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  <CR><LF>
<CR><LF>
<CR><LF>
Backtrace: 0x40184b05:0x3ffb15f0 0x40084ece:0x3ffb1620 0x4008413d:0x3ffb1650 0x40086b1a:0x3ffb8ec0 0x40084b0e:0x3ffb8ee0 0x40084b0e:0x3ffb8f00 <CR><LF>
<CR><LF>
<CR><LF>
<CR><LF>
<CR><LF>
ELF file SHA256: fb2ffa2d15221416<CR><LF>
<CR><LF>
<ESC>[0;32mI (14130) esp_core_dump_flash: Save core dump to flash...<ESC>[0m<CR><LF>
<ESC>[0;32mI (14137) esp_core_dump_flash: Erase flash 24576 bytes @ 0xc10000<ESC>[0m<CR><LF>
<ESC>[0;32mI (14534) esp_core_dump_flash: Write end offset 0x5ee4, check sum length 4<ESC>[0m<CR><LF>
<ESC>[0;32mI (14534) esp_core_dump_flash: Core dump has been saved to flash.<ESC>[0m<CR><LF>

0x40184b05
C:/Sunsight/esp/esp-idf-v4.4/components/hal/i2c_hal.c:170
0x40084ece
C:/Sunsight/esp/esp-idf-v4.4/components/driver/i2c.c:497
0x4008413d
C:/Sunsight/esp/esp-idf-v4.4/components/freertos/port/xtensa/xtensa_vectors.S:1111
0x40086b1a
C:/Sunsight/esp/esp-idf-v4.4/components/spi_flash/cache_utils.c:122
0x40084b0e
C:/Sunsight/esp/esp-idf-v4.4/components/esp_ipc/src/esp_ipc.c:74
0x4008fc85
C:/Sunsight/esp/esp-idf-v4.4/components/freertos/port/xtensa/port.c:131

Looking in cache_utils.c in the area around line 122, it says you have problems accessing PSRAM on the other CPU.  I suspect that to be actual cause.  A little latter it talks about a long running ISR. 

How do I further diagnose this problem and then fix it?

-Russ

SunnyRuss, 2022-07-29:
I used IDF.PY MENUCONFIG to disable the second core and the crash does not happen.

ESP-Marius, 2022-08-01:
Hi @SunnyRuss 

Which compiler optimization level (menuconfig->Compiler Options->Optimization level) are you running with?  
I suspect there might be an issue with the i2c driver if running with `-O0`/`COMPILER_OPTIMIZATION_NONE`

chipweinberger, 2022-08-01:
see https://github.com/espressif/esp-idf/issues/9129

mythbuster5, 2022-08-01:
@SunnyRuss Could you please try to apply https://github.com/espressif/esp-idf/commit/00229547b20ed1752dc152a055350cfb480e4522 this commit to your project? I think this commit might fix your issue.

SunnyRuss, 2022-08-05:
I applied 0022954.  It caused no harm.  The crash is/was intermittent and has not occurred after applying 0022954.  I will close this ticket.

closed 2022-08-05 by SunnyRuss