espressif/esp-idf #9461: [Hang][ESP-WROOM32][v4.4.1][VSCode Linux w/Espressif plugin][USB] uart_hal_write_txfifo causes Interrupt wdt timeout

sbltlh, 2022-07-29:
Related to #9155? I have applied the fix_v4.4.zip from #9129.

Running a NimBLE SPP profile and Wi-Fi station. BLE is not connected/being used and Wi-Fi is not either, it is just connected. Crash is very random. Usually takes a day or two to happen.

Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
PC      : 0x4009195c  PS      : 0x00060b35  A0      : 0x8008ac52  A1      : 0x3ffbeb00  
0x4009195c: vPortExitCritical at /mnt/MORE/projects/esp_tools/esp-idf/components/freertos/port/xtensa/port.c:305

A2      : 0x00000001  A3      : 0x3ffbeb30  A4      : 0x3ffbeb38  A5      : 0x00000000  
A6      : 0x3ffd4c74  A7      : 0x3ffd4cc8  A8      : 0x80091de8  A9      : 0x3ffbeae0  
A10     : 0x3ffd4cc8  A11     : 0x3ffd4cd0  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x3ffd4c74  A15     : 0x3ffd4cc8  SAR     : 0x0000001c  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
Core  0 was running in ISR context:
EPC1    : 0x400d3693  EPC2    : 0x00000000  EPC3    : 0x4000bff0  EPC4    : 0x40091ba2
0x400d3693: uart_hal_write_txfifo at /mnt/MORE/projects/esp_tools/esp-idf/components/hal/uart_hal_iram.c:35

0x40091ba2: prvCopyDataToQueue at /mnt/MORE/projects/esp_tools/esp-idf/components/freertos/queue.c:2267

Backtrace:0x40091959:0x3ffbeb000x4008ac4f:0x3ffbeb30 0x40082ca1:0x3ffbeb60 0x4008ac59:0x3fff3c20 0x400835bb:0x3fff3c60 0x4010b3ea:0x3fff3c80 
0x40091959: xPortEnterCriticalTimeout at /mnt/MORE/projects/esp_tools/esp-idf/components/freertos/port/xtensa/port.c:302

0x4008ac4f: i2c_isr_handler_default at /mnt/MORE/projects/esp_tools/esp-idf/components/driver/i2c.c:506

0x40082ca1: _xt_lowint1 at /mnt/MORE/projects/esp_tools/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1111

0x4008ac59: i2c_isr_handler_default at /mnt/MORE/projects/esp_tools/esp-idf/components/driver/i2c.c:531

0x400835bb: queue_recv_hlevel_wrapper at /mnt/MORE/projects/esp_tools/esp-idf/components/bt/controller/esp32/bt.c:954

0x4010b3ea: btdm_controller_task at ??:?

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/9222639/sdkconfig.txt)

chipweinberger, 2022-08-01:
Interesting that you are hitting this with the fix_v4.4.zip patch.

Note: I closed https://github.com/espressif/esp-idf/issues/9155 sort of preemptively without testing a great deal.

blueMoodBHD, 2024-02-27:
Hi @sbltlh 

Is the issue still present? Could you please try the latest release version?

Alvin1Zhang, 2024-07-24:
Thanks for reporting, sorry for the very slow turnaround, will close due to short of feedback, feel free to reopen with more updates. Thanks.

closed 2024-07-24 by Alvin1Zhang