espressif/esp-idf #9462: [Feature Request] sdkconfig: "Print registers and reboot after delay" would be useful

chipweinberger, 2022-07-29:
1. During development, "print registers and reboot" is annoying because terminals get filled with more text, you can't read it. 
2. However, during beta testing, "print registers and halt" is not good for users.

We need a middle approach, with a configurable amount of seconds delay before reboot. For example 10 seconds.

Thanks for your consideration.

chipweinberger, 2022-08-11:
Bump. 

A `CONFIG_ESP_SYSTEM_PANIC_PRINT_DELAY_REBOOT` option would let me keep it the same between development and production, and it just generally more useful IMO. 20htz boot loops are annoying =)

chipweinberger, 2022-08-11:
@ESP-Marius, @igrr, @Alvin1Zhang 

Just want to make sure 1 person sees this.

igrr, 2022-08-11:
Hi @chipweinberger,
All issues posted on Github are synced to our internal issue tracker, categorized and land on of of the teams' backlog. We've seen this feature request but currently consider it relatively low priority. Most efforts right now are focused on v5.0 release and various bug fixes.

Have you checked this example demonstrating how to set up builds for multiple configurations (development/production)? https://github.com/espressif/esp-idf/blob/master/examples/build_system/cmake/multi_config/README.md
I hope it can satisfy your requirement for now.

chipweinberger, 2022-08-11:
Thanks for your input Ivan, as always. 

That is a useful page, for now =)

**Use case:** For Beta testing, reboot after delay is most ideal. A beta device may need to be debugged "live", but must still reboot automatically.

chipweinberger, 2022-10-12:
Pull Request: https://github.com/espressif/esp-idf/pull/9962

@igrr, hope you consider it!

chipweinberger, 2023-02-21:
merged into master.

closed 2023-02-21 by chipweinberger