espressif/esp-idf #9463: modbus function code 0x06 MB_FUNC_WRITE_REGISTER

mwh-github, 2022-07-30:
+ When I use the modbus master rtu, I only find that there is only a function code for writing multiple registers, but there is no function for writing a single register.
+ There are some modbus devices that only support writing a single register(function code 0x06), not multiple registers(function code 0x10).
```
static uint8_t mbc_serial_master_get_command(mb_param_type_t param_type, mb_param_mode_t mode)
{
    uint8_t command = 0;
    switch(param_type)
    { //
        case MB_PARAM_HOLDING:
            command = (mode == MB_PARAM_WRITE) ?
                        MB_FUNC_WRITE_MULTIPLE_REGISTERS :
                        MB_FUNC_READ_HOLDING_REGISTER;
            break;
        case MB_PARAM_INPUT:
            command = MB_FUNC_READ_INPUT_REGISTER;
            break;
        case MB_PARAM_COIL:
            command = (mode == MB_PARAM_WRITE) ?
                        MB_FUNC_WRITE_MULTIPLE_COILS :
                        MB_FUNC_READ_COILS;
            break;
        case MB_PARAM_DISCRETE:
            if (mode != MB_PARAM_WRITE) {
                command = MB_FUNC_READ_DISCRETE_INPUTS;
            } else {
                ESP_LOGE(MB_MASTER_TAG, "%s: Incorrect mode (%u)",
                            __FUNCTION__, (uint8_t)mode);
            }
            break;
        default:
            ESP_LOGE(MB_MASTER_TAG, "%s: Incorrect param type (%u)",
                            __FUNCTION__, param_type);
            break;
    }
    return command;
}
```

alisitsyn, 2022-08-01:
Hello @mwh-github,

Thank you for the issue.
During development of this port  it was supposed that the command `MB_FUNC_WRITE_MULTIPLE_REGISTERS`  =  0x10 is enough to write registers because it can write one register as well as multiple. However I agree that some devices do not support this command and this looks strange to me because 0x10 is more universal. This command will be added to the list of commands with low priority. By the way what is the device you are using?

mwh-github, 2022-09-25:
> Hello @mwh-github,
>
> Thank you for the issue. During development of this port it was supposed that the command `MB_FUNC_WRITE_MULTIPLE_REGISTERS` = 0x10 is enough to write registers because it can [...]

> Hello @mwh-github,
>
> Thank you for the issue. During development of this port it was supposed that the command `MB_FUNC_WRITE_MULTIPLE_REGISTERS` = 0x10 is enough to write registers because it can [...]

Thanks for your reply. I am using TECO's S310 and C310 inverter models.The S310  supports 0x06 and 0x10 commands, but the C310 only supports 0x06 when writing.

alisitsyn, 2023-02-20:
Hello @mwh-github ,

Sorry for even very late response. 
There are several variants to do this. The most simple way to implement the command 0x06 is to use the cpp wrapper code and change the command to 6 here: [url]https://github.com/alisitsyn/modbus_support/blob/mb_support/modbus_add_cpp_wrapper/modbus_simple_master/main/ModbusMaster.cpp#L170[/url]. This will require to do a cycle writing the registers if your code needs to write several registers. The same way it can be implemented in pure C.

closed 2023-06-13 by Alvin1Zhang