espressif/esp-idf #9464: ulp_riscv_adc.h missing #ifdef __cplusplus closing bracket

203Null, 2022-07-30:
## Environment

- Module or chip used:  ESP32-S3
- IDF version (run ``git describe --tags`` to find it):
    master 36f49f3 - Merge branch 'bugfix/esp_netif_clean_deps_lwip' into 'master'

## Problem Description

https://github.com/espressif/esp-idf/blob/master/components/ulp/ulp_riscv/include/ulp_riscv_adc.h
This file is missing the closing bracket for #ifdef __cplusplus 

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"
#include "ulp_riscv_adc.h"

void app_main()
{

}

```

ESP-Marius, 2022-08-01:
You're right, I've pushed an internal MR to fix this.

Thanks for reporting this.

203Null, 2022-08-05:
Seems to be resolved

closed 2022-08-05 by 203Null