espressif/esp-idf #9465: [example request]Is there any example on usb serial/jtag controller to act like a normal serial communicating with computer

kaidegit, 2022-07-30:
It is easy to output logs to this controller using menuconfig, however, it seems to have no example on how to use it by ourselves.

FanhuaCloud, 2022-08-01:
You can use the CDC serial port

kaidegit, 2022-08-01:
However, esp32c3 has no usb peripheral to set up as a cdc device, but has a usb serial/jtag controller. I also see the `Arduino-esp32` project offered the usage of this controller.  So why not offering an example to help us to use it?

igrr, 2022-08-01:
Hi @kaidegit, indeed there is no example of the usb_serial_jtag driver yet. Until it is added, please refer to the comments in this header file: https://github.com/espressif/esp-idf/blob/release/v4.4/components/driver/include/driver/usb_serial_jtag.h.

ESP-Marius, 2024-03-24:
I think with https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb_serial_jtag/usb_serial_jtag_echo we can consider this request fulfilled. 

@kaidegit feel free to re-open this again if you think it is still lacking.

closed 2024-03-24 by ESP-Marius