espressif/esp-idf #9466: AAC decoder not finding correct sample_rate for some streams

espradio, 2022-07-30:
Noticed AAC decoder not finding the sample rate for some stream such as http://streamasiacdn.atc-labs.com/rdsaanthwanam.aac , 
For example, This stream is actually 44100Hz sample rate. But decoder is detecting as 22050.

How to produce:
============
Able to reproduce with examples/player/pipeline_living_stream/.
Apply following patch.

```
diff --git a/examples/player/pipeline_living_stream/main/play_living_stream_example.c b/examples/player/pipeline_living_stream/main/play_living_stream_example.c
index 91a8c43..398c5ad 100644
--- a/examples/player/pipeline_living_stream/main/play_living_stream_example.c
+++ b/examples/player/pipeline_living_stream/main/play_living_stream_example.c
@@ -37,7 +37,7 @@
 
 static const char *TAG = "HTTP_LIVINGSTREAM_EXAMPLE";
 
-#define AAC_STREAM_URI "http://open.ls.qingting.fm/live/274/64k.m3u8?format=aac"
+#define AAC_STREAM_URI "http://streamasiacdn.atc-labs.com/rdsaanthwanam.aac"
```

### Log:
` (9744) HTTP_LIVINGSTREAM_EXAMPLE: [ * ] Receive music info from aac decoder, sample_rates=22050, bits=16, ch=2
` 
Attaching the VLC decoder output for
`vlc http://streamasiacdn.atc-labs.com/rdsaanthwanam.aac
`
[image]

suda-morris, 2022-08-01:
This example doesn't exist in esp-idf, suggest reporting the issue to [esp-adf](https://github.com/espressif/esp-adf/issues).

Alvin1Zhang, 2022-08-01:
Thanks for reporting, will close now, please help create issue on https://github.com/espressif/esp-adf/issues, thanks.

closed 2022-08-07 by igrr