espressif/esp-idf #9468: compile_commands.json is not complete

ldmik, 2022-08-01:
**Is your feature request related to a problem? Please describe.**
I'm using clangd in vscode to parse the code, seems the generated compile_commands.json in build folder is not complete, e.g., the ledc.c and tasks.c of freerots will not be recorded in compile_commands.json. This causes some error during code parsing.

A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
Update the compile scripts to generate a right compile_commands.json
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
none
A clear and concise description of any alternative solutions or features you've considered.

Please give as many details as you can. Include suggestions for useful APIs or interfaces if relevant.

**Additional context**
1. vscode with clangd extension
2. idf 4.4.1
3. windows 10
Add any other context or screenshots about the feature request here.

igrr, 2022-08-01:
@ldmik could you please attach the generated compile_commands.json file where you observe this issue? Does this issue occur also with the default IDF examples, such as hello_world?

ldmik, 2022-08-02:
[compile_commands.zip](https://github.com/espressif/esp-idf/files/9238628/compile_commands.zip)
[hello_world_main.zip](https://github.com/espressif/esp-idf/files/9238633/hello_world_main.zip)
Hi @igrr , here's my compile_commands.json file which generated by a modified hello world project based on example of the master branch.

I see there are ledc.c and tasks.c file in compile_commands.json file, but when I using clangd to goto the definition, always it can't go to the expected source code but to the declaration position.

igrr, 2022-08-07:
If the compile_commands.json is generated correctly, perhaps the issue is with clangd extension?
Please have a look also at https://github.com/espressif/esp-idf/issues/6721 where other users have had some limited success in using clangd.

igrr, 2023-02-24:
I'm going to close this issue since compile_commands.json was generated correctly, and there was no further response.

closed 2023-02-24 by igrr