espressif/esp-idf #9469: MEM Release (8 bits capabilities) after BLE

jrHernandezSio, 2022-08-01:
Hi,

ENVIRONMENT: espressif32@3.3.2 -> ESP-IDF 4.3

I want to free memory after using BLE in the same boot, but I am finding troubles because it seems the memoery is not freed properly.

My BLE driver is set (by menucfg) to be using dynamiv memory allocation.

Before releasing the bt driver memory the driver is denited as follows:

```
    esp_ble_gap_stop_advertising();

    esp_bluedroid_disable();
    esp_bluedroid_deinit();

    esp_bt_controller_disable();
    esp_bt_controller_deinit();

  // -> Release bt memory done here
```

I use this function to check memory: `heap_caps_print_heap_info(MALLOC_CAP_8BIT);`
**Mem at wakeup**:     
free **209360** allocated 15856 min_free **209360** largest_free_block 65536

I have tried several ways to release memory:

`esp_bt_controller_mem_release( ESP_BT_MODE_BLE );`
free **25664** allocated 199552 min_free **9312** largest_free_block 8192
`esp_bt_mem_release( ESP_BT_MODE_BLE ); `
free **73936** allocated 154828 min_free **9708** largest_free_block 16384
`esp_bt_mem_release( ESP_BT_MODE_BTDM );`
free **104736** allocated 154928 min_free **37688** largest_free_block 16384

But in any case, in spite of the **free** value is OK for my application the **min_free** value is not enough for me and get run out of memory.

Why the min_free value is not recovered if mem has been released?
Is there anything I am missing when unloading BLE driver?

Nor sure if this issue is related to BLE driver.

Regards, Ramon.

igrr, 2022-08-01:
Hi Ramon,
`min_free` is the value also known as "minimum ever free memory" — the lowest amount of memory free in the given memory region that has ever occurred during program execution. So it is not a value which can be "recovered" by freeing some memory.

If the allocation failure occurs after BLE is deinitialized, then the relevant numbers you need to check are:
* "free" — total free memory
* "largest_free_block" — largest amount of memory which can be allocated in a single call to "malloc"

If your application is trying to allocate a large chunk of memory (larger than "largest_free_block" but smaller than "free"), then the allocation will fail because there is no chunk large enough to satisfy this request, even if the total amount of heap memory is sufficient.

jrHernandezSio, 2022-08-01:
Hi, 

Thanks for your fast esponse.

Understood, but then I do not understand the following:

Have a look at the values before and after unloading the BLE driver (The mem release is don inside BLE disconnect).

-DEBUG-: [1659346379,484 s] Session dead
Heap summary for capabilities 0x00000004:
  At 0x3ff80518 len 6872 free 144 allocated 5104 min_free 144
    largest_free_block 0 alloc_blocks 35 free_blocks 0 total_blocks 35
  At 0x3ffb6388 len 7288 free 84 allocated 5580 min_free 68
    largest_free_block 0 alloc_blocks 20 free_blocks 0 total_blocks 20
  At 0x3ffb9a20 len 16648 free 308 allocated 14716 min_free 308
    largest_free_block 0 alloc_blocks 76 free_blocks 0 total_blocks 76
  At 0x3ffce4a0 len 72544 free 5188 allocated 65732 min_free 1552
    largest_free_block 2048 alloc_blocks 375 free_blocks 5 total_blocks 380
  At 0x3ffe0440 len 129984 free 17828 allocated 110532 min_free 7104
    largest_free_block 8192 alloc_blocks 14 free_blocks 4 total_blocks 18
  Totals:
    free 23552 allocated 201664 **min_free 9176** largest_free_block 8192
-DEBUG-: [1659346379,548 s] BLE connected (1)
**-DEBUG-: [1659346379,552 s] BLE disconnect...**
Heap summary for capabilities 0x00000004:
  At 0x3ffc294c len 2584 free 960 allocated 0 min_free 924
    largest_free_block 512 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x3ffb8000 len 6688 free 5028 allocated 36 min_free 5028
    largest_free_block 4096 alloc_blocks 1 free_blocks 1 total_blocks 2
  At 0x3ffb0000 len 25480 free 23820 allocated 36 min_free 23820
    largest_free_block 16384 alloc_blocks 1 free_blocks 1 total_blocks 2
  At 0x3ffae6e0 len 6192 free 4532 allocated 36 min_free 4532
    largest_free_block 4096 alloc_blocks 1 free_blocks 1 total_blocks 2
  At 0x3ff80518 len 6872 free 156 allocated 5092 min_free 144
    largest_free_block 8 alloc_blocks 34 free_blocks 1 total_blocks 35
  At 0x3ffb6388 len 7288 free 108 allocated 5556 min_free 68
    largest_free_block 8 alloc_blocks 18 free_blocks 2 total_blocks 20
  At 0x3ffb9a20 len 16648 free 324 allocated 14700 min_free 308
    largest_free_block 16 alloc_blocks 75 free_blocks 1 total_blocks 76
  At 0x3ffce4a0 len 72544 free 45324 allocated 25596 min_free 1552
    largest_free_block 16384 alloc_blocks 120 free_blocks 9 total_blocks 129
  At 0x3ffe0440 len 129984 free 17828 allocated 110532 min_free 7104
    largest_free_block 8192 alloc_blocks 14 free_blocks 4 total_blocks 18
  Totals:
    free 98080 allocated 161584 **min_free 43480** largest_free_block 16384

As long as you say min_free should have remaiened at 9176, but why is bigger now?

jrHernandezSio, 2022-08-01:
See what´s happening when allocating 36864 bytes (when largest block is 32K) :

**Heap summary for capabilities 0x00000004:**
  At 0x3ff80518 len 6872 free 144 allocated 5104 min_free 144
    largest_free_block 0 alloc_blocks 35 free_blocks 0 total_blocks 35
  At 0x3ffb6388 len 7288 free 72 allocated 5592 min_free 72
    largest_free_block 0 alloc_blocks 17 free_blocks 0 total_blocks 17
  At 0x3ffb9a20 len 16648 free 320 allocated 14704 min_free 308
    largest_free_block 8 alloc_blocks 75 free_blocks 1 total_blocks 76
  At 0x3ffce4a0 len 72544 free 52436 allocated 18484 min_free 52040
    largest_free_block 32768 alloc_blocks 116 free_blocks 2 total_blocks 118
  At 0x3ffe0440 len 129984 free 20836 allocated 107524 min_free 20836
    largest_free_block 16384 alloc_blocks 1 free_blocks 1 total_blocks 2
  Totals:
    **free 73808 allocated 151408 min_free 73400 largest_free_block 32768**

` drops_buffer = (char*) malloc(sizeof(char) * (36864) );`

**Heap summary for capabilities 0x00000004:**
  At 0x3ff80518 len 6872 free 144 allocated 5104 min_free 144
    largest_free_block 0 alloc_blocks 35 free_blocks 0 total_blocks 35
  At 0x3ffb6388 len 7288 free 72 allocated 5592 min_free 72
    largest_free_block 0 alloc_blocks 17 free_blocks 0 total_blocks 17
  At 0x3ffb9a20 len 16648 free 320 allocated 14704 min_free 308
    largest_free_block 8 alloc_blocks 75 free_blocks 1 total_blocks 76
  At 0x3ffce4a0 len 72544 free 15572 allocated 55348 min_free 15572
    largest_free_block 8192 alloc_blocks 117 free_blocks 2 total_blocks 119
  At 0x3ffe0440 len 129984 free 20836 allocated 107524 min_free 20836
    largest_free_block 16384 alloc_blocks 1 free_blocks 1 total_blocks 2
  **Totals:
    free 36944 allocated 188272 min_free 36932 largest_free_block 16384**

igrr, 2022-08-01:
> As long as you say min_free should have remaiened at 9176, but why is bigger now?

This is because releasing BLE controller memory has created several new (previously unused) heap blocks. Since we don't track the timeline of `min_free` values, this new block got added to the "total" min_free value. This is indeed confusing, I agree. The better way would be to keep track of the overall "min_free" value at every moment in time; however doing so would also incur some overhead.

Regarding the case when malloc(36864) succeeds with largest_free_block=16384, I can't explain this behavior. Perhaps some memory gets released between the moment when you print the heap summary and the moment you try to allocate? Could you share the code required to reproduce this? This would help us investigate the issue.

jrHernandezSio, 2022-08-01:
Hi,

Just a remark. The allocation of 36864 is done when min_free is 32768.

I can not provide full program, but I hope this piece can help:
If there is not enough memory I send less data.
`
    heap_caps_print_heap_info(MALLOC_CAP_8BIT);

    char *drops_buffer;
    send_max_drops = 6;
    drops_buffer = (char*) malloc(sizeof(char) * (send_max_drops * MAX_PIC_LEN) );  // #define MAX_PIC_LEN     6U*1024U 
    if(drops_buffer == NULL)
    {
        //  Send less data inside POST
        printf("\n\nError malloc !!! -> 2 drops/post mode\n\n");
        send_max_drops = 2;
        drops_buffer = (char*) malloc(sizeof(char) * (send_max_drops * MAX_PIC_LEN) );
    
        if(drops_buffer == NULL)
            printf("\n\nError malloc !!!\n\n");
    }
    heap_caps_print_heap_info(MALLOC_CAP_8BIT);
`

SoucheSouche, 2022-08-08:
Hi @jrHernandezSio,

I tried reproducing the behavior of a successful allocation of 36864 bytes when the largest_free_block is inferior to the requested allocation size but there is still enough free memory available. For this I allocated 7 times 35000 bytes leaving me with 47012 bytes of free memory and 26112 bytes of largest_free_block. as shown below.

```
Heap summary for capabilities 0x00000004:
  At 0x3ffae6e0 len 6432 free 4 allocated 4676 min_free 4
    largest_free_block 0 alloc_blocks 32 free_blocks 0 total_blocks 32
  At 0x3ffb3210 len 183792 free 166372 allocated 15740 min_free 165372
    largest_free_block 163840 alloc_blocks 14 free_blocks 1 total_blocks 15
  At 0x3ffe0440 len 15072 free 13448 allocated 0 min_free 13448
    largest_free_block 13312 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x3ffe4350 len 113840 free 112216 allocated 0 min_free 112216
    largest_free_block 110592 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 292040 allocated 20416 min_free 291040 largest_free_block 163840
------------allocated 35000 bytes of memory at 3ffb7620
Heap summary for capabilities 0x00000004:
  At 0x3ffae6e0 len 6432 free 4 allocated 4676 min_free 4
    largest_free_block 0 alloc_blocks 32 free_blocks 0 total_blocks 32
  At 0x3ffb3210 len 183792 free 131368 allocated 50740 min_free 131368
    largest_free_block 131072 alloc_blocks 15 free_blocks 1 total_blocks 16
  At 0x3ffe0440 len 15072 free 13448 allocated 0 min_free 13448
    largest_free_block 13312 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x3ffe4350 len 113840 free 112216 allocated 0 min_free 112216
    largest_free_block 110592 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 257036 allocated 55416 min_free 257036 largest_free_block 131072

------------allocated 35000 bytes of memory at 3ffbfedc
Heap summary for capabilities 0x00000004:
  At 0x3ffae6e0 len 6432 free 4 allocated 4676 min_free 4
    largest_free_block 0 alloc_blocks 32 free_blocks 0 total_blocks 32
  At 0x3ffb3210 len 183792 free 96364 allocated 85740 min_free 96364
    largest_free_block 96256 alloc_blocks 16 free_blocks 1 total_blocks 17
  At 0x3ffe0440 len 15072 free 13448 allocated 0 min_free 13448
    largest_free_block 13312 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x3ffe4350 len 113840 free 112216 allocated 0 min_free 112216
    largest_free_block 110592 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 222032 allocated 90416 min_free 222032 largest_free_block 110592

------------allocated 35000 bytes of memory at 3ffc8798
Heap summary for capabilities 0x00000004:
  At 0x3ffae6e0 len 6432 free 4 allocated 4676 min_free 4
    largest_free_block 0 alloc_blocks 32 free_blocks 0 total_blocks 32
  At 0x3ffb3210 len 183792 free 61360 allocated 120740 min_free 61360
    largest_free_block 60416 alloc_blocks 17 free_blocks 1 total_blocks 18
  At 0x3ffe0440 len 15072 free 13448 allocated 0 min_free 13448
    largest_free_block 13312 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x3ffe4350 len 113840 free 112216 allocated 0 min_free 112216
    largest_free_block 110592 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 187028 allocated 125416 min_free 187028 largest_free_block 110592

------------allocated 35000 bytes of memory at 3ffd1054
Heap summary for capabilities 0x00000004:
  At 0x3ffae6e0 len 6432 free 4 allocated 4676 min_free 4
    largest_free_block 0 alloc_blocks 32 free_blocks 0 total_blocks 32
  At 0x3ffb3210 len 183792 free 26356 allocated 155740 min_free 26356
    largest_free_block 26112 alloc_blocks 18 free_blocks 1 total_blocks 19
  At 0x3ffe0440 len 15072 free 13448 allocated 0 min_free 13448
    largest_free_block 13312 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x3ffe4350 len 113840 free 112216 allocated 0 min_free 112216
    largest_free_block 110592 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 152024 allocated 160416 min_free 152024 largest_free_block 110592

------------allocated 35000 bytes of memory at 3ffe49ac
Heap summary for capabilities 0x00000004:
[... 44 lines cut ...]
  At 0x3ffae6e0 len 6432 free 4 allocated 4676 min_free 4
    largest_free_block 0 alloc_blocks 32 free_blocks 0 total_blocks 32
  At 0x3ffb3210 len 183792 free 26356 allocated 155740 min_free 26356
    largest_free_block 26112 alloc_blocks 18 free_blocks 1 total_blocks 19
  At 0x3ffe0440 len 15072 free 13448 allocated 0 min_free 13448
    largest_free_block 13312 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x3ffe4350 len 113840 free 7204 allocated 105000 min_free 7204
    largest_free_block 7168 alloc_blocks 3 free_blocks 1 total_blocks 4
  Totals:
    free 47012 allocated 265416 min_free 47012 largest_free_block 26112

Error malloc !!! -> 2 drops/post mode

Heap summary for capabilities 0x00000004:
  At 0x3ffae6e0 len 6432 free 4 allocated 4676 min_free 4
    largest_free_block 0 alloc_blocks 32 free_blocks 0 total_blocks 32
  At 0x3ffb3210 len 183792 free 14064 allocated 168028 min_free 14064
    largest_free_block 13824 alloc_blocks 19 free_blocks 1 total_blocks 20
  At 0x3ffe0440 len 15072 free 13448 allocated 0 min_free 13448
    largest_free_block 13312 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x3ffe4350 len 113840 free 7204 allocated 105000 min_free 7204
    largest_free_block 7168 alloc_blocks 3 free_blocks 1 total_blocks 4
  Totals:
    free 34720 allocated 277704 min_free 34720 largest_free_block 13824
```

SoucheSouche, 2022-09-15:
Hi @jrHernandezSio,
I will consider this issue as closed since the problem you faced couldn't be reproduced.
Feel free to reopen the ticket if needed and provide a mean to reproduce the encountered issue.

closed 2022-11-21 by Alvin1Zhang