espressif/esp-idf #9470: LICENSE is missing copyright info

gril002, 2022-08-01:
Currently the license in the LICENSE file contains " Copyright [yyyy] [name of copyright owner]" ie. , the year and copyright owner are not filled in.

igrr, 2022-08-02:
Hi @gril002,
The part of LICENSE file you are referring to is an appendix explaining how to apply the license to a copyrighted work: https://github.com/espressif/esp-idf/blob/fde4afc67a2035a1f461dc511cd85dde1ef90368/LICENSE#L179
Essentially it provides a template which can be used in source or header files.

If you are looking for the information on the copyrights relevant to the ESP-IDF, please refer to the following documentation page: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/COPYRIGHT.html

gril002, 2022-08-02:
Whoops, that's my bad. But the copyright info is exactly what I was looking for, so thanks for pointing me to it!

closed 2022-08-02 by gril002