espressif/esp-idf #9471: `reattempt_conn` in `nimble` lib works incorrectly

ermacv, 2022-08-01:
## Environment

- Development Kit:     none
- Kit version none
- Module or chip used:  ESP32-C3
- IDF version (run ``git describe --tags`` to find it): v5.0-dev-3954-g271f4e4c40
- Build System:        idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 11.2.0
- Operating System:     Ubuntu 22.04
- Using an IDE?: No
- Power Supply:        USB
## Problem Description

When using `nimble` as BLE library `CONFIG_BT_NIMBLE_ENABLE_CONN_REATTEMPT` option `reattempt_conn` memory access is done incorrectly.
Internally the option source code uses an array `reattempt_conn` that size is equal to `CONFIG_BT_NIMBLE_MAX_CONNECTIONS`. When `CONFIG_BT_NIMBLE_MAX_CONNECTIONS` is equal to 1 then I expect that array access index should be 0 only. But in code `reattempt_conn[ev->conn_handle]` `ev->conn_handle` is equal to 1. So there is array out of bonds error. I analyzed the generated `.map` file and in my case it sets `ble_att_preferred_mtu_val` variable to 0 (that should be unchanged) and the `nimble` stack works incorrectly from this point. Of course it is environment specific and any other variable could be unexpectedly set to 0.
### Expected Behavior
`ble_att_preferred_mtu_val` stays unchanged after BLE conect/disconnect events if there are no `ble_att_set_preferred_mtu` explicit calls when `CONFIG_BT_NIMBLE_ENABLE_CONN_REATTEMPT` is enabled and `CONFIG_BT_NIMBLE_MAX_CONNECTIONS` is set to 1.

### Actual Behavior
`ble_att_preferred_mtu_val` is set to 0 after BLE conect/disconnect events.

### Steps to reproduce
You can use `examples/bluetooth/nimble/blehr` example to reproduce the issue. Default `sdkconfig` could be used instead of `CONFIG_BT_NIMBLE_MAX_CONNECTIONS` default option value (it should be modified from 3 to 1). Then you need to connect with BLE central (nRF Connect mobile app e.g.) and then disconnect.

Then you can add a following snippet at the end of `app_main()` to read current default `mtu`:
```c
    while (true) {
        uint16_t mtu = ble_att_preferred_mtu();
        ESP_LOGI(tag, "mtu = %u", mtu);
        vTaskDelay(pdMS_TO_TICKS(5000));
    }
```
Before the disconnection step it should be equal to 256. After the disconnection it will be 0.

Another way to reproduce the issue is to enable extra runtime assertions by using `CONFIG_BT_NIMBLE_DEBUG` option set to `y`. Then after disconnect step you need to connect again and set the MTU (request MTU in a nRF Connect app) to 23 for example. Then runtime assertion is triggered: `assert failed: ble_att_chan_mtu ble_att.c:450 (mtu >= 23)`.

In any case it is environment specific and it works only if `ble_att_preferred_mtu_val` is placed just after `reattempt_conn` in FW file. If  you have another variable placed after `reattempt_conn` then it will be zeroed (see generated `.map` file in this case).

### Code to reproduce this issue
See comment above.

## Debug Logs

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

rahult-github, 2022-08-03:
Hi @ermacv ,  

 can you please confirm if attached patch works  ?
 
 Thanks. 

[0001-Nimble-Fix-memory-corruption-introduced-due-to-wrong.txt](https://github.com/espressif/esp-idf/files/9248963/0001-Nimble-Fix-memory-corruption-introduced-due-to-wrong.txt)

ermacv, 2022-08-03:
Hello @rahult-github,
Thank you for the provided patch. It seems to work now. I don't see any array out of bound access anymore.

rahult-github, 2022-08-04:
Hi @ermacv , 

Thanks for confirming. ev->conn_handle has range defined in spec. So it can be more than 1 (i.e. multiple simultaneous connections) . ESP controllers give conn handle starting with value 1. 

Thanks,

ermacv, 2022-08-10:
Hi @rahult-github 
Fine for me. I guess the issue could be closed.