espressif/esp-idf #9474: PLL_D2_CLK clock source support for LCD Peripheral

rmshub, 2022-08-01:
LCD hal driver only supports two clock sources PLL_F160M_CLK & XTAL_CLK. Is there any limitation with PLL_D2_CLK ? Can you please provide support for PLL_D2_CLK clock source. I can drive maximum of 8 MHz as LCD PCLK if use PLL_F160M_CLK  as source. If I configured as LCD PCLK as 10MHz got distortion in display.

LCD Controller: ILI9488 16bit (supports up to 20MHz) 480x320
Module: ESP32-S3-WROOM-1-N16R2
Graphical library: LVGL
LVGL buffers: Internal memory

suda-morris, 2022-08-02:
Hi @rmshub 
This commit enabled the support of PLL240MHz: https://github.com/espressif/esp-idf/commit/a149afb0c064dc1144d37e745c914cd6dc08b8e8

rmshub, 2022-08-02:
It is enabled only for RGB panel. I need support for i8080 bus also.

suda-morris, 2022-08-03:
> I can drive maximum of 8 MHz as LCD PCLK if use PLL_F160M_CLK as source. If I configured as LCD PCLK as 10MHz got distortion in display.

So have you checked the PCLK frequency is 10MHz indeed? The i80 bus resolution is 80MHz by default (if you select the PLL160M as the clock source). So if you set PCLK=10M, which means the divisor is 80/10=8, it resides in a valid range. 

This [example](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/lcd/i80_controller/main/i80_controller_example_main.c#L28) also uses 10MHz PCLK, but we haven't noticed the distortion.

rmshub, 2022-08-03:
My LCD is working properly up to 20 MHz clock with https://github.com/lovyan03/LovyanGFX.
Example: https://github.com/Makerfabs/Makerfabs-ESP32-S3-Parallel-TFT-with-Touch/blob/main/firmware/SD16_3.5/SD16_3.5.ino

LCD configuration can be found here: https://github.com/lovyan03/LovyanGFX/blob/master/src/lgfx/v1/platforms/esp32s3/Bus_Parallel16.cpp

suda-morris, 2022-08-04:
> LCD configuration can be found here: https://github.com/lovyan03/LovyanGFX/blob/master/src/lgfx/v1/platforms/esp32s3/Bus_Parallel16.cpp

I saw you allocated the i80 bus by `esp_lcd_new_i80_bus`, but after that, I didn't see you call the `esp_lcd_new_panel_io_i80` to allocate an i80 device. So you're not using the esp-idf driver but writing your own one? If so, please take care of the clock calculation part, AFAIK, there is some hardware bug there. The driver used in esp-idf has somehow workaround that issue by not using the fractional clock, and also avoiding setting the `lcd_clkcnt_n` to `1`.

rmshub, 2022-08-23:
20MHz lcd clock is working after updating esp-idf version to [v5.0-beta1](https://github.com/espressif/esp-idf/releases/tag/v5.0-beta1).

closed 2022-08-23 by rmshub