espressif/esp-idf #9477: using mdns with esp-idf v4 & v5

nopnop2002, 2022-08-02:
## Environment
```
$ idf.py --version
ESP-IDF v4.4.1-405-g6c5fb29c2c
```

## Problem Description

Since ESP-IDF V5, mdns is now a component.

When using mdns with ESP-IDF V5, idf_component.yml is required.

```
dependencies:
  espressif/mdns: "^1.0.3"
  idf:
     version: ">=4.1.0"
```

An error will occur if i build same project with ESP-IDF V4.4.
```
  idf_component_manager.version_solver.mixology.failure.SolverFailure:
  Because no versions of espressif/mdns match >1.0.3,<1.0.4 || >1.0.4,<1.0.5
  || >1.0.5,<2.0.0

   and espressif/mdns (1.0.3) depends on idf (>=5.0), espressif/mdns (>=1.0.3,<1.0.4 || >1.0.4,<1.0.5 || >1.0.5,<2.0.0) requires idf (>=5.0).

  And because espressif/mdns (1.0.4) depends on both idf (>=5.0) and idf
  (>=5.0), espressif/mdns (>=1.0.3,<2.0.0) requires idf (>=5.0.0).

  So, because no versions of idf match >=5.0.0

   and project depends on espressif/mdns (^1.0.3), version solving failed.

Call Stack (most recent call first):
  /home/nop/esp-idf/tools/cmake/project.cmake:384 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "/home/nop/rtos/esp-idf-mqtt-client/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
```

Is there a way to build the same project with IDF V4 and V5?

igrr, 2022-08-02:
Hi @nopnop2002,
You can add the condition to install the managed component only for IDF v5.0, like so: https://github.com/espressif/esp-rainmaker/blob/5c81ecf755432ebeffaadc966b67799ea2f7c9d3/components/esp_rainmaker/idf_component.yml#L6

nopnop2002, 2022-08-02:
@igrr 

Thank you.

I fonund these in github:

```
    rules: # will add "optional_component" only when all if clauses are True
      - if: "idf_version >=3.3,<5.0" # supports all SimpleSpec grammars (https://python-semanticversion.readthedocs.io/en/latest/reference.html#semantic_version.SimpleSpec)
      - if: "target in [esp32, esp32c3]" # supports boolean operator ==, !=, in, not in.
```

closed 2022-08-02 by nopnop2002