espressif/esp-idf #9478: Printf function not working properly

ankit-thealchemist, 2022-08-02:
## Environment

- Development Kit:      [esp32 dev kit cv4

- IDF version  v4.4.1
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     UBUNTU 22.04 LTS

- Using an IDE?:vscode
- Power Supply:         [USB]

## Problem Description

printf function not working as expected

### Expected Behavior

print the argument

### Actual Behavior

does not print any thing. Only print when using the "\n". example printf("hello world\n") will be printed

### Code to reproduce this issue

```cpp

#include <stdio.h>

void app_main(void)
{
    printf("hello world");
}

```

igrr, 2022-08-02:
Hi @ankit-thealchemist,
Standard I/O streams in ESP-IDF are line buffered.

(Quote of the relevant part of the standard follows:)
> When a stream is unbuffered, characters are intended to appear from the source or at the destination as soon as possible. Otherwise characters may be accumulated and transmitted to or from the host environment as a block. When a stream is fully buffered, characters are intended to be transmitted to or from the host environment as a block when a buffer is filled. When a stream is line buffered, characters are intended to be transmitted to or from the host environment as a block when a new-line character is encountered. Furthermore, characters are intended to be transmitted as a block to the host environment when a buffer is filled, when input is requested on an unbuffered stream, or when input is requested on a line buffered stream that requires the transmission of characters from the host environment. Support for these characteristics is implementation-defined, and may be affected via the setbuf and setvbuf functions.

You can use `setvbuf` function to make the stream unbuffered. Alternatively, you call call fflush and fsync functions to manually flush the output stream, when necessary.

ankit-thealchemist, 2022-08-02:
Thanks for sharing the information!

closed 2022-08-02 by ankit-thealchemist