espressif/esp-idf #9479: A2dp sink volume control turning off

qombajn, 2022-08-02:
I would like to "lock" volume on my esp32 sink module and i don't know where i have to change received value for my constant value. I was trying to change volume_set_by_controller and volume_set_by_local_host but it gives no response. Only printing right value in terminal. How can I turn off this feature and don't receive volume from host(phone). If this possible to set 100% on my phone and 10% volume on esp sink??

Wth-Esp, 2022-08-02:
Maybe refer to https://github.com/espressif/esp-idf/issues/7875#issuecomment-965997813

qombajn, 2022-08-02:
> Maybe refer to [#7875 (comment)](https://github.com/espressif/esp-idf/issues/7875#issuecomment-965997813)
Maybe it would work but I need it to work with every devices without doing changes in the phone.

fracho123, 2022-08-25:
I'm pretty sure you can do this by modifying the write_ringbuf(2) function in the bt_app_core.c file. The a2dp_sink example doesn't implement actual physical volume changes from what I can tell. I was able to get volume control working by using @mooalot's solution [here](https://www.esp32.com/viewtopic.php?t=15505).

BetterJincheng, 2022-09-28:
@qombajn a2dp_sink example does not implement physical volume changed on ESP32. It just prints the volume value for an informational purpose. The static variable `s_volume` is left to the user to implement the physical volume control.

If you want to disable the log of volume change, please set `s_volume_notify` to `false`.

Turn to us anytime.
Thanks!

closed 2023-03-29 by Alvin1Zhang