espressif/esp-idf #9480: ESP32-S3 BLE 5.0 backwards compatibility with older devices

krupis, 2022-08-02:
Hello. Its been a while. I have been further testing this with LightBlue and NRF connect app with different phones and there is my results:

1. **Using wire shark connected to NRF52840 :**

[image]

2. Using LightBlue app on the Samsung galaxy S8 edge Android version 9. The ESP32 BLE does not show up in the scanned  device list. 
[image]

3. Using NRF connect app on the Samsung galaxy S8 edge Android version 9. The ESP32 BLE does not show up in the scanned device list

[image]

4. Using Lightblue app on the Pocophone F1 Android version 9
[image]

5. Using NRF connect on Pocophone F1 android version 9
[image]

6. Using Lightblue on Huawei P10 android version 9
[image]

7. Using NRF Connect on Huawei P10 android version 9
[image]

8. Using NRF Connect on IPhone 12 pro max
[image]

Results summed up:
only the Iphone 12 pro max and Pocohpone F1 were able to capture the ESP32 5.0 BLE device.

Samsung galaxy s8 edge and Huawei p10 did not discover ESP32-S3 5.0 BLE device.

Is BLE 5.0 not backwards compatible with BLE 4? I was under impression that if the phone does not support BLE 5.0 I would still be able to connect to this device. Could someone clarify to me how does backwards compatibility work and how to overcome this issue?

What if I am developing code with ESP32-S3 and ble 5.0 and the client develop app for it. How to ensure that older phones that do not support BLE 5.0 can still use the app to connect to my device?

esp-cjh, 2022-08-02:
The ble50_security_server example use extended advertising. Therefore, the phone doesn't support BLE 5.0 (extended scan) cannot receive extended advertising packet. If you want to backwards compatibility with older devices, you can send legacy advertising packet.

krupis, 2022-08-02:
> The ble50_security_server example use extended advertising. Therefore, the phone doesn't support BLE 5.0 (extended scan) cannot receive extended advertising packet. If you want to backwards compatibility with older devices, you can send legacy advertising packet.

Thanks for rapid response. Can you answer the following:

1. If 5.0 is backwards compatible with 4.2, why it wouldnt show up?
2. What exactly is the difference between extended advertising and legacy advertising?
3. Multi-advertisment example program relies on extended ble functions to achieve what I want. A sketch from the program:
```
    // 1M phy extend adv, Connectable advertising
    FUNC_SEND_WAIT_SEM(esp_ble_gap_ext_adv_set_params(0, &ext_adv_params_1M), test_sem);
    FUNC_SEND_WAIT_SEM(esp_ble_gap_ext_adv_set_rand_addr(0, addr_1m), test_sem);
    FUNC_SEND_WAIT_SEM(esp_ble_gap_config_ext_adv_data_raw(0, 30, &raw_adv_data2[0]), test_sem);

    // 2M phy extend adv, Scannable advertising
    FUNC_SEND_WAIT_SEM(esp_ble_gap_ext_adv_set_params(1, &ext_adv_params_2M), test_sem);
    FUNC_SEND_WAIT_SEM(esp_ble_gap_ext_adv_set_rand_addr(1, addr_2m), test_sem);
    FUNC_SEND_WAIT_SEM(esp_ble_gap_config_ext_adv_data_raw(1, sizeof(raw_adv_data_2m), &raw_adv_data_2m[0]), test_sem);

    // start all adv
    FUNC_SEND_WAIT_SEM(esp_ble_gap_ext_adv_start(2, &ext_adv[0]), test_sem);
```

The above will create 2 different advertisment profiles. Could you advice me how can I create the same using legacy functions?

esp-cjh, 2022-08-02:
1. BLE 5.0 support extended advertising, 4.2 not support.
2. You can refer to Core Spec 5.0 Vol6 PartB 4.4.2.2.
3. Legacy advertising cannot create multi adv profiles, but you can use extended adv to create multi legacy adv. Also refer to Multi-adv example.
```
    // 1M phy legacy adv, ADV_IND
    FUNC_SEND_WAIT_SEM(esp_ble_gap_ext_adv_set_params(2, &legacy_adv_params), test_sem);
    FUNC_SEND_WAIT_SEM(esp_ble_gap_ext_adv_set_rand_addr(2, addr_legacy), test_sem);
    FUNC_SEND_WAIT_SEM(esp_ble_gap_config_ext_adv_data_raw(2, sizeof(legacy_adv_data), &legacy_adv_data[0]), test_sem);
    FUNC_SEND_WAIT_SEM(esp_ble_gap_config_ext_scan_rsp_data_raw(2, sizeof(legacy_scan_rsp_data), &legacy_scan_rsp_data[0]), test_sem);
```

krupis, 2022-08-02:
> 1. BLE 5.0 support extended advertising, 4.2 not support.
> 2. You can refer to Core Spec 5.0 Vol6 PartB 4.4.2.2.
> 3. Legacy advertising cannot create multi adv profiles, but you can use extended adv [...]

Amazing help. Thanks a lot for clarifying!

closed 2023-01-12 by Alvin1Zhang