espressif/esp-idf #9481: LCD接口例程

gao-zhida, 2022-08-02:
在使用lcd接口时，调用官方例程时发现并没有SSD1963IC的驱动，官方大大可不可以添加一个进来？谢谢

suda-morris, 2022-08-03:
Hi @gao-zhida 

We won't add more LCD controller drivers into esp-idf repo, but will consider adding that as a managed component, which you can find in https://components.espressif.com/search/lcd

Also you're welcome to submit any PRs to https://github.com/espressif/esp-bsp/tree/master/components/lcd

suda-morris, 2022-08-03:
BTW, alternatively, you can take [iot-solution](https://github.com/espressif/esp-iot-solution/tree/master/components/display/screen/controller_driver/ssd1963) driver as the starting point, plus some clean-up to adapter to the esp_lcd interface.

gao-zhida, 2022-08-03:
> 顺便说一句，或者，您可以将[iot-solution](https://github.com/espressif/esp-iot-solution/tree/master/components/display/screen/controller_driver/ssd1963)驱动程序作为起点，加上一些清理以适应 esp_lcd 接口。
okay, thank you

closed 2024-12-25 by suda-morris