espressif/esp-idf #9482: Hidden stack overflow

jrHernandezSio, 2022-08-02:
Hi,

I am having a crash when starting my device, but it is not happening consistently, it happens only sometimes and there is no way to reproduce it.

I would like to know if this may be related to stack overflow problem. If so, how can I increase stack size for main task?
The main task is the only task running when creashing.

```
Core  0 register dump:
PC      : 0x40093aad  PS      : 0x00060330  A0      : 0x8012f41a  A1      : 0x3ffbacc0
A2      : 0xc000b83f  A3      : 0xffffffff  A4      : 0x00060b20  A5      : 0x08000000
A6      : 0x00000009  A7      : 0x3ffbab94  A8      : 0x3ffc3e14  A9      : 0x00000001
A10     : 0x00000000  A11     : 0x00060323  A12     : 0x3f400464  A13     : 0x3f4001f8
A14     : 0x00000400  A15     : 0x3f400bdc  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0xc000b87f  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc

Backtrace:0x40093aaa:0x3ffbacc0 0x4012f417:0x3ffbad00 0x400dc207:0x3ffbad40 0x400dd08d:0x3ffbad80 0x400dd09f:0x3ffbada0 0x400dd307:0x3ffbadc0 0x400dd371:0x3ffbade0 0x400d9075:0x3ffbae00 0x400d3a47:0x3ffbae20 0x400d4b01:0x3ffbae40 0x400d577e:0x3ffbae60 0x400d6cb3:0x3ffbae90 0x401c5b27:0x3ffbafd0 0x40095db9:0x3ffbaff0
  #0  0x40093aaa:0x3ffbacc0 in xQueueSemaphoreTake at C:\Users\jr.hernandez\.platformio\packages\framework-espidf\components\freertos/queue.c:1447 (discriminator 1)
  #1  0x4012f417:0x3ffbad00 in uart_flush_input at C:\Users\jr.hernandez\.platformio\packages\framework-espidf\components\driver/uart.c:1270 (discriminator 2)
  #2  0x400dc207:0x3ffbad40 in SI_execute at lib/SERIAL_INTERFACE/serial_interface.c:297
  #3  0x400dd08d:0x3ffbad80 in is_attached at lib_prj/BG95/bg95.c:535
  #4  0x400dd09f:0x3ffbada0 in detach at lib_prj/BG95/bg95.c:553
  #5  0x400dd307:0x3ffbadc0 in BG95_off_avd at lib_prj/BG95/bg95.c:758
  #6  0x400dd371:0x3ffbade0 in BG95_off at lib_prj/BG95/bg95.c:732
  #7  0x400d9075:0x3ffbae00 in SEND_get_barcodes_to_file at src/send_manager.c:2178
  #8  0x400d3a47:0x3ffbae20 in HD_rx_barcodes_to_file at src/main.c:970
  #9  0x400d4b01:0x3ffbae40 in SYSTEM_notify at src/main.c:1714
  #10 0x400d577e:0x3ffbae60 in SYSTEM_setup at src/main.c:1917
  #11 0x400d6cb3:0x3ffbae90 in app_main at src/main.c:6600
  #12 0x401c5b27:0x3ffbafd0 in main_task at C:\Users\jr.hernandez\.platformio\packages\framework-espidf\components\freertos\port/port_common.c:133 (discriminator 2)
  #13 0x40095db9:0x3ffbaff0 in vPortTaskWrapper at C:\Users\jr.hernandez\.platformio\packages\framework-espidf\components\freertos\port\xtensa/port.c:168

ELF file SHA256: 3639363134373436

Rebooting...

Boot!...
```

I would like to know if this may be related to stack overflow problem. If so, how can I increase stack size for main task?

I have tried to check usStackHighWaterMark, but not sure how to get xHandle for main task.

Regards, Ramon.

jrHernandezSio, 2022-08-02:
I have decreased main stack from 3584 to 2000 and problem did not happen.

ESP-Marius, 2022-08-03:
Hi,

Your panic output does seem to indicate some kind of memory corruption (See the invalid address`EXCVADDR = 0xc000b87f`)
There are some hints on how to debug heap corruption [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html#finding-heap-corruption). 

Try to enable [Heap poisning](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html#configuration) and [stackoverflow detection](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-freertos-check-stackoverflow) before running it again and see if that can help you track down the issue.

> I have decreased main stack from 3584 to 2000 and problem did not reproduce. Below 2000 it crashses, but caused by other funtions.

Did you try to increase the stacksize a lot and see if that solved the issue? E.g. increase it to 8000 bytes.

> I have tried to check usStackHighWaterMark, but not sure how to get xHandle for main task.
Calling `uxTaskGetStackHighWaterMark(NULL)` from your task should be sufficient to give you the watermark.

jrHernandezSio, 2022-08-03:
Thanks for the tip !

Yes, I increased stack to 4000 and enable detection. Let´s see if gets reproduced.

ESP-Marius, 2022-09-06:
Closing this issue due to lack of activity, feel free to reopen it if the issue still occurs and you want to continue the investigation.

closed 2022-09-06 by ESP-Marius