espressif/esp-idf #9483: BLE failed to get charactreristic value after reconnection

jrHernandezSio, 2022-08-02:
Hi,

**ENVIRONMENT**: 

- espressif32@3.3.2 -> framework-espidf @ 3.40300.0 (4.3.0)
- BLE driver: Bluedroid (loaded dynamically)

**DESCRIPTION**
I am trying to perform a new connection and BLE transfer after being connected or have tried being connected. But I am finding troubles when reading characterictics value.

It only works ok properly when coming freshly from deep_sleep (charactristic value is read properly by ESP when phone writes via BLE). But during the same boot I always find the same trouble. The fisrt characteristic is not read properly, read value is not the expected one.

I have tried several things:
 **- Unregister ble** app via `esp_ble_gatts_app_unregister` -> no success

```
    esp_ble_gap_stop_advertising();
    esp_bluedroid_disable();
    esp_bluedroid_deinit();

    esp_bt_controller_disable();
    esp_bt_controller_deinit();

    esp_ble_gatts_app_unregister(gl_profile_tab[PROFILE_A_APP_ID].gatts_if);
```

 **- Unload full BLE driver** via -> no success
```
    esp_ble_gap_stop_advertising();
    esp_bluedroid_disable();
    esp_bluedroid_deinit();

    esp_bt_controller_disable();
    esp_bt_controller_deinit();

    esp_bt_mem_release( ESP_BT_MODE_BLE );   // OK for deep sleep  // crash from idle

```

Any help will be welcome.

esp-cjh, 2022-08-03:
Could you please provide complete and detailed steps to reproduce this problem? Could you please provide more information such as complete log, chip, esp-idf version (4.3.0?), and example? You can also try with esp-idf 4.4.1 to see if this problem occurs.

jrHernandezSio, 2022-08-03:
Hi,

Thanks for the support. 
It is 4.3.0.
Hard to share the full project to reproduce. Let me see the behaviour with 4.4.1 and get back to you.

jrHernandezSio, 2022-08-03:
**Updated to framework-espidf @ 3.40401.0 (4.4.1)**

Behaviour is the same.
I am afraid I can not povide code to reproduce, but let me give you more details.

**Connection right fafter deep sleep:**

BLE connection is ok and the characteristic is read:
Read_char         = 0x D4 77 B8 E1 AC 0A 26 72 FC 4C 56 53 A8 94 57 A8 58 A7 34 0D 74 50 67 8B
Token_encrypt  = 0x D4 77 B8 E1 AC 0A 26 72 FC 4C 56 53 A8 94 57 A8 58 A7 34 0D 74 50 67 8B (matched ok)

Read_char         = 0x E0 F0 D2 82 A1 E2 52 E9 91 8B 23 35 B7 55 13 D5 CE B7 53 F0 30 66 9B 30
Token_encrypt  = 0x E0 F0 D2 82 A1 E2 52 E9 91 8B 23 35 B7 55 13 D5 CE B7 53 F0 30 66 9B 30 (matched ok)

**2nd connection (after first failed connection):**

Read_char         = 0x D4 77 B8 E1 AC 0A 26 72 FC 4C 56 53 A8 94 57 A8 58 A7 34 0D 74 50 67 8B
Token_encrypt  = 0x 98 5E 4A 69 28 E6 DF 04 2B 3A 9E 07 9D 7E 47 9F 75 85 C9 C5 15 E6 C5 11 (Read_char is supposed to match this)

**2nd connection (after first failed connection) ( Step by step):

first failed connection
Token_encrypt(Reg) = 0x 10 42 94 7E FD 94 CA 66 46 C1 8D 97 15 C2 36 FC E6 74 67 E9 7E 9E 0D 6B

2nd connection
Token_encrypt(Reg) = 0x A1 AE AB FE 32 5F 68 14 67 49 FB BD 8B F4 58 E4 F9 83 E9 B6 34 38 DD 6B

Read_char              = 0x 0A C0 E7 95 35 70 4E 1F 3E A7 F1 05 37 3E 86 A5 44 B7 B8 6C F8 B6 80 A2 (but appears this instead)
Token_encrypt       = 0x A1 AE AB FE 32 5F 68 14 67 49 FB BD 8B F4 58 E4 F9 83 E9 B6 34 38 DD 6B (Matched ko)

It seems somehing related to initialization inmy end, but I have tracked dow and I do not know why the read charateristic value is sth different to the expected one.

Let me enable info debug leve and provide full log.

Regards, Ramon.

esp-cjh, 2022-08-03:
Sure, you don't need to provide the source code.

jrHernandezSio, 2022-08-03:
**Connection right fafter deep sleep:**

`Token_encrypt(Reg) = 0x AD C2 1C 44 17 33 FD A9 8C 0F 5C F2 33 86 39 B4 89 05 79 85 6A 3D 55 22
-DEBUG-: [        36,093 s]
-DEBUG-: [        36,102 s] BLE stat: BLUDRD(2) BT (2)
-DEBUG-: [        36,102 s] BLE connecting...
-DEBUG-: [        36,106 s] gatts_event_handler......
I (4008) GATTS_TABLE: create attribute table successfully, the number handle = 13

-DEBUG-: [        36,120 s] gatts_event_handler......
I (4028) GATTS_DEMO: SERVICE_START_EVT, status 0, service_handle 40

-DEBUG-: [        36,131 s] BLE conn ......... (0)
-DEBUG-: [        36,372 s] BLE conn ......... (0)
-DEBUG-: [        36,460 s] gatts_event_handler......
I (4358) GATTS_DEMO: ESP_GATTS_CONNECT_EVT, conn_id 0, remote 75:7a:c5:64:f5:9e:
-DEBUG-: [        36,462 s] set 3 for gatts_if_for_indicate
-DEBUG-: [        36,602 s] BLE conn ......... (0)
-DEBUG-: [        36,842 s] BLE conn ......... (0)
-DEBUG-: [        37,084 s] BLE conn ......... (0)
-DEBUG-: [        37,332 s] BLE conn ......... (0)
-DEBUG-: [        37,572 s] BLE conn ......... (0)
-DEBUG-: [        37,673 s] gatts_event_handler......
I (5578) GATTS_DEMO: ESP_GATTS_MTU_EVT, MTU 156
-DEBUG-: [        37,785 s] gatts_event_handler......
I (5688) GATTS_TABLE: ESP_GATTS_READ_EVT
-DEBUG-: [        37,812 s] BLE conn ......... (0)
-DEBUG-: [        37,855 s] gatts_event_handler......
I (5758) GATTS_TABLE: GATT_WRITE_EVT, handle = 42, value len = 24, value :
I (5758) GATTS_TABLE: ad c2 1c 44 17 33 fd a9 8c 0f 5c f2 33 86 39 b4 
I (5768) GATTS_TABLE: 89 05 79 85 6a 3d 55 22 
I (5768) GATTS_TABLE: GATT_WRITE_EVT, handle = 47
-DEBUG-: [        37,875 s] length = 18

-DEBUG-: [        37,878 s] Read_char      = 0x AD C2 1C 44 17 33 FD A9 8C 0F 5C F2 33 86 39 B4 89 05 79 85 6A 3D 55 22
-DEBUG-: [        37,889 s]
Token_encrypt = 0x AD C2 1C 44 17 33 FD A9 8C 0F 5C F2 33 86 39 B4 89 05 79 85 6A 3D 55 22
-DEBUG-: [        37,900 s]
-DEBUG-: [        37,902 s] EQUAL TOKENNN !!
-DEBUG-: [        37,907 s] gatts_event_handler......
I (5808) GATTS_TABLE: ESP_GATTS_READ_EVT
-DEBUG-: [        37,916 s] gatts_event_handler......
-DEBUG-: [        37,943 s] gatts_event_handler......
I (5848) GATTS_TABLE: ESP_GATTS_READ_EVT
-DEBUG-: [        37,988 s] gatts_event_handler......
I (5888) GATTS_TABLE: ESP_GATTS_READ_EVT
-DEBUG-: [        38,054 s] gatts_event_handler......
I (5958) GATTS_TABLE: GATT_WRITE_EVT, handle = 45, value len = 2, value :
I (5958) GATTS_TABLE: 01 00 
I (5958) GATTS_TABLE: GATT_WRITE_EVT, handle = 47
-DEBUG-: [        38,122 s] BLE conn ......... (1)
-DEBUG-: [        38,408 s] BLE Connected :)`

**2nd connection (after first failed connection):**

`-DEBUG-: [1659511941,518 s] BLE advertising...
-DEBUG-: [1659511941,522 s] BLE stat: BLUDRD(0) BT (0)
I (3978) BTDM_INIT: BT controller compile version [5688ed5]
I (3988) system_api: Base MAC address is not set
I (3988) system_api: read default base MAC address from EFUSE
I (3988) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (4248) GAP: register callback
-DEBUG-: [1659511942,153 s] gatts_event_handler......
I (4248) GATTS_DEMO: REGISTER_APP_EVT, status 0, app_id 0

-DEBUG-: [1659511942,155 s]
Token_encrypt(Reg) = 0x 10 42 94 7E FD 94 CA 66 46 C1 8D 97 15 C2 36 FC E6 74 67 E9 7E 9E 0D 6B
-DEBUG-: [1659511942,166 s]
-DEBUG-: [1659511942,174 s] BLE stat: BLUDRD(2) BT (2)
-DEBUG-: [1659511942,175 s] BLE connecting...
[... 163 lines cut ...]
I (44988) GATTS_TABLE: GATT_WRITE_EVT, handle = 47
-DEBUG-: [1659511982,821 s] length = 18

-DEBUG-: [1659511982,825 s] Read_char      = 0x A5 18 B3 78 34 96 D0 97 58 4F A3 11 D0 65 9E D6 5D C1 81 B6 56 17 C9 00
-DEBUG-: [1659511982,836 s]
Token_encrypt = 0x 26 79 86 97 08 9E 95 C3 13 6A 70 8A 18 F1 BF 21 67 47 D2 35 D9 A2 AC 80
-DEBUG-: [1659511982,847 s]
-DEBUG-: [1659511982,849 s] WRONG TOKENNN !!
-DEBUG-: [1659511982,854 s] gatts_event_handler......
I (45028) GATTS_TABLE: ESP_GATTS_READ_EVT
-DEBUG-: [1659511982,863 s] gatts_event_handler......
I (45038) GAP: Close session beacuse 2 in operation result
-DEBUG-: [1659511982,873 s] Wrong token -> disconnect BLE
-DEBUG-: [1659511982,891 s] gatts_event_handler......
I (45068) GATTS_TABLE: ESP_GATTS_READ_EVT
-DEBUG-: [1659511982,957 s] gatts_event_handler......
I (45128) GATTS_TABLE: ESP_GATTS_READ_EVT
-DEBUG-: [1659511984,058 s] gatts_event_handler......
I (46228) GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT, disconnect reason 0x13
-DEBUG-: [1659511987,869 s] BLE stat: BLUDRD(2) BT (2) 
W (50048) BT_APPL: bta_dm_disable BTA_DISABLE_DELAY set to 200 ms
I (50058) GAP: Stop adv successfully
-DEBUG-: [1659511988,085 s] BLE stat: BLUDRD(0) BT (0) 
-DEBUG-: [1659511988,086 s] BLE conn ......... (0)
I (50488) GAP: Close session beacuse 2 in operation result`

esp-cjh, 2022-08-03:
You write a token to a characteristic and then read the same characteristic to get the token, right? If so, the problem is that the token you wrote is incorrect or the data received by the ESP32 is incorrect. Could you please capture the air interface packets and provide the capture file?

jrHernandezSio, 2022-08-04:
Yes, just like that. A phone gets connected to the ESP32 and write this token in the ESP32.

How can I capture the air interface pakets? 
I guess I could run "nRf connect" or similar in the phone, but the BT is taken by the App running already connected to the ESP32.

esp-cjh, 2022-08-04:
You can use nrf sniffer which is a dongle plugged into the computer and use Wireshark software to capture the air packets.

jrHernandezSio, 2022-08-04:
I do not have it righ now, but I think I can use only nRF connect and mock the App controlling the Token.
Let me try and get back to you.

jrHernandezSio, 2022-08-05:
Trying to get  the log I have found out the root cause. There was a buffer not being reset properly and the token encryption result was incorrect.

So sorry.

Thanks anyway. Ramón.

closed 2022-08-05 by jrHernandezSio

esp-cjh, 2022-08-05:
It does not matter.