espressif/esp-idf #9486: wifi sometimes does not connect and emits strange log output

0xFEEDC0DE64, 2022-08-02:
on the esp32 i noticed that sometimes the wifi connection does not work until power cycled, and I had a case now with usb programmer attached and noticed a lot of "wifi:Send SA Query req with transaction id XXXX" lines:

```
I (552951) wifi:Send SA Query req with transaction id 237b                                                                                                                                                                                    
I (553151) wifi:Send SA Query req with transaction id 5bb7                                                                                                                                                                                    
I (553361) wifi:Send SA Query req with transaction id b22e                                                                                                                                                                                    
I (553561) wifi:Send SA Query req with transaction id 842a                                                                                                                                                                                    
I (553771) wifi:Send SA Query req with transaction id d66f                                                                                                                                                                                    
I (553971) wifi:Send SA Query req with transaction id 4731                                                                                                                                                                                    
I (554181) wifi:Send SA Query req with transaction id 5dbe                                                                                                                                                                                    
I (554381) wifi:Send SA Query req with transaction id 870c                                                                                                                                                                                    
I (554591) wifi:Send SA Query req with transaction id 6840                                                                                                                                                                                    
I (554791) wifi:Send SA Query req with transaction id 93c5                                                                                                                                                                                    
I (555001) wifi:Send SA Query req with transaction id 4a87                                                                                                                                                                                    
I (555201) wifi:Send SA Query req with transaction id b31                                                                                                                                                                                     
I (555401) wifi:Send SA Query req with transaction id 6537                                                                                                                                                                                    
I (555611) wifi:Send SA Query req with transaction id 68f9                                                                                                                                                                                    
I (555811) wifi:Send SA Query req with transaction id 6506                                                                                                                                                                                    
I (556021) wifi:Send SA Query req with transaction id 6270                                                                                                                                                                                    
I (556221) wifi:Send SA Query req with transaction id 4158                                                                                                                                                                                    
I (556431) wifi:Send SA Query req with transaction id 68d9                                                                                                                                                                                    
I (556631) WIFI_STACK espwifistack.cpp:1631 wifi_event_cb(): event_base=WIFI_EVENT event_id=WIFI_EVENT_AP_STADISCONNECTED                                                                                                                     
I (556631) WIFI_STACK espwifistack.cpp:1634 wifi_event_cb(): AP Station Disconnected: MAC: 34:cf:f6:6b:03:cd, AID: 1                                                                                                                          
I (556631) wifi:new:<11,2>, old:<11,2>, ap:<11,2>, sta:<11,0>, prof:1                                                                                                                                                                         
I (556651) wifi:Send SA Query req with transaction id ccf7                                                                                                                                                                                    
I (556861) wifi:Send SA Query req with transaction id 51e6                                                                                                                                                                                    
I (557061) wifi:Send SA Query req with transaction id eb93                                                                                                                                                                                    
I (557271) wifi:Send SA Query req with transaction id 5334                                                                                                                                                                                    
I (557471) wifi:Send SA Query req with transaction id 3928                                                                                                                                                                                    
I (557681) wifi:Send SA Query req with transaction id 704b                                                                                                                                                                                    
I (557881) wifi:Send SA Query req with transaction id 6b58                                                                                                                                                                                    
I (558091) wifi:Send SA Query req with transaction id 7256                                                                                                                                                                                    
I (558231) wifi:new:<11,2>, old:<11,2>, ap:<11,2>, sta:<11,0>, prof:1                                                                                                                                                                         
I (558231) wifi:Send SA Query req with transaction id cc06                                                                                                                                                                                    
I (558431) wifi:Send SA Query req with transaction id 4077                                                                                                                                                                                    
I (558641) wifi:Send SA Query req with transaction id 4985                                                                                                                                                                                    
I (558841) wifi:Send SA Query req with transaction id 863c                                                                                                                                                                                    
[... cut ...]
I (574841) wifi:Send SA Query req with transaction id 4d41
I (575051) wifi:Send SA Query req with transaction id 1b
I (575251) wifi:Send SA Query req with transaction id 89d
I (575461) wifi:Send SA Query req with transaction id 8d0
I (575661) wifi:Send SA Query req with transaction id db22
I (575871) wifi:Send SA Query req with transaction id 7dbe
I (576071) wifi:Send SA Query req with transaction id ed3b
I (576271) wifi:Send SA Query req with transaction id 124f
```

Is there anything I could try to fix this randomly occuring issue?

Alvin1Zhang, 2022-08-03:
Thanks for reporting.

MaxwellAlan, 2022-08-04:
Hi @0xFEEDC0DE64 

When AP and STA has a secure PMF Association, AP or STA will send SA Query Request frame when it received Deauth or Associate/Reassociate req frame, which will triger SA Query Procedure.  We will look into this issue.
You can disable PMF feature to workaround for this issue now.

Thanks.

nachiketkukade, 2022-08-04:
@0xFEEDC0DE64 , can you please share below details? -
1. IDF version (run ``git describe --tags`` to find it)
2. Any steps or commands after which this issue is seen

0xFEEDC0DE64, 2022-08-05:
version: v5.0-dev-3055-ge51d850393 (around 20-30 commits are my own patches, but I do not patch any wifi related stuff)

no steps, I usually just power-cycle. next time I see this on my dev setup I will try more stuff, for example the access point

nachiketkukade, 2022-08-06:
That IDF version is probably showing a private commit, can you please share the upstream commit from where you started adding your changes?
Also there is one recent change (commit id https://github.com/espressif/esp-idf/commit/2f4ec4e27d6f6839e8e072dd09fc2b05b831e226 ) that has fixes related to this feature.  If this isn't present in your repo, I'd suggest to rebase on top of this commit and check if the issue still exists.

0xFEEDC0DE64, 2022-08-08:
we have had lots of issues with tcp with newer idf versions, for example websocket clients did not reconnect anymore, i will try to cherry-pick that new wifi blob into our codebase, thanks for finding it :)

closed 2022-09-09 by nachiketkukade