espressif/esp-idf #9488: Blufi Check Sum Error in IDF 5.0

Nick-tecpal, 2022-08-03:
## Environment

- Development Kit: ESP32-S3-DevKitC-1
- Kit version:v1
- Module or chip used:  ESP32-S3
- IDF version :master/v5.0-dev-4547-gfde4afc67a
- Build System:         [Make|CMake|idf.py]
- Compiler version:xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:Linux
- Using an IDE?:No
- Power Supply: USB

## Problem Description

We’re using the IDF4.4, and we plan to upgrade to 5.0. But we get a check sum error when use the Blufi sample to test the IDF 5.0

https://github.com/espressif/esp-idf/tree/release/v5.0/examples/bluetooth/blufi

### Expected Behavior

Pass the test with Blufi sample

### Actual Behavior

Check sum error with Blufi sample

### Steps to reproduce

Download and run the Blufi sample.

## Debug Logs
[image]

## Other items if possible

esp-cjh, 2022-08-10:
Hi, @Nick-tecpal 
In IDF 5.0, we enable hardware AES acceleration(CONFIG_MBEDTLS_HARDWARE_AES=y) which caused error in blufi ase decrypt. You can disable hardware AES acceleration temporarily by menuconfig. We'll fix this asap.

esp-cjh, 2022-08-15:
Hi, @Nick-tecpal. 
We have found the reason of this problem. API usage for `mbedtls_dhm_calc_secret` was incorrect in IDF 5.0. We will fix it ASAP.

closed 2022-08-17 by espressif-bot