espressif/esp-idf #9489: Memory allocation Failure after looping through ota check too many times

tobyloki, 2022-08-03:
I am trying to run a variant of the pre_encrypted_ota sample from https://github.com/espressif/esp-idf/tree/release/v5.0/examples/system/ota/pre_encrypted_ota. My code basically just runs the main code infinitely so that it always checks for new ota updates. If any step fails, it restarts the loop, attempts to clean up some memory, waits 2s, and restarts. The code is shown below. However, after several loops, eventually the memory starts to run out and it gives me this error. How can I fix this?

```
E (413035) Dynamic Impl: alloc(16749 bytes) failed
E (413035) esp-tls-mbedtls: read error :-0x7F00:
E (413045) TRANSPORT_BASE: esp_tls_conn_read error, errno=Success
E (413045) esp_https_ota: Connection closed, errno = 0
E (413055) esp_https_ota: Complete headers were not received
```

```c
bool broken = false;
esp_decrypt_handle_t decrypt_handle = NULL;
esp_https_ota_handle_t https_ota_handle = NULL;

while (1)
{
    if (broken)
    {
        if (decrypt_handle != NULL)
        {
            esp_encrypted_img_decrypt_end(decrypt_handle);
            decrypt_handle = NULL;
        }

        if (https_ota_handle != NULL)
        {
            esp_https_ota_abort(https_ota_handle);
            https_ota_handle = NULL;
        }

        broken = false;
        vTaskDelay(2000 / portTICK_PERIOD_MS);
    }

    esp_err_t ota_finish_err = ESP_OK;
    esp_http_client_config_t config = {
        .url = "https://someurl/program_secure.bin",
        .cert_pem = server_cert_pem_start,
        .timeout_ms = 5000,
        .keep_alive_enable = true,
    };

    esp_decrypt_cfg_t cfg = {};
    cfg.rsa_pub_key = otaKey;
    cfg.rsa_pub_key_len = strlen(otaKey) + 1;
    // cfg.rsa_pub_key = rsa_private_pem_start;
    // cfg.rsa_pub_key_len = rsa_private_pem_end - rsa_private_pem_start;
    decrypt_handle = esp_encrypted_img_decrypt_start(&cfg);
    if (!decrypt_handle)
    {
        ESP_LOGE(TAG, "OTA upgrade failed");

        broken = true;
        continue;
    }

    esp_https_ota_config_t ota_config = {
        .http_config = &config,
        // .partial_http_download = true,
        // .max_http_request_size = 300000,
        .decrypt_cb = _decrypt_cb,
        .decrypt_user_ctx = (void *)decrypt_handle,
    };

    esp_err_t err = esp_https_ota_begin(&ota_config, &https_ota_handle);
    if (err != ESP_OK)
    {
        ESP_LOGE(TAG, "ESP HTTPS OTA Begin failed");

[... 147 lines cut ...]
E (402475) Dynamic Impl: alloc(16322 bytes) failed
E (402475) esp-tls-mbedtls: read error :-0x7F00:
E (402485) TRANSPORT_BASE: esp_tls_conn_read error, errno=Success
E (402485) esp_https_ota: Connection closed, errno = 0
E (402495) esp_https_ota: Complete headers were not received
I (404505) esp_encrypted_img: Starting Decryption Process
I (406575) esp_https_ota: Starting OTA...
I (406575) esp_https_ota: Writing to partition subtype 17 at offset 0x1b0000
I (406585) esp_encrypted_img: Magic Verified
I (406585) esp_encrypted_img: Reading RSA private key
I (408795) native_ota_example: Running firmware version: 1.0.1
W (408805) native_ota_example: New firmware version 1.0.1 is older than or equal to the running firmware version 1.0.1. Will not begin ota.
E (408805) esp_https_ota: Decrypt callback failed -1
E (408815) esp_https_ota: Decryption of image header failed
E (408825) esp_encrypted_img: Invalid operation
I (410825) esp_encrypted_img: Starting Decryption Process
I (413025) esp_https_ota: Starting OTA...
I (413025) esp_https_ota: Writing to partition subtype 17 at offset 0x1b0000
E (413035) Dynamic Impl: alloc(16749 bytes) failed
E (413035) esp-tls-mbedtls: read error :-0x7F00:
E (413045) TRANSPORT_BASE: esp_tls_conn_read error, errno=Success
E (413045) esp_https_ota: Connection closed, errno = 0
E (413055) esp_https_ota: Complete headers were not received
I (415065) esp_encrypted_img: Starting Decryption Process
```

hmalpani, 2022-08-04:
Hello @tobyloki 
Thanks for reporting the issue. This is being caused because memory is not freed in [`_decrypt_cb()`](https://github.com/espressif/esp-idf/blob/master/examples/system/ota/pre_encrypted_ota/main/pre_encrypted_ota.c#L63) function in the pre_encrypted_ota example.
I am sharing a patch here. Can you try the example with it and confirm if it solves the issue?
[IDFGH_7978.txt](https://github.com/espressif/esp-idf/files/9256925/IDFGH_7978.txt)

tobyloki, 2022-08-04:
Thanks. That seemed to fix the problem. I do have one question though. Even if `err` is equal to `ESP_OK`, is it still ok to make a call to `free(pargs.data_out);`?

hmalpani, 2022-08-05:
No. When `err` is equal to `ESP_OK`, `pargs.data_out` is freed after writing into the partition. If we try to free it, it will give error.

mahavirj, 2022-08-05:
@tobyloki Reopening the issue. This shall be (auto) closed when fix gets merged.

closed 2022-08-09 by espressif-bot