espressif/esp-idf #9490: BLE Mesh Generic on/off client server example: Invalid Response

Shane555, 2022-08-03:
## Environment
- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for DevKitC): |v3|
- IDF version: v4.3.3 release
- Build System:         [idf.py]
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp-2022r1-RC1) 11.2.0
- Operating System:     [Linux]
- Using an IDE?: [VSCode]
- Power Supply:         [USB]

## Problem Description

I ran the generic on/off client and server example with 2 esp32c3m1 devkit. One acting as the client, the other acting as a server. I have not provisioned the two nodes into a network yet at this point. When i tracked the output from idf.py monitor, the callbacks are not running as intended. The event `ESP_BLE_MESH_GENERIC_CLIENT_SET_STATE_EVT` keeps running when i send a set state message to nowhere with `esp_ble_mesh_generic_client_set_state(&common, &set);` which is odd because in the [tutorial](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/esp_ble_mesh/ble_mesh_node/onoff_client/tutorial/BLE_Mesh_Node_OnOff_Client_Example_Walkthrough.md), the event should only trigger when it receives an ACK from the server model yet it keeps triggering even without connecting to any other nodes. 
 
### Steps to reproduce

1. run the espblemesh example for esp_ble_node and flash the devkit via usb. Power it using the usb as well
2. run idf.py monitor for the port

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
void app_main(void)
{
  esp_err_t err;

  ESP_LOGI(TAG, "Initializing...");

  board_init();

  err = nvs_flash_init();
  if (err == ESP_ERR_NVS_NO_FREE_PAGES)
  {
    ESP_ERROR_CHECK(nvs_flash_erase());
    err = nvs_flash_init();
  }
  ESP_ERROR_CHECK(err);

  err = bluetooth_init();
  if (err)
  {
    ESP_LOGE(TAG, "esp32_bluetooth_init failed (err %d)", err);
    return;
  }

  /* Open nvs namespace for storing/restoring mesh example info */
  err = ble_mesh_nvs_open(&NVS_HANDLE);
  if (err)
  {
    return;
  }

  ble_mesh_get_dev_uuid(dev_uuid);

  /* Initialize the Bluetooth Mesh Subsystem */
  err = ble_mesh_init();
  if (err)
  {
    ESP_LOGE(TAG, "Bluetooth mesh init failed (err %d)", err);
  }
  while (1)
  {
    example_ble_mesh_send_gen_onoff_set(); //added this line
    vTaskDelay(5000 / portTICK_PERIOD_MS); //added this line
  }
  return;
}

```

## Debug Logs

output looping from monitor
```
I (292476) EXAMPLE_NVS: Store, key "onoff_client", length 6
I (292476) EXAMPLE_NVS: Store, data: 00 00 00 00 01 71 
E (297476) BLE_MESH: Invalid NetKeyIndex 0x0000
E (297476) BLE_MESH: Failed to send client message 0x00008202
I (297476) EXAMPLE: Generic client, event 1, error code -22, opcode is 0x8202
I (297476) EXAMPLE: ESP_BLE_MESH_GENERIC_CLIENT_SET_STATE_EVT
I (297486) EXAMPLE: ESP_BLE_MESH_MODEL_OP_GEN_ONOFF_SET, onoff 0
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

Shane555, 2022-08-03:
I suspect that the description may be wrong. The client node seems to be triggering the ESP_BLE_MESH_GENERIC_CLIENT_PUBLISH_EVT only when the server node sends back a message from client node request. In that case, the description in the [tutorial](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/esp_ble_mesh/ble_mesh_node/onoff_client/tutorial/BLE_Mesh_Node_OnOff_Client_Example_Walkthrough.md) may not be correct. 
[image]

forx157, 2022-08-26:
Hi, @Shane555 
In your first question, the ESP_BLE_MESH_GENERIC_CLIENT_SET_STATE_EVT event is indeed as you said, not only when the reply is replied to, but also when the reply fails, as can be referred to in the following code：`example_ble_mesh_send_gen_onoff_set ` => ` esp_ble_mesh_generic_client_set_state` => ` btc_ble_mesh_generic_client_call_handler` => 
```c
        cb.error_code = bt_mesh_generic_client_set_state(&common, arg->generic_client_set_state.set_state);
        if (cb.error_code) {
            /* If send failed, callback error_code to app layer immediately */
            btc_ble_mesh_generic_client_callback(&cb, ESP_BLE_MESH_GENERIC_CLIENT_SET_STATE_EVT);
        }
```

hipal31, 2022-08-26:
@Shane555 
callback indicates you about error.
So in callback, you should consider only if error code is 0.

Shane555, 2022-08-29:
Ok, that clears things up. Its a misunderstanding on my end. Thanks.

closed 2022-08-29 by Shane555