espressif/esp-idf #9492: ECONNABORTED periodically on MQTT nodes

KonssnoK, 2022-08-03:
Using esp32s3, mesh-networking, MQTT, 11 devices.

Devices form a network and connect individually to a MQTT Broker.

Periodically, once per hour, i see this on the terminal:

Looking at the log, i would say this happens always after an NTP synchronization, but it shouldn't influence the MQTT connection

```
I (12:57:18.901) mqtt_app: MQTT_EVENT_PUBLISHED, msg_id=53052
E (12:57:25.566) TRANSPORT_BASE: poll_read select error 113, errno = Software caused connection abort, fd = 54
E (12:57:25.567) MQTT_CLIENT: Poll read error: 119, aborting connection
I (12:57:25.575) mqtt_app: MQTT_EVENT_DISCONNECTED
I (12:57:26.536) network: Network connected
I (12:57:26.536) network: IP: 10.0.0.99
I (12:57:26.537) network: Netmask: 255.255.0.0
I (12:57:26.539) network: Gateway: 10.0.0.1
W (12:57:26.544) mqtt_app: MQTT Client is already running!
I (12:57:35.580) mqtt_app: MQTT_EVENT_BEFORE_CONNECT
I (12:57:36.845) mqtt_app: MQTT_EVENT_CONNECTED
```

```
I (11:56:48.209) mqtt_app: sent publish returned msg_id=0
W (11:56:55.715) ntp: state change: [2 (ESP), 1]
I (11:56:55.716) processor: NTP event: 1
W (11:56:58.762) ntp: state change: [1 (ESP), 2]
I (11:57:00.276) mqtt_app: sent publish returned msg_id=48805
I (11:57:00.309) mqtt_app: MQTT_EVENT_PUBLISHED, msg_id=48805
I (11:57:01.775) mqtt_app: sent publish returned msg_id=18438
I (11:57:01.845) mqtt_app: MQTT_EVENT_PUBLISHED, msg_id=18438
E (11:57:23.371) TRANSPORT_BASE: poll_read select error 113, errno = Software caused connection abort, fd = 54
E (11:57:23.372) MQTT_CLIENT: Poll read error: 119, aborting connection
I (11:57:23.380) mqtt_app: MQTT_EVENT_DISCONNECTED
I (11:57:24.330) network: Network connected
I (11:57:24.330) network: IP: 10.0.0.74
I (11:57:24.330) network: Netmask: 255.255.0.0
I (11:57:24.334) network: Gateway: 10.0.0.1
W (11:57:24.338) mqtt_app: MQTT Client is already running!
I (11:57:33.384) mqtt_app: MQTT_EVENT_BEFORE_CONNECT
I (11:57:34.701) mqtt_app: MQTT_EVENT_CONNECTED

```

euripedesrocha, 2022-08-04:
Hi @KonssnoK, could you provide the version of idf and esp-mqtt you are using? 
We recently fixed an issue espressif/esp-mqtt#225 that had similar behavior.

KonssnoK, 2022-08-04:
@euripedesrocha Sorry my bad for forgetting

ESP-IDF v4.4
f0a3ccc73201c2d94e135daaa36973f2bd389e3b

which picks from
27eb4726067465c5c67d4ecdca5ddccd26f02580
of esp-mqtt repo

KonssnoK, 2022-08-04:
The fix you are referring to is not backported to v4.X branch

AxelLin, 2022-10-27:
> The fix you are referring to is not backported to v4.X branch

https://github.com/espressif/esp-mqtt/commit/166741fd6d68fa7b2b8ebadc4cdb352761e5ee9c

AxelLin, 2022-11-11:
> The fix you are referring to is not backported to v4.X branch

@euripedesrocha 

It really takes too much time waitting for backport fix to 4.x branches.
There is a big time gap between a fix available in master (https://github.com/espressif/esp-mqtt/commit/89e5c6014f8dbcfcd98af35fb507ca7b96ac8aee Jul 5) v.s. fix available in idf_v4.x branch (https://github.com/espressif/esp-mqtt/commit/166741fd6d68fa7b2b8ebadc4cdb352761e5ee9c Oct 26).
And another big time gap to update esp-idf submodule. (2 week+)
I understand you need to arrange test to avoid regressions, but such long backport fix time does not make sense at all.
Any chance to improve the backport bug fix process?

euripedesrocha, 2022-11-11:
Hi @AxelLin, we are aware of the long time it takes, and we are already working to improve those times.

AxelLin, 2022-11-30:
I'm still waiting for the esp-idf submodule update for v4.3 branch.

euripedesrocha, 2022-12-01:
Hi @AxelLin, I'm sorry for the long waiting. We are working to merge it soon. This particular one is a bit tricky because, despite being simple, we have a change in the behavior of the component.

AxelLin, 2022-12-08:
> Hi @AxelLin, I'm sorry for the long waiting. We are working to merge it soon. This particular one is a bit tricky because, despite being simple, we have a change in the behavior of the component.

The esp-mqtt idf_v4.x branch is supposed to be used for v4.x branches.
So that's the difference of backporting to v4.4 and v4.3 branch?

euripedesrocha, 2022-12-08:
Hi @AxelLin, idf_v4x is used for v4.x branches, but you can see that in IDF we point to different commits on that branch. IDF release branches are sometimes locked, and we need to wait for the merging window. In this particular case, there are changes in the MR that are being evaluated to back port because 4.3 and 4.4 are in different stages of their lifetime support. We are working to make it easier to have a particular mqtt version available for usage in a project using the component manager. 

A workaround, to not be blocked by our process, is to have a separated mqtt component in your project pointing to a convenient commit and append it to `EXTRA_COMPONENT_DIRS` to be selected instead of idf's internal version.

closed 2022-12-22 by KonssnoK