espressif/esp-idf #9493: memspi: no response!, when using w25q128 for external spi flash on HSPI

muratdemirtas, 2022-08-03:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No|Yes (please give details)]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to reproduce

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

 `[00:16:53:888] <0x1b>[0;32mI (00:00:02.082) Factory JAN V2.3: MOSI: 13   MISO: 12   SCLK: 14   CS:  0<0x1b>[0m␍␊
[00:16:53:888] <0x1b>[0;32mI (00:00:02.084) Factory JAN V2.3:  SPI FLASH 1 <0x1b>[0m␍␊
[00:16:53:888] D (00:00:02.090) spi: SPI2 use iomux pins.<0x1b>[0m␍␊
[00:16:53:888] <0x1b>[0;32mI (00:00:02.094) Factory JAN V2.3:  SPI FLASH 2<0x1b>[0m␍␊
[00:16:53:888] V (00:00:02.099) bus_lock: device registered on bus 1 slot 0.<0x1b>[0m␍␊
[00:16:53:931] D (00:00:02.105) FLASH_HAL: extra_dummy: 0<0x1b>[0m␍␊
[00:16:53:931] V bus_lock: dev 0 acquired.␍␊
[00:16:53:931] V bus_lock: SPI dev changed from -1 to 0␍␊
[00:16:53:931] V (00:00:02.115) bus_lock: dev 0 released.<0x1b>[0m␍␊
[00:16:53:931] <0x1b>[0;32mI (00:00:02.119) Factory JAN V2.3:  SPI FLASH 3<0x1b>[0m␍␊
[00:16:53:931] V bus_lock: dev 0 acquired.␍␊
[00:16:53:931] V (2939) memspi: raw_chip_id: 1840EF␍␊
[00:16:53:931] <0x1b>[0m␍␊
[00:16:53:931] V (2939) memspi: chip_id: EF4018␍␊
[00:16:53:931] <0x1b>[0m␍␊
[00:16:53:931] V (2939) memspi: raw_chip_id: 0␍␊
[00:16:53:931] <0x1b>[0m␍␊
[00:16:53:931] <0x1b>[0;31mE (2949) memspi: no response␍␊
[00:16:53:931] <0x1b>[0m␍␊
[00:16:53:931] V (00:00:02.141) bus_lock: dev 0 released.<0x1b>[0m␍␊
[00:16:53:974] <0x1b>[0;31mE (00:00:02.146) Factory JAN V2.3: FLASH error: UNKNOWN ERROR (0x108)<0x1b>[0m␍␊`

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

nopnop2002, 2022-08-04:
Plz try this.

https://github.com/nopnop2002/esp-idf-w25q64

muratdemirtas, 2022-08-04:
> Plz try this.
> 
> https://github.com/nopnop2002/esp-idf-w25q64 [image]

[11:01:58:500] <0x1b>[0;32mI (675) W25Q64: Page Write(Sector=0 Address=10) len=26<0x1b>[0m␍␊
[11:01:58:500] <0x1b>[0;32mI (685) W25Q64: Fast Read Data: len=256<0x1b>[0m␍␊
[11:01:58:500] ----------------------------------------------------------␍␊
[11:01:58:500] 00000: ff ff ff ff ff ff ff ff ff ff 41 42 43 44 45 46 |8b ␍␊
[11:01:58:500] 00010: 47 48 49 4a 4b 4c 4d 4e 4f 50 51 52 53 54 55 56 |e8 ␍␊
[11:01:58:544] 00020: 57 58 59 5a ff ff ff ff ff ff ff ff ff ff ff ff |56 ␍␊
[11:01:58:544] 00030: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:544] 00040: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:544] 00050: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:544] 00060: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:544] 00070: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:544] 00080: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:544] 00090: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:587] 000a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:587] 000b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:587] 000c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:587] 000d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:587] 000e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:587] 000f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:587] ----------------------------------------------------------␍␊
[11:01:58:587]        90 92 94 96 3c 3d 3e 3f 40 41 84 86 88 8a 8c 8e |f9 ␍␊
[11:01:58:630] ␍␊
[11:01:58:630] <0x1b>[0;32mI (785) W25Q64: Page Write(Sector=0 Address=0) len=10<0x1b>[0m␍␊
[11:01:58:630] <0x1b>[0;32mI (795) W25Q64: Fast Read Data: len=256<0x1b>[0m␍␊
[11:01:58:630] ----------------------------------------------------------␍␊
[11:01:58:630] 00000: 30 31 32 33 34 35 36 37 38 39 41 42 43 44 45 46 |a2 ␍␊
[11:01:58:630] 00010: 47 48 49 4a 4b 4c 4d 4e 4f 50 51 52 53 54 55 56 |e8 ␍␊
[11:01:58:630] 00020: 57 58 59 5a ff ff ff ff ff ff ff ff ff ff ff ff |56 ␍␊
[11:01:58:630] 00030: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:630] 00040: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:630] 00050: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:673] 00060: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:673] 00070: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:673] 00080: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:673] 00090: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:673] 000a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:673] 000b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:673] 000c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:673] 000d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:715] 000e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:715] 000f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:715] ----------------------------------------------------------␍␊
[11:01:58:715]        c1 c4 c7 ca 71 73 75 77 79 7b 84 86 88 8a 8c 8e |10 ␍␊
[11:01:58:715] ␍␊
[11:01:58:715] <0x1b>[0;32mI (895) W25Q64: Success All Test<0x1b>[0m␍␊

muratdemirtas, 2022-08-04:
> Plz try this.
> 
> https://github.com/nopnop2002/esp-idf-w25q64 [image]

`[11:01:58:153] <0x1b>[0;32mI (326) cpu_start: Starting scheduler on PRO CPU.<0x1b>[0m␍␊
[11:01:58:153] <0x1b>[0;32mI (0) cpu_start: Starting scheduler on APP CPU.<0x1b>[0m␍␊
[11:01:58:153] <0x1b>[0;32mI (335) W25Q64: MISO_GPIO=12<0x1b>[0m␍␊
[11:01:58:153] <0x1b>[0;32mI (335) W25Q64: MOSI_GPIO=13<0x1b>[0m␍␊
[11:01:58:153] <0x1b>[0;32mI (345) W25Q64: SCLK_GPIO=14<0x1b>[0m␍␊
[11:01:58:153] <0x1b>[0;32mI (345) W25Q64: CS_GPIO=0<0x1b>[0m␍␊
[11:01:58:153] <0x1b>[0;32mI (345) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 <0x1b>[0m␍␊
[11:01:58:196] <0x1b>[0;32mI (355) W25Q64: readStatusReg1 : 0<0x1b>[0m␍␊
[11:01:58:196] <0x1b>[0;32mI (365) W25Q64: readStatusReg2 : 0<0x1b>[0m␍␊
[11:01:58:196] <0x1b>[0;32mI (365) W25Q64: readUniqieID : d2-66-78-64-43-43-18-29<0x1b>[0m␍␊
[11:01:58:196] <0x1b>[0;32mI (375) W25Q64: readManufacturer : ef-40-18<0x1b>[0m␍␊
[11:01:58:196] <0x1b>[0;32mI (375) W25Q64: Fast Read Data: len=256<0x1b>[0m␍␊
[11:01:58:196] ----------------------------------------------------------␍␊
[11:01:58:196] 00000: 30 31 32 33 34 35 36 37 38 39 41 42 43 44 45 46 |a2 ␍␊
[11:01:58:196] 00010: 47 48 49 4a 4b 4c 4d 4e 4f 50 51 52 53 54 55 56 |e8 ␍␊
[11:01:58:239] 00020: 57 58 59 5a ff ff ff ff ff ff ff ff ff ff ff ff |56 ␍␊
[11:01:58:239] 00030: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:239] 00040: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:239] 00050: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:239] 00060: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:239] 00070: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:239] 00080: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:239] 00090: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:282] 000a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:282] 000b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:282] 000c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:282] 000d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:282] 000e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:282] 000f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:282] ----------------------------------------------------------␍␊
[11:01:58:282]        c1 c4 c7 ca 71 73 75 77 79 7b 84 86 88 8a 8c 8e |10 ␍␊
[11:01:58:298] ␍␊
[11:01:58:414] <0x1b>[0;32mI (585) W25Q64: Read Data: len=256<0x1b>[0m␍␊
[11:01:58:414] ----------------------------------------------------------␍␊
[11:01:58:414] 00000: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:414] 00010: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:414] 00020: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:414] 00030: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:414] 00040: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:414] 00050: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:414] 00060: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:457] 00070: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:457] 00080: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:457] 00090: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:457] 000a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:457] 000b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:457] 000c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:457] 000d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:457] 000e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:500] 000f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:500] ----------------------------------------------------------␍␊
[11:01:58:500]        f0 f0 f0 f0 f0 f0 f0 f0 f0 f0 f0 f0 f0 f0 f0 f0 |00 ␍␊
[11:01:58:500] ␍␊
[11:01:58:500] <0x1b>[0;32mI (675) W25Q64: Page Write(Sector=0 Address=10) len=26<0x1b>[0m␍␊
[11:01:58:500] <0x1b>[0;32mI (685) W25Q64: Fast Read Data: len=256<0x1b>[0m␍␊
[11:01:58:500] ----------------------------------------------------------␍␊
[11:01:58:500] 00000: ff ff ff ff ff ff ff ff ff ff 41 42 43 44 45 46 |8b ␍␊
[11:01:58:500] 00010: 47 48 49 4a 4b 4c 4d 4e 4f 50 51 52 53 54 55 56 |e8 ␍␊
[11:01:58:544] 00020: 57 58 59 5a ff ff ff ff ff ff ff ff ff ff ff ff |56 ␍␊
[11:01:58:544] 00030: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:544] 00040: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:544] 00050: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:544] 00060: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:544] 00070: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:544] 00080: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:544] 00090: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[... 7 lines cut ...]
[11:01:58:587]        90 92 94 96 3c 3d 3e 3f 40 41 84 86 88 8a 8c 8e |f9 ␍␊
[11:01:58:630] ␍␊
[11:01:58:630] <0x1b>[0;32mI (785) W25Q64: Page Write(Sector=0 Address=0) len=10<0x1b>[0m␍␊
[11:01:58:630] <0x1b>[0;32mI (795) W25Q64: Fast Read Data: len=256<0x1b>[0m␍␊
[11:01:58:630] ----------------------------------------------------------␍␊
[... cut ...]
[11:01:58:673] 000c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:673] 000d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:715] 000e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:715] 000f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:715] ----------------------------------------------------------␍␊
[11:01:58:715]        c1 c4 c7 ca 71 73 75 77 79 7b 84 86 88 8a 8c 8e |10 ␍␊
[11:01:58:715] ␍␊
[11:01:58:715] <0x1b>[0;32mI (895) W25Q64: Success All Test<0x1b>[0m␍␊`

muratdemirtas, 2022-08-04:
> Plz try this.
> 
> https://github.com/nopnop2002/esp-idf-w25q64 [image]

[11:01:58:153] <0x1b>[0;32mI (326) cpu_start: Starting scheduler on PRO CPU.<0x1b>[0m␍␊
[11:01:58:153] <0x1b>[0;32mI (0) cpu_start: Starting scheduler on APP CPU.<0x1b>[0m␍␊
[11:01:58:153] <0x1b>[0;32mI (335) W25Q64: MISO_GPIO=12<0x1b>[0m␍␊
[11:01:58:153] <0x1b>[0;32mI (335) W25Q64: MOSI_GPIO=13<0x1b>[0m␍␊
[11:01:58:153] <0x1b>[0;32mI (345) W25Q64: SCLK_GPIO=14<0x1b>[0m␍␊
[11:01:58:153] <0x1b>[0;32mI (345) W25Q64: CS_GPIO=0<0x1b>[0m␍␊
[11:01:58:153] <0x1b>[0;32mI (345) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 <0x1b>[0m␍␊
[11:01:58:196] <0x1b>[0;32mI (355) W25Q64: readStatusReg1 : 0<0x1b>[0m␍␊
[11:01:58:196] <0x1b>[0;32mI (365) W25Q64: readStatusReg2 : 0<0x1b>[0m␍␊
[11:01:58:196] <0x1b>[0;32mI (365) W25Q64: readUniqieID : d2-66-78-64-43-43-18-29<0x1b>[0m␍␊
[11:01:58:196] <0x1b>[0;32mI (375) W25Q64: readManufacturer : ef-40-18<0x1b>[0m␍␊
[11:01:58:196] <0x1b>[0;32mI (375) W25Q64: Fast Read Data: len=256<0x1b>[0m␍␊
[11:01:58:196] ----------------------------------------------------------␍␊
[11:01:58:196] 00000: 30 31 32 33 34 35 36 37 38 39 41 42 43 44 45 46 |a2 ␍␊
[11:01:58:196] 00010: 47 48 49 4a 4b 4c 4d 4e 4f 50 51 52 53 54 55 56 |e8 ␍␊
[11:01:58:239] 00020: 57 58 59 5a ff ff ff ff ff ff ff ff ff ff ff ff |56 ␍␊
[11:01:58:239] 00030: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:239] 00040: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:239] 00050: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:239] 00060: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:239] 00070: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:239] 00080: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:239] 00090: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:282] 000a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:282] 000b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:282] 000c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:282] 000d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:282] 000e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:282] 000f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:282] ----------------------------------------------------------␍␊
[11:01:58:282] c1 c4 c7 ca 71 73 75 77 79 7b 84 86 88 8a 8c 8e |10 ␍␊
[11:01:58:298] ␍␊
[11:01:58:414] <0x1b>[0;32mI (585) W25Q64: Read Data: len=256<0x1b>[0m␍␊
[11:01:58:414] ----------------------------------------------------------␍␊
[11:01:58:414] 00000: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:414] 00010: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:414] 00020: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:414] 00030: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:414] 00040: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:414] 00050: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:414] 00060: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:457] 00070: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:457] 00080: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:457] 00090: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:457] 000a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:457] 000b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:457] 000c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:457] 000d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:457] 000e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:500] 000f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:500] ----------------------------------------------------------␍␊
[11:01:58:500] f0 f0 f0 f0 f0 f0 f0 f0 f0 f0 f0 f0 f0 f0 f0 f0 |00 ␍␊
[11:01:58:500] ␍␊
[11:01:58:500] <0x1b>[0;32mI (675) W25Q64: Page Write(Sector=0 Address=10) len=26<0x1b>[0m␍␊
[11:01:58:500] <0x1b>[0;32mI (685) W25Q64: Fast Read Data: len=256<0x1b>[0m␍␊
[11:01:58:500] ----------------------------------------------------------␍␊
[11:01:58:500] 00000: ff ff ff ff ff ff ff ff ff ff 41 42 43 44 45 46 |8b ␍␊
[11:01:58:500] 00010: 47 48 49 4a 4b 4c 4d 4e 4f 50 51 52 53 54 55 56 |e8 ␍␊
[11:01:58:544] 00020: 57 58 59 5a ff ff ff ff ff ff ff ff ff ff ff ff |56 ␍␊
[11:01:58:544] 00030: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:544] 00040: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:544] 00050: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:544] 00060: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:544] 00070: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:544] 00080: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:544] 00090: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[... 7 lines cut ...]
[11:01:58:587] 90 92 94 96 3c 3d 3e 3f 40 41 84 86 88 8a 8c 8e |f9 ␍␊
[11:01:58:630] ␍␊
[11:01:58:630] <0x1b>[0;32mI (785) W25Q64: Page Write(Sector=0 Address=0) len=10<0x1b>[0m␍␊
[11:01:58:630] <0x1b>[0;32mI (795) W25Q64: Fast Read Data: len=256<0x1b>[0m␍␊
[11:01:58:630] ----------------------------------------------------------␍␊
[... cut ...]
[11:01:58:673] 000c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:673] 000d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:715] 000e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:715] 000f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff |f0 ␍␊
[11:01:58:715] ----------------------------------------------------------␍␊
[11:01:58:715] c1 c4 c7 ca 71 73 75 77 79 7b 84 86 88 8a 8c 8e |10 ␍␊
[11:01:58:715] ␍␊
[11:01:58:715] <0x1b>[0;32mI (895) W25Q64: Success All Test<0x1b>[0m␍␊

muratdemirtas, 2022-08-04:
> Plz try this.
> 
> https://github.com/nopnop2002/esp-idf-w25q64 [image]

i have tried with different frequencies between 5-40Mhz, Looks like your code is working and tests passing without errors.

But esp-idf doesnt detect my chip.

//static const int SPI_Frequency = 5000000;
static const int SPI_Frequency = 40000000;

nopnop2002, 2022-08-04:
It may not matter, but Your code is not set  cs_io_num:
```
const esp_flash_spi_device_config_t device_config = {
		.host_id = HSPI_HOST,
		.cs_io_num = 0,
		.io_mode = SPI_FLASH_FASTRD,
		.speed = ESP_FLASH_40MHZ,
		.cs_id = 0,
};
```

muratdemirtas, 2022-08-04:
> There is official example: https://github.com/espressif/esp-idf/tree/master/examples/storage/ext_flash_fatfs
> 
> ```
>   const esp_flash_spi_device_config_t device_config = {
>       .host_id = HOST_ID,
>       .cs_id = 0,
>       .cs_io_num = PIN_CS,
>       .io_mode = SPI_FLASH_DIO,
>       .freq_mhz = EXAMPLE_FLASH_FREQ_MHZ,
>   };
> ```
> 
> Your code is different from the official example.
> 
> ```
> const esp_flash_spi_device_config_t device_config = {
> 		.host_id = HSPI_HOST,
> 		.cs_io_num = 0,
> 		.io_mode = SPI_FLASH_FASTRD,
> 		.speed = ESP_FLASH_40MHZ,
> 		.cs_id = 0,
> };
> ```
> 
> Plz try official: https://github.com/espressif/esp-idf/tree/master/examples/storage/ext_flash_fatfs

We are using esp idf 4.4. I'm using the official example. 
https://github.com/espressif/esp-idf/blob/release/v4.4/examples/storage/ext_flash_fatfs/main/ext_flash_fatfs_example_main.c

Result is same.

muratdemirtas, 2022-08-04:
I tried with another spi flash 4MB w25q32 but it gives me

[01:42:39:190] <0x1b>[0;32mI (1292) example: Initializing external SPI Flash<0x1b>[0m␍␊
[01:42:39:190] <0x1b>[0;32mI (1292) example: Pin assignments:<0x1b>[0m␍␊
[01:42:39:190] <0x1b>[0;32mI (1302) example: MOSI: 13   MISO: 12   SCLK: 14   CS: 20<0x1b>[0m␍␊
[01:42:39:190] D (1302) spi: SPI2 use iomux pins.<0x1b>[0m␍␊
[01:42:39:190] V (1312) bus_lock: device registered on bus 1 slot 0.<0x1b>[0m␍␊
[01:42:39:190] D (1312) FLASH_HAL: extra_dummy: 0<0x1b>[0m␍␊
[01:42:39:190] V bus_lock: dev 0 acquired.␍␊
[01:42:39:190] V bus_lock: SPI dev changed from -1 to 0␍␊
[01:42:39:190] V (1322) bus_lock: dev 0 released.<0x1b>[0m␍␊
[01:42:39:233] V bus_lock: dev 0 acquired.␍␊
[01:42:39:233] V (1332) memspi: raw_chip_id: 16405E␍␊
[01:42:39:233] <0x1b>[0m␍␊
[01:42:39:233] V (1332) memspi: chip_id: 5E4016␍␊
[01:42:39:233] <0x1b>[0m␍␊
[01:42:39:233] V (1342) memspi: raw_chip_id: 5E1640␍␊
[01:42:39:233] <0x1b>[0m␍␊
[01:42:39:233] V (1342) memspi: chip_id: 40165E␍␊
[01:42:39:233] <0x1b>[0m␍␊
[01:42:39:233] V (1342) bus_lock: dev 0 released.<0x1b>[0m␍␊
[01:42:39:233] V bus_lock: dev 0 acquired.␍␊
[01:42:39:233] V (1352) memspi: raw_chip_id: 405E16␍␊
[01:42:39:233] <0x1b>[0m␍␊
[01:42:39:233] V (1352) memspi: chip_id: 165E40␍␊
[01:42:39:233] <0x1b>[0m␍␊
[01:42:39:233] V (1362) memspi: raw_chip_id: 16405E␍␊
[01:42:39:233] <0x1b>[0m␍␊
[01:42:39:233] V (1362) memspi: chip_id: 5E4016␍␊
[01:42:39:233] <0x1b>[0m␍␊
[01:42:39:233] V (1362) bus_lock: dev 0 released.<0x1b>[0m␍␊
[01:42:39:233] V bus_lock: dev 0 acquired.␍␊
[01:42:39:276] V (1372) memspi: raw_chip_id: 5E1640␍␊
[01:42:39:276] <0x1b>[0m␍␊
[01:42:39:276] V (1372) memspi: chip_id: 40165E␍␊
[01:42:39:276] <0x1b>[0m␍␊
[01:42:39:276] V (1382) memspi: raw_chip_id: 405E16␍␊
[01:42:39:276] <0x1b>[0m␍␊
[01:42:39:276] V (1382) memspi: chip_id: 165E40␍␊
[01:42:39:276] <0x1b>[0m␍␊
[01:42:39:276] V (1382) bus_lock: dev 0 released.<0x1b>[0m␍␊
[01:42:39:276] V bus_lock: dev 0 acquired.␍␊
[01:42:39:276] V (1392) memspi: raw_chip_id: 16405E␍␊
[01:42:39:276] <0x1b>[0m␍␊
[01:42:39:276] V (1392) memspi: chip_id: 5E4016␍␊
[01:42:39:276] <0x1b>[0m␍␊
[01:42:39:276] V (1402) memspi: raw_chip_id: 5E1640␍␊
[01:42:39:276] <0x1b>[0m␍␊
[01:42:39:276] V (1402) memspi: chip_id: 40165E␍␊
[01:42:39:276] <0x1b>[0m␍␊
[01:42:39:276] V (1402) bus_lock: dev 0 released.<0x1b>[0m␍␊
[01:42:39:276] V bus_lock: dev 0 acquired.␍␊
[01:42:39:319] V (1412) memspi: raw_chip_id: 405E16␍␊
[01:42:39:319] <0x1b>[0m␍␊
[01:42:39:319] V (1412) memspi: chip_id: 165E40␍␊
[01:42:39:319] <0x1b>[0m␍␊
[01:42:39:319] V (1422) memspi: raw_chip_id: 16405E␍␊
[01:42:39:319] <0x1b>[0m␍␊
[01:42:39:319] V (1422) memspi: chip_id: 5E4016␍␊
[01:42:39:319] <0x1b>[0m␍␊
[01:42:39:319] V (1432) bus_lock: dev 0 released.<0x1b>[0m␍␊
[01:42:39:319] V bus_lock: dev 0 acquired.␍␊
[01:42:39:319] V (1432) memspi: raw_chip_id: 5E1640␍␊
[01:42:39:319] <0x1b>[0m␍␊
[01:42:39:319] V (1442) memspi: chip_id: 40165E␍␊
[01:42:39:319] <0x1b>[0m␍␊
[01:42:39:319] V (1442) memspi: raw_chip_id: 405E16␍␊
[01:42:39:319] <0x1b>[0m␍␊
[01:42:39:319] V (1442) memspi: chip_id: 165E40␍␊
[01:42:39:319] <0x1b>[0m␍␊
[01:42:39:319] V (1452) bus_lock: dev 0 released.<0x1b>[0m␍␊
[01:42:39:319] V bus_lock: dev 0 acquired.␍␊
[01:42:39:319] V (1452) memspi: raw_chip_id: 16405E␍␊
[01:42:39:362] <0x1b>[0m␍␊
[01:42:39:362] V (1462) memspi: chip_id: 5E4016␍␊
[01:42:39:362] <0x1b>[0m␍␊
[01:42:39:362] V (1462) memspi: raw_chip_id: 5E1640␍␊
[01:42:39:362] <0x1b>[0m␍␊
[01:42:39:362] V (1462) memspi: chip_id: 40165E␍␊
[01:42:39:362] <0x1b>[0m␍␊
[01:42:39:362] V (1472) bus_lock: dev 0 released.<0x1b>[0m␍␊
[01:42:39:362] V bus_lock: dev 0 acquired.␍␊
[01:42:39:362] V (1472) memspi: raw_chip_id: 405E16␍␊
[01:42:39:362] <0x1b>[0m␍␊
[01:42:39:362] V (1482) memspi: chip_id: 165E40␍␊
[01:42:39:362] <0x1b>[0m␍␊
[01:42:39:362] V (1482) memspi: raw_chip_id: 16405E␍␊
[01:42:39:362] <0x1b>[0m␍␊
[01:42:39:362] V (1482) memspi: chip_id: 5E4016␍␊
[01:42:39:362] <0x1b>[0m␍␊
[01:42:39:362] V (1492) bus_lock: dev 0 released.<0x1b>[0m␍␊
[01:42:39:362] V bus_lock: dev 0 acquired.␍␊
[01:42:39:362] V (1492) memspi: raw_chip_id: 5E1640␍␊
[01:42:39:405] <0x1b>[0m␍␊
[01:42:39:405] V (1502) memspi: chip_id: 40165E␍␊
[01:42:39:405] <0x1b>[0m␍␊
[01:42:39:405] V (1502) memspi: raw_chip_id: 405E16␍␊
[01:42:39:405] <0x1b>[0m␍␊
[01:42:39:405] V (1502) memspi: chip_id: 165E40␍␊
[01:42:39:405] <0x1b>[0m␍␊
[01:42:39:405] V (1512) bus_lock: dev 0 released.<0x1b>[0m␍␊
[01:42:39:405] V bus_lock: dev 0 acquired.␍␊
[01:42:39:405] V (1512) memspi: raw_chip_id: 16405E␍␊
[01:42:39:405] <0x1b>[0m␍␊
[01:42:39:405] V (1522) memspi: chip_id: 5E4016␍␊
[01:42:39:405] <0x1b>[0m␍␊
[01:42:39:405] V (1522) memspi: raw_chip_id: 5E1640␍␊
[01:42:39:405] <0x1b>[0m␍␊
[01:42:39:405] V (1522) memspi: chip_id: 40165E␍␊
[01:42:39:405] <0x1b>[0m␍␊
[01:42:39:405] V (1532) bus_lock: dev 0 released.<0x1b>[0m␍␊
[01:42:39:405] V bus_lock: dev 0 acquired.␍␊
[01:42:39:405] V (1532) memspi: raw_chip_id: 405E16␍␊
[01:42:39:405] <0x1b>[0m␍␊
[01:42:39:405] V (1542) memspi: chip_id: 165E40␍␊
[01:42:39:442] <0x1b>[0m␍␊
[01:42:39:442] V (1542) memspi: raw_chip_id: 16405E␍␊
[01:42:39:442] <0x1b>[0m␍␊
[01:42:39:442] V (1552) memspi: chip_id: 5E4016␍␊
[01:42:39:442] <0x1b>[0m␍␊
[01:42:39:442] V (1552) bus_lock: dev 0 released.<0x1b>[0m␍␊
[01:42:39:442] <0x1b>[0;31mE (1552) example: Failed to initialize external Flash: ESP_ERR_FLASH_NOT_INITIALISED (0x6003)<0x1b>[0m␍␊

muratdemirtas, 2022-08-04:
bool use_iomux = spicommon_bus_using_iomux(config->host_id);
and put 'use_iomux=false;' under it.

**fix that bug.**

 I'll ask my colleagues to find and integrate a better solution for this.

[01:49:27:551] D (1302) spi: SPI2 use iomux pins.<0x1b>[0m␍␊
[01:49:27:551] V (1312) bus_lock: device registered on bus 1 slot 0.<0x1b>[0m␍␊
[01:49:27:551] anan␍␊
[01:49:27:551] D (1312) FLASH_HAL: extra_dummy: 0<0x1b>[0m␍␊
[01:49:27:551] V bus_lock: dev 0 acquired.␍␊
[01:49:27:551] V bus_lock: SPI dev changed from -1 to 0␍␊
[01:49:27:551] V (1322) bus_lock: dev 0 released.<0x1b>[0m␍␊
[01:49:27:594] V bus_lock: dev 0 acquired.␍␊
[01:49:27:594] V (1332) memspi: raw_chip_id: 16405E␍␊
[01:49:27:594] <0x1b>[0m␍␊
[01:49:27:594] V (1332) memspi: chip_id: 5E4016␍␊
[01:49:27:594] <0x1b>[0m␍␊
[01:49:27:594] V (1342) memspi: raw_chip_id: 16405E␍␊
[01:49:27:594] <0x1b>[0m␍␊
[01:49:27:594] V (1342) memspi: chip_id: 5E4016␍␊
[01:49:27:594] <0x1b>[0m␍␊
[01:49:27:594] V (1342) bus_lock: dev 0 released.<0x1b>[0m␍␊
[01:49:27:594] D (1352) spi_flash: trying chip: issi<0x1b>[0m␍␊
[01:49:27:594] V bus_lock: dev 0 acquired.␍␊
[01:49:27:594] V (1352) bus_lock: dev 0 released.<0x1b>[0m␍␊
[01:49:27:594] D (1362) spi_flash: trying chip: gd<0x1b>[0m␍␊
[01:49:27:594] V bus_lock: dev 0 acquired.␍␊
[01:49:27:594] V (1362) bus_lock: dev 0 released.<0x1b>[0m␍␊
[01:49:27:594] D (1372) spi_flash: trying chip: mxic<0x1b>[0m␍␊
[01:49:27:637] V bus_lock: dev 0 acquired.␍␊
[01:49:27:637] V (1372) bus_lock: dev 0 released.<0x1b>[0m␍␊
[01:49:27:637] D (1382) spi_flash: trying chip: winbond<0x1b>[0m␍␊
[01:49:27:637] V bus_lock: dev 0 acquired.␍␊
[01:49:27:637] V (1382) bus_lock: dev 0 released.<0x1b>[0m␍␊
[01:49:27:637] D (1392) spi_flash: trying chip: generic<0x1b>[0m␍␊
[01:49:27:637] V bus_lock: dev 0 acquired.␍␊
[01:49:27:637] V (1392) bus_lock: dev 0 released.<0x1b>[0m␍␊
[01:49:27:637] <0x1b>[0;32mI (1402) spi_flash: detected chip: generic<0x1b>[0m␍␊
[01:49:27:637] V bus_lock: dev 0 acquired.␍␊
[01:49:27:637] V (1402) bus_lock: dev 0 released.<0x1b>[0m␍␊
[01:49:27:637] <0x1b>[0;32mI (1412) spi_flash: flash io: fastrd<0x1b>[0m␍␊
[01:49:27:681] V bus_lock: dev 0 acquired.␍␊
[01:49:27:681] V (1412) bus_lock: dev 0 released.<0x1b>[0m␍␊
[01:49:27:681] V bus_lock: dev 0 acquired.␍␊
[01:49:27:681] V (1422) memspi: raw_chip_id: 16405E␍␊
[01:49:27:681] <0x1b>[0m␍␊
[01:49:27:681] V (1422) memspi: chip_id: 5E4016␍␊
[01:49:27:681] <0x1b>[0m␍␊
[01:49:27:681] V (1432) bus_lock: dev 0 released.<0x1b>[0m␍␊
[01:49:27:681] <0x1b>[0;32mI (1432) example: Initialized external Flash, size=4096 KB, ID=0x5e4016<0x1b>[0m␍␊

muratdemirtas, 2022-08-04:
@igrr

nopnop2002, 2022-08-04:
This is my result:
No problem.
Mayby your wire is not correct.
```
I (0) cpu_start: Starting scheduler on APP CPU.
I (344) example: Initializing external SPI Flash
I (354) example: Pin assignments:
I (354) example: MOSI: 23   MISO: 19   SCLK: 18   CS:  5
I (364) example: DMA CHANNEL: 3
I (364) spi_flash: detected chip: winbond
I (364) spi_flash: flash io: dio
I (374) example: Initialized external Flash, size=16384 KB, ID=0xef4018
I (384) example: Adding external Flash as a partition, label="storage", size=16384 KB
I (384) example: Listing data partitions:
I (394) example: - partition 'nvs', subtype 2, offset 0x9000, size 24 kB
I (404) example: - partition 'phy_init', subtype 1, offset 0xf000, size 4 kB
I (404) example: - partition 'storage', subtype 129, offset 0x0, size 16384 kB
I (414) example: Mounting FAT filesystem
I (424) example: FAT FS: 16212 kB total, 16208 kB free
I (424) example: Opening file
I (674) example: File written
I (674) example: Reading file
I (674) example: Read from file: 'Written using ESP-IDF v5.0-dev-4037-g9b8c558e63-dirty'
```

muratdemirtas, 2022-08-04:
> > We are using esp idf 4.4. I'm using the official example.
> 
> This is my result of official example:
> 
> ID=0xef4018 is w25q128.
> 
> No problem. Mayby your code is not correct.
> 
> ```
> I (0) cpu_start: Starting scheduler on APP CPU.
> I (344) example: Initializing external SPI Flash
> I (354) example: Pin assignments:
> I (354) example: MOSI: 23   MISO: 19   SCLK: 18   CS:  5
> I (364) example: DMA CHANNEL: 3
> I (364) spi_flash: detected chip: winbond
> I (364) spi_flash: flash io: dio
> I (374) example: Initialized external Flash, size=16384 KB, ID=0xef4018
> I (384) example: Adding external Flash as a partition, label="storage", size=16384 KB
> I (384) example: Listing data partitions:
> I (394) example: - partition 'nvs', subtype 2, offset 0x9000, size 24 kB
> I (404) example: - partition 'phy_init', subtype 1, offset 0xf000, size 4 kB
> I (404) example: - partition 'storage', subtype 129, offset 0x0, size 16384 kB
> I (414) example: Mounting FAT filesystem
> I (424) example: FAT FS: 16212 kB total, 16208 kB free
> I (424) example: Opening file
> I (674) example: File written
> I (674) example: Reading file
> I (674) example: Read from file: 'Written using ESP-IDF v5.0-dev-4037-g9b8c558e63-dirty'
> ```

https://esp32.com/viewtopic.php?t=27631
that solution is fixed my issue,
thanks for your help.
iam using original esp example code.

muratdemirtas, 2022-08-04:
[01:49:27:551] <0x1b>[0;32mI (1292) example: Initializing external SPI Flash<0x1b>[0m␍␊
[01:49:27:551] <0x1b>[0;32mI (1292) example: Pin assignments:<0x1b>[0m␍␊
[01:49:27:551] <0x1b>[0;32mI (1302) example: MOSI: 13   MISO: 12   SCLK: 14   CS:  0<0x1b>[0m␍␊
[01:49:27:551] D (1302) spi: SPI2 use iomux pins.<0x1b>[0m␍␊
[01:49:27:551] V (1312) bus_lock: device registered on bus 1 slot 0.<0x1b>[0m␍␊
[01:49:27:551] anan␍␊
[01:49:27:551] D (1312) FLASH_HAL: extra_dummy: 0<0x1b>[0m␍␊
[01:49:27:551] V bus_lock: dev 0 acquired.␍␊
[01:49:27:551] V bus_lock: SPI dev changed from -1 to 0␍␊
[01:49:27:551] V (1322) bus_lock: dev 0 released.<0x1b>[0m␍␊
[01:49:27:594] V bus_lock: dev 0 acquired.␍␊
[01:49:27:594] V (1332) memspi: raw_chip_id: 16405E␍␊
[01:49:27:594] <0x1b>[0m␍␊
[01:49:27:594] V (1332) memspi: chip_id: 5E4016␍␊
[01:49:27:594] <0x1b>[0m␍␊
[01:49:27:594] V (1342) memspi: raw_chip_id: 16405E␍␊
[01:49:27:594] <0x1b>[0m␍␊
[01:49:27:594] V (1342) memspi: chip_id: 5E4016␍␊
[01:49:27:594] <0x1b>[0m␍␊
[01:49:27:594] V (1342) bus_lock: dev 0 released.<0x1b>[0m␍␊
[01:49:27:594] D (1352) spi_flash: trying chip: issi<0x1b>[0m␍␊
[01:49:27:594] V bus_lock: dev 0 acquired.␍␊
[01:49:27:594] V (1352) bus_lock: dev 0 released.<0x1b>[0m␍␊
[01:49:27:594] D (1362) spi_flash: trying chip: gd<0x1b>[0m␍␊
[01:49:27:594] V bus_lock: dev 0 acquired.␍␊
[01:49:27:594] V (1362) bus_lock: dev 0 released.<0x1b>[0m␍␊
[01:49:27:594] D (1372) spi_flash: trying chip: mxic<0x1b>[0m␍␊
[01:49:27:637] V bus_lock: dev 0 acquired.␍␊
[01:49:27:637] V (1372) bus_lock: dev 0 released.<0x1b>[0m␍␊
[01:49:27:637] D (1382) spi_flash: trying chip: winbond<0x1b>[0m␍␊
[01:49:27:637] V bus_lock: dev 0 acquired.␍␊
[01:49:27:637] V (1382) bus_lock: dev 0 released.<0x1b>[0m␍␊
[01:49:27:637] D (1392) spi_flash: trying chip: generic<0x1b>[0m␍␊
[01:49:27:637] V bus_lock: dev 0 acquired.␍␊
[01:49:27:637] V (1392) bus_lock: dev 0 released.<0x1b>[0m␍␊
[01:49:27:637] <0x1b>[0;32mI (1402) spi_flash: detected chip: generic<0x1b>[0m␍␊
[01:49:27:637] V bus_lock: dev 0 acquired.␍␊
[01:49:27:637] V (1402) bus_lock: dev 0 released.<0x1b>[0m␍␊
[01:49:27:637] <0x1b>[0;32mI (1412) spi_flash: flash io: fastrd<0x1b>[0m␍␊
[01:49:27:681] V bus_lock: dev 0 acquired.␍␊
[01:49:27:681] V (1412) bus_lock: dev 0 released.<0x1b>[0m␍␊
[01:49:27:681] V bus_lock: dev 0 acquired.␍␊
[01:49:27:681] V (1422) memspi: raw_chip_id: 16405E␍␊
[01:49:27:681] <0x1b>[0m␍␊
[01:49:27:681] V (1422) memspi: chip_id: 5E4016␍␊
[01:49:27:681] <0x1b>[0m␍␊
[01:49:27:681] V (1432) bus_lock: dev 0 released.<0x1b>[0m␍␊
[01:49:27:681] <0x1b>[0;32mI (1432) example: Initialized external Flash, size=4096 KB, ID=0x5e4016<0x1b>[0m␍␊
[01:49:27:681] <0x1b>[0;32mI (1442) example: Adding external Flash as a partition, label="storage", size=4096 KB<0x1b>[0m␍␊
[01:49:27:681] D (1452) partition: Loading the partition table<0x1b>[0m␍␊
[01:49:27:681] V (1452) calculated md5: 0x3ffb5998   f4 ad 4f 45 38 56 4b 5d  74 35 b6 2c 75 b6 95 24  |..OE8VK]t5.,u..$|<0x1b>[0m␍␊
[01:49:27:723] V (1462) stored md5: 0x3f418070   f4 ad 4f 45 38 56 4b 5d  74 35 b6 2c 75 b6 95 24  |..OE8VK]t5.,u..$|<0x1b>[0m␍␊
[01:49:27:723] D (1472) partition: Partition table MD5 verified<0x1b>[0m␍␊
[01:49:27:723] <0x1b>[0;32mI (1472) example: Listing data partitions:<0x1b>[0m␍␊
[01:49:27:723] <0x1b>[0;32mI (1482) example: - partition 'nvs', subtype 2, offset 0x9000, size 24 kB<0x1b>[0m␍␊
[01:49:27:723] <0x1b>[0;32mI (1492) example: - partition 'phy_init', subtype 1, offset 0xf000, size 4 kB<0x1b>[0m␍␊
[01:49:27:723] <0x1b>[0;32mI (1492) example: - partition 'storage', subtype 129, offset 0x0, size 4096 kB<0x1b>[0m␍␊
[01:49:27:767] <0x1b>[0;32mI (1502) example: Mounting FAT filesystem<0x1b>[0m␍␊
[01:49:27:767] V (1512) wl_flash: config start_addr=0x00000000, full_mem_size=0x00400000, page_size=0x00001000, sector_size=0x00001000, updaterate=0x00000010, wr_size=0x00000010, version=0x00000002, temp_buff_size=0x00000020<0x1b>[0m␍␊
[01:49:27:767] D (1532) wl_flash: config - config result: state_size=0x00005000, cfg_size=0x00001000, addr_cfg=0x003ff000, addr_state1=0x003f5000, addr_state2=0x003fa000, flash_size=0x003f4000<0x1b>[0m␍␊
[01:49:27:810] V bus_lock: dev 0 acquired.␍␊
[01:49:27:810] V (1542) bus_lock: dev 0 released.<0x1b>[0m␍␊
[01:49:27:810] V bus_lock: dev 0 acquired.␍␊
[01:49:27:810] V (1552) bus_lock: dev 0 released.<0x1b>[0m␍␊
[01:49:27:810] D (1552) wl_flash: init - config ID=2, stored ID=2, access_count=0, block_size=4096, max_count=16, pos=0, move_count=0x00000000<0x1b>[0m␍␊
[01:49:27:810] D (1572) wl_flash: init starts: crc1= 0xb4a236d0, crc2 = 0xb4a236d0, this->state.crc= 0xb4a236d0, state_copy->crc= 0xb4a236d0, version=2, read_version=2<0x1b>[0m␍␊
[01:49:27:810] D (1582) wl_flash: init: crc1=0xb4a236d0, crc2 = 0xb4a236d0, result= 0x00000000<0x1b>[0m␍␊
[01:49:27:853] V (1592) wl_flash: recoverPos start<0x1b>[0m␍␊
[01:49:27:853] V bus_lock: dev 0 acquired.␍␊
[01:49:27:853] V (1592) bus_lock: dev 0 released.<0x1b>[0m␍␊
[... 79 lines cut ...]
[01:49:28:327] V (2082) ff_diskio_spiflash: ff_wl_write - pdrv=0, sector=2, count=1␍␊
[01:49:28:327] <0x1b>[0m␍␊
[01:49:28:327] D (2082) wl_flash: erase_range - start_address= 0x00002000, size= 0x00001000<0x1b>[0m␍␊
[01:49:28:327] D (2092) wl_flash: erase_sector - sector= 0x00000002<0x1b>[0m␍␊
[01:49:28:327] V (2092) wl_flash: calcAddr - addr= 0x00002000 -> result= 0x00003000, dummy_addr= 0x00000000<0x1b>[0m␍␊
[01:49:28:343] V bus_lock: dev 0 acquired.␍␊
[... cut ...]
[01:49:28:439] V (2182) ff_diskio_spiflash: ff_wl_read - pdrv=0, sector=6, count=1␍␊
[01:49:28:439] <0x1b>[0m␍␊
[01:49:28:439] D (2192) wl_flash: read - src_addr= 0x00006000, size= 0x00001000<0x1b>[0m␍␊
[01:49:28:439] V (2202) wl_flash: calcAddr - addr= 0x00006000 -> result= 0x00007000, dummy_addr= 0x00000000<0x1b>[0m␍␊
[01:49:28:439] V bus_lock: dev 0 acquired.␍␊
[01:49:28:439] V (2212) bus_lock: dev 0 released.<0x1b>[0m␍␊
[01:49:28:439] <0x1b>[0;32mI (2212) example: Read from file: 'Written using ESP-IDF v4.4.1-593-g6d85d53cee-dirty'<0x1b>[0m␍␊

closed 2022-08-04 by muratdemirtas

nopnop2002, 2022-08-04:
```
chip_id: 5E4016
```

There are chips that work properly and chips that don't.

I tested these.
But I couldn't get it working.

GD25Q64
SST25VF016B
SST25VF032B

muratdemirtas, 2022-08-04:
W25Q128FV initialized successfully.

[02:11:41:380] <0x1b>[0;32mI (1292) example: Initializing external SPI Flash<0x1b>[0m␍␊
[02:11:41:380] <0x1b>[0;32mI (1292) example: Pin assignments:<0x1b>[0m␍␊
[02:11:41:380] <0x1b>[0;32mI (1302) example: MOSI: 13   MISO: 12   SCLK: 14   CS:  0<0x1b>[0m␍␊
[02:11:41:380] D (1302) spi: SPI2 use iomux pins.<0x1b>[0m␍␊
[02:11:41:380] V (1312) bus_lock: device registered on bus 1 slot 0.<0x1b>[0m␍␊
[02:11:41:380] anan␍␊
[02:11:41:380] D (1312) FLASH_HAL: extra_dummy: 0<0x1b>[0m␍␊
[02:11:41:380] V bus_lock: dev 0 acquired.␍␊
[02:11:41:380] V bus_lock: SPI dev changed from -1 to 0␍␊
[02:11:41:380] V (1322) bus_lock: dev 0 released.<0x1b>[0m␍␊
[02:11:41:423] V bus_lock: dev 0 acquired.␍␊
[02:11:41:423] V (1332) memspi: raw_chip_id: 1840EF␍␊
[02:11:41:423] <0x1b>[0m␍␊
[02:11:41:423] V (1332) memspi: chip_id: EF4018␍␊
[02:11:41:423] <0x1b>[0m␍␊
[02:11:41:423] V (1342) memspi: raw_chip_id: 1840EF␍␊
[02:11:41:423] <0x1b>[0m␍␊
[02:11:41:423] V (1342) memspi: chip_id: EF4018␍␊
[02:11:41:423] <0x1b>[0m␍␊
[02:11:41:423] V (1342) bus_lock: dev 0 released.<0x1b>[0m␍␊
[02:11:41:423] D (1352) spi_flash: trying chip: issi<0x1b>[0m␍␊
[02:11:41:423] V bus_lock: dev 0 acquired.␍␊
[02:11:41:423] V (1352) bus_lock: dev 0 released.<0x1b>[0m␍␊
[02:11:41:423] D (1362) spi_flash: trying chip: gd<0x1b>[0m␍␊
[02:11:41:423] V bus_lock: dev 0 acquired.␍␊
[02:11:41:423] V (1362) bus_lock: dev 0 released.<0x1b>[0m␍␊
[02:11:41:423] D (1372) spi_flash: trying chip: mxic<0x1b>[0m␍␊
[02:11:41:466] V bus_lock: dev 0 acquired.␍␊
[02:11:41:466] V (1372) bus_lock: dev 0 released.<0x1b>[0m␍␊
[02:11:41:466] D (1382) spi_flash: trying chip: winbond<0x1b>[0m␍␊
[02:11:41:466] V bus_lock: dev 0 acquired.␍␊
[02:11:41:466] V (1382) bus_lock: dev 0 released.<0x1b>[0m␍␊
[02:11:41:466] <0x1b>[0;32mI (1392) spi_flash: detected chip: winbond<0x1b>[0m␍␊
[02:11:41:466] V bus_lock: dev 0 acquired.␍␊
[02:11:41:466] V (1392) bus_lock: dev 0 released.<0x1b>[0m␍␊
[02:11:41:466] <0x1b>[0;32mI (1402) spi_flash: flash io: fastrd<0x1b>[0m␍␊
[02:11:41:466] V bus_lock: dev 0 acquired.␍␊
[02:11:41:466] V (1402) bus_lock: dev 0 released.<0x1b>[0m␍␊
[02:11:41:466] V bus_lock: dev 0 acquired.␍␊
[02:11:41:466] V (1412) memspi: raw_chip_id: 1840EF␍␊
[02:11:41:509] <0x1b>[0m␍␊
[02:11:41:509] V (1412) memspi: chip_id: EF4018␍␊
[02:11:41:509] <0x1b>[0m␍␊
[02:11:41:509] V (1422) bus_lock: dev 0 released.<0x1b>[0m␍␊
[02:11:41:509] <0x1b>[0;32mI (1422) example: Initialized external Flash, size=16384 KB, ID=0xef4018<0x1b>[0m␍␊
[02:11:41:509] <0x1b>[0;32mI (1432) example: Adding external Flash as a partition, label="storage", size=16384 KB<0x1b>[0m␍␊
[02:11:41:509] D (1442) partition: Loading the partition table<0x1b>[0m␍␊
[02:11:41:509] V (1442) calculated md5: 0x3ffb5998   f4 ad 4f 45 38 56 4b 5d  74 35 b6 2c 75 b6 95 24  |..OE8VK]t5.,u..$|<0x1b>[0m␍␊
[02:11:41:509] V (1452) stored md5: 0x3f418070   f4 ad 4f 45 38 56 4b 5d  74 35 b6 2c 75 b6 95 24  |..OE8VK]t5.,u..$|<0x1b>[0m␍␊
[02:11:41:553] D (1462) partition: Partition table MD5 verified<0x1b>[0m␍␊
[02:11:41:553] <0x1b>[0;32mI (1472) example: Listing data partitions:<0x1b>

muratdemirtas, 2022-08-04:
> ```
> chip_id: 5E4016
> ```
> 
> There are chips that work properly and chips that don't.
> 
> I tested these. But I couldn't get it working.
> 
> GD25Q64 SST25VF016B SST25VF032B

maybe you might be do this.

https://esp32.com/viewtopic.php?t=27631