espressif/esp-idf #9494: examples/protocols/smtp_client

nopnop2002, 2022-08-04:
## Environment
```
$ idf.py --version
ESP-IDF v5.0-dev-4037-g9b8c558e63-dirty
```

## Problem Description

To help keep your account secure, from May 30, 2022, Google no longer supports the use of third-party apps or devices which ask you to sign in to your Google Account using only your username and password.

As a result, this sample does not work properly.

## Debug Logs

```
220 smtp.googlemail.com ESMTP im22-20020a170902bb1600b0016bdeb58611sm2901483plb.112 - gsmtp
I (14233) smtp_example: Writing EHLO to server...

250-smtp.googlemail.com at your service, [61.211.140.68]
250-SIZE 35882577
250-8BITMIME
250-STARTTLS
250-ENHANCEDSTATUSCODE
S
250-PIPELINING
250-CHUNKING
250 SMTPUTF8
I (14443) smtp_example: Writing STARTTLS to server...

220 2.0.0 Ready to start TLS
I (14623) smtp_example: Performing the SSL/TLS handshake...
I (15973) smtp_example: Verifying peer X.509 certificate...
I (15973) smtp_example: Certificate verified.
I (15973) smtp_example: Cipher suite is TLS-ECDHE-ECDSA-WITH-AES-128-GCM-SHA256
I (15983) smtp_example: Authentication...
I (15983) smtp_example: Write AUTH LOGIN
I (16273) smtp_example: Write USER NAME
I (16483) smtp_example: Write PASSWORD
I (16893) smtp_example: write_ssl_and_get_response=535
E (16903) smtp_example: Last error was: -0xfffffde9 - UNKNOWN ERROR CODE (0200) : UNKNOWN ERROR CODE (0017)
```

vikramdattu, 2022-08-04:
Hi @nopnop2002 thanks for pointing out that google stopped Less secure apps option.

There however is a workaround to still make this example useful. You need to generate special app password.
I found this on StackOverflow:  https://stackoverflow.com/a/72626684/3249583

nopnop2002, 2022-08-04:
@mahavirj 

>There however is a workaround to still make this example useful.

I tried.
It's work fine.

However, the procedure is complicated.
I would like to wait for an official fix.