espressif/esp-idf #9495: ESP32-C3 pulls up GPIO19 because it is used by USB

DCSBL, 2022-08-04:
## Environment

- Development Kit:      None
- Kit version (for WroverKit/PicoKit/DevKitC): na
- Module or chip used:  ESP32-C3
- IDF version (run ``git describe --tags`` to find it): v4.4.1
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): riscv32-esp-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0 (or xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0)
- Operating System:     macOS
- (Windows only) environment type: Na
- Using an IDE?: VSCode
- Power Supply:         external 5V

## Problem Description
The GPIO pull-up configuration to configure GPIO19 does not work as expected.

### Actual Behavior
After configuring gpio with the config below, the input is still pulled up. 
```c
gpio_config_t io_conf = {};
io_conf.pin_bit_mask = BIT64(GPIO_NUM_19);
io_conf.mode = GPIO_MODE_INPUT;
io_conf.pull_up_en = 0;
io_conf.pull_down_en = 0;
gpio_config(&io_conf);
```  

To fix this, we have to clear the `USB_SERIAL_JTAG_DP_PULLUP` bit with `CLEAR_PERI_REG_MASK(USB_SERIAL_JTAG_CONF0_REG, USB_SERIAL_JTAG_DP_PULLUP);`. That works but only after a second reset. -> 

> https://www.espressif.com/sites/default/files/documentation/esp32-c3_datasheet_en.pdf page 21
> USB - GPIO18 and GPIO19 are USB pins. The pull-up value of a USB pin is controlled by the pin’s pull-up value together with USB pull-up value. If any of the two pull-up values is 1, the pin’s pull-up resistor will be enabled. The pull-up resistors of USB pins are controlled by USB_SERIAL_JTAG_DP_PULLUP bit.

### Expected Behavior
We expect that GPIO19 is put in the desired state without other config. Or that this, or 'a' workaround is documented somewhere (if that is not the case)

closed 2022-08-24 by espressif-bot

DCSBL, 2022-08-24:
Thanks @zikalino !

michaesc, 2024-01-16:
Does anyone happen to have a **ESP-IDF** (**5.1.2** or similar) project that demonstrates (re)using the **GPIO18** and **GPIO19** of a **ESP32-C3** for any simple purpose?

I'm trying to put a simple **SPST** tactical switch (push button) on each pin, which drives **GPIO18** and **GPIO19** low when pressed. The **MCU** should get an interrupt, but this is not happening. I'm not sure why, and I've written to the forum asking for help.

wormyrocks, 2024-01-18:
Assuming you aren't using the USB JTAG feature, have you tried setting the eFuse bit for DIS_USB_SERIAL_JTAG?
`espefuse.py -p COMx burn_efuse DIS_USB_SERIAL_JTAG 1`
Anecdotally, it disables the pull-up on IO19 on boot, but I haven't tested whether I can subsequently reuse the pin as you describe.
Keep in mind it's a one-time programmable setting.

michaesc, 2024-01-18:
I'm trying to not use the eFuse, in order to allow users to selectively disable USB-JTAG.

It makes no sense to me that the feature cannot be disabled, and worse yet there is no documentation stating whether it can or cannot be disabled. I'm still trying to debug the problem.