espressif/esp-idf #9496: Undefined reference to function when using cpp project and cpp component

krupis, 2022-08-04:
Hello. I have been playing around with ESP32-S3 multi-adv example. I have written a code and made some minor changes to build code for c++ project.
I have created a BLE component and my project structure is as following:
[image]

my CMakeLists.txt under BLE component:

```
idf_component_register(SRCS "BLE_custom.cpp"
                    INCLUDE_DIRS "."
                    REQUIRES bt)

target_compile_options(${COMPONENT_LIB} PRIVATE -Wno-unused-const-variable)
target_compile_options(${COMPONENT_LIB} PRIVATE -Wno-missing-field-initializers)
target_compile_options(${COMPONENT_LIB} PRIVATE -Wno-maybe-uninitialized)
```

my CMakeLists.txt under main :
```
idf_component_register(SRCS "main.cpp"
                    INCLUDE_DIRS ".")

target_compile_options(${COMPONENT_LIB} PRIVATE -Wno-unused-const-variable)
```

My main.cpp code:
```
extern "C" {
   void app_main(); 
   
}

#include "BLE_custom.h"

#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "esp_bt.h"
#include "sdkconfig.h"

void app_main(void)
{
    esp_err_t ret;
    ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK( ret );

    BLE_setup();

    return;
}

```

As you can see from main.cpp, I #include "BLE_custom.h" and try to call function BLE_setup() which is defined in BLE_custom.cpp. I also have a function prototype inside BLE_custom.h so everything should be fine.

My BLE_custom.h:

```

#ifndef BLE_CUSTOM_H
#define BLE_CUSTOM_H

#ifdef __cplusplus
extern "C" {
#endif

#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "esp_bt.h"

#include "esp_gap_ble_api.h"
#include "esp_gatts_api.h"
#include "esp_bt_defs.h"
#include "esp_bt_main.h"
#include "esp_gatt_common_api.h"
#include "sdkconfig.h"
#include "freertos/semphr.h"

#define LOG_TAG "MULTI_ADV_DEMO"
#define GATTS_TABLE_TAG "SEC_GATTS_DEMO"
#define HEART_PROFILE_NUM                         1
#define HEART_PROFILE_APP_IDX                     0
#define ESP_HEART_RATE_APP_ID                     0x55
#define HEART_RATE_SVC_INST_ID                    0
#define EXT_ADV_HANDLE                            0
#define NUM_EXT_ADV_SET                           1
#define EXT_ADV_DURATION                          0
#define EXT_ADV_MAX_EVENTS                        0
#define GATTS_DEMO_CHAR_VAL_LEN_MAX               0x40

#define ESP_FW_VERSION "1.1.3"
#define ESP_HW_VERSION "1.0.1"

#define ENDIAN_CHANGE_U16(x) ((((x)&0xFF00)>>8) + (((x)&0xFF)<<8))
#define BLE_PACKET_SPLITTER         20
#define CHAR_DECLARATION_SIZE       (sizeof(uint8_t))
#define SVC_INST_ID                 0

#define FUNC_SEND_WAIT_SEM(func, sem) do {\
        esp_err_t __err_rc = (func);\
        if (__err_rc != ESP_OK) { \
            ESP_LOGE(LOG_TAG, "%s, message send fail, error = %d", __func__, __err_rc); \
        } \
        xSemaphoreTake(sem, portMAX_DELAY); \
} while(0);

void build_ble_response_packet(uint8_t op_code, uint8_t message_id, uint16_t total_length, uint8_t* payload,esp_gatt_if_t gatts_if,uint16_t conn_id);
void Multi_packet_response(uint16_t data_length,uint8_t* data,esp_gatt_if_t gatts_if,uint16_t conn_id);
void BLE_setup();

void change_BLE_password(uint32_t new_password);

enum
{
    BLE_SERVICE,
    BLE_WRITE_CHARACTERISTIC,
    BLE_WRITE_VALUE,
    BLE_READ_CHARACTERISTIC,
    BLE_READ_VALUE,
    BLE_WRITE_DECLARATION,
    BLE_IDX_NB,
};

struct gatts_profile_inst {
    esp_gatts_cb_t gatts_cb;
    uint16_t gatts_if;
    uint16_t app_id;
    uint16_t conn_id;
    uint16_t service_handle;
    esp_gatt_srvc_id_t service_id;
    uint16_t char_handle;
    esp_bt_uuid_t char_uuid;
    esp_gatt_perm_t perm;
    esp_gatt_char_prop_t property;
    uint16_t descr_handle;
    esp_bt_uuid_t descr_uuid;
};

#ifdef __cplusplus
}
#endif

#endif

```

The Error message I receive:
```
FAILED: multi_adv_demos.elf 
cmd.exe /C "cd . && C:\Users\petrikas.lu\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s3-elf\bin\xtensa-esp32s3-elf-g++.exe -mlongcalls  @CMakeFiles\multi_adv_demos.elf.rsp -o multi_adv_demos.elf  && cd ."
c:/users/petrikas.lu/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/main/libmain.a(main.cpp.obj):(.literal.app_main+0x1c): undefined reference to `BLE_setup'
c:/users/petrikas.lu/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/main/libmain.a(main.cpp.obj): in function `app_main':
C:\Users\petrikas.lu\Desktop\WORK\ESP32_MULTI_GAP2\build/../main/main.cpp:31: undefined reference to `BLE_setup'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

I am not sure what is the issue here. I use the same technique for my another project for ESP32 (First time using ESP32-S3) and everything seemed fine. Could someone explain to me what I am doing wrong?

igrr, 2022-08-04:
In BLE_custom.cpp, have you defined `void BLE_setup()` as `extern "C"`? If not, the main.cpp will be looking for BLE_setup with C linkage, but only a BLE_setup with C++ linkage (name mangled) will be defined.

krupis, 2022-08-04:
I see what you mean. Thank you

closed 2022-08-04 by krupis