espressif/esp-idf #9499: Secure Boot causes invalid header

tobyloki, 2022-08-05:
I have tried enabling secure boot on top of flash encryption, and it causing the following error. Before adding secure boot, I was only using flash encryption and it was working just fine and I was able to see the output logs.
```cmd
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xddae3f82
invalid header: 0xddae3f82
invalid header: 0xddae3f82
invalid header: 0xddae3f82
invalid header: 0xddae3f82
invalid header: 0xddae3f82
invalid header: 0xddae3f82
ets Jul 29 2019 12:21:46
```

Here are a list of my sdkconfig.defaults regarding secure boot and flash encryption.
```
# Flash encryption & NVS encryption
CONFIG_FLASH_ENCRYPTION_ENABLED=y
CONFIG_SECURE_FLASH_ENCRYPTION_MODE_RELEASE=y
CONFIG_NVS_ENCRYPTION=y
CONFIG_PARTITION_TABLE_OFFSET=0xc000
# Secure boot
CONFIG_SECURE_BOOT=y
CONFIG_SECURE_BOOT_V1_ENABLED=y
CONFIG_SECURE_BOOTLOADER_REFLASHABLE=y
CONFIG_SECURE_BOOT_SIGNING_KEY="credentials/keys/secure_boot_signing_key.pem"
```

Here are the commands I ran to install the program. I built everything first, and the bootloader, partition table, and app image are signed (not sure about ota_data_initial.bin as it doesn't generate a file like ota_data_initial_unsigned.bin). I then pre encrypt the binaries on the host machine. I then upload the encrypted bootloader first. Then I then upload the encrypted partition table, ota_data_initial,  and app image.
```
idf.py build

espsecure.py encrypt_flash_data --keyfile credentials/keys/flash_encryption_key.bin --address 0x1000 --output factoryFiles/bootloader-reflash-digest.bin build\bootloader\bootloader-reflash-digest.bin

espsecure.py encrypt_flash_data --keyfile credentials/keys/flash_encryption_key.bin --address 0xc000 --output factoryFiles/partition-table.bin build\partition_table\partition-table.bin

espsecure.py encrypt_flash_data --keyfile credentials/keys/flash_encryption_key.bin --address 0x16000 --output factoryFiles/ota_data_initial.bin build\ota_data_initial.bin

espsecure.py encrypt_flash_data --keyfile credentials/keys/flash_encryption_key.bin --address 0x20000 --output factoryFiles/myprogram.bin build\myprogram.bin
# Note: I did get a warning about: "Padding with 12 bytes of random data (encrypted data must be multiple of 16 bytes long)"

esptool.py -b 460800 --before default_reset --after no_reset --chip esp32  write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 factoryFiles/bootloader-reflash-digest.bin

esptool.py -b 460800 --before default_reset --after no_reset --chip esp32  write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0xc000 factoryFiles/partition-table.bin 0x16000 factoryFiles/ota_data_initial.bin 0x20000 factoryFiles/myprogram.bin

idf.py monitor
```

Here is my partition table:
```
sec_cert,    0x3F, ,         ,         0x3000, ,
nvs,         data, nvs,      ,         0x6000, ,
otadata,     data, ota,      ,         0x2000, encrypted,
phy_init,    data, phy,      ,         0x1000, encrypted,
ota_0,       app,  ota_0,    ,         1600K,  encrypted,
ota_1,       app,  ota_1,    ,         1600K,  encrypted,
```

After running `idf.py monitor`, I get these logs, and the program doesn't run. How can I fix this problem?
```cmd
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xddae3f82
invalid header: 0xddae3f82
invalid header: 0xddae3f82
invalid header: 0xddae3f82
invalid header: 0xddae3f82
invalid header: 0xddae3f82
invalid header: 0xddae3f82
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xddae3f82
invalid header: 0xddae3f82
invalid header: 0xddae3f82
invalid header: 0xddae3f82
invalid header: 0xddae3f82
invalid header: 0xddae3f82
invalid header: 0xddae3f82
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xddae3f82
invalid header: 0xddae3f82
invalid header: 0xddae3f82
invalid header: 0xddae3f82
invalid header: 0xddae3f82
invalid header: 0xddae3f82
invalid header: 0xddae3f82
ets Jul 29 2019 12:21:46
```

mahavirj, 2022-08-23:
@tobyloki 

Image `bootloader-reflash-digest.bin` contains both (secure boot) digest and bootloader binary. Its offset must be 0x0 (and not 0x1000) as digest occupies very 1st sector and rest image is bootloader binary. Can you please try replacing offset from `0x1000` to `0x0` and recheck (in both image generation and flashing commands)?

Relevant: https://github.com/espressif/esp-idf/issues/3548

tobyloki, 2022-08-23:
Thanks. That fixed the problem. It is working now.

closed 2022-08-24 by mahavirj