espressif/esp-idf #9500: Function from another cpp component is not accessible inside header file but accessible in cpp file

krupis, 2022-08-05:
Hello. I have issues when I try to call a function that is included from another CPP file inside my static function.

Project structure is as following:

[image]

Inside logging.cpp I have declared a function:
```
bool logging_30_min_data()
```

Inside logging.h I have added a function prototype:
```

#ifndef LOGGING_H
#define LOGGING_H

#ifdef __cplusplus
extern "C" {
#endif

#include "M95M04.h"
#include "Thermostat.h"
#include "stdint.h"
#include "stdio.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"
#include "esp_sntp.h"
#include <time.h>
#include <sys/time.h>
#include "esp_log.h"
#include <math.h>
#include "params.h"

extern QueueHandle_t logging_queue;
extern QueueHandle_t alarm_logging_queue;

void Logging_task(void* param);
void alarm_logging_task(void* param);

void EEPROM_erase_all(spi_device_handle_t spi);

void EEPROM_setup();

uint32_t endian_swap_32b(uint32_t val);
uint16_t endian_swap_16b(uint16_t val);
s_thermostat_logging convert_struct_to_big_endian_log(s_thermostat_logging* data);
alarm_logging_s convert_struct_to_big_endian_alarm(alarm_logging_s* data);

bool logging_30_min_data();
bool logging_alarm(uint16_t alarm_mask);

#ifdef __cplusplus
}
#endif

#endif

```

In my thermostat.h I include #include "logging.h".

```
#ifndef THERMOSTAT_H
#define THERMOSTAT_H

#ifdef __cplusplus
extern "C" {
#endif

#include "stdint.h"
#include "stdio.h"
#include "params.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"
#include "Temperature_sensor.h"
#include "Motion.h"
#include "esp_timer.h"
#include <cstring>
#include "esp_sntp.h"
#include <time.h>
#include <sys/time.h>
#include "logging.h"

#ifdef __cplusplus
}
#endif

#endif

static void test_function(void* arg){
logging_30_min_data(); // cannot be called.
}

```

I can use the functions that are declared in logging.cpp inside thermostat.cpp but not thermostat.h. What is the issue here and how can I reach the logging functions inside thermostat header file?

igrr, 2022-08-07:
@krupis this looks like an exact same problem as https://github.com/espressif/esp-idf/issues/9496, doesn't it?
The function is declared in the header file with C linkage (in an extern "C" block) but is implemented only with C++ linkage.
When compiling the translation unit of the caller, the compiler creates a reference to the function with C linkage. But since such function is not defined anywhere, linking fails.
The solution is to either:
- declare the function outside of extern C block — works if all your code is C++
- define the function also as extern "C" (same as suggested in the linked issue).

I also kindly recommend reading about the difference between function declaration and definition in C++, i think your question might becaused by misunderstanding what the declaration does.

igrr, 2022-09-29:
@krupis I'm going to close the issue as I think my comment above should solve the problem. If the problem is still not fixed, please feel free to reopen the issue.

closed 2022-09-29 by igrr