espressif/esp-idf #9501: i2c.c anomalies

ammaree, 2022-08-05:
I have tried to understand the logic relating to the 'clear_bus_cnt' variable but so far have failed completely. 
The lines listed below are all the references related in/directly to this variable.

90.     #define I2C_ACKERR_CNT_MAX             (10)
1492   static uint8_t clear_bus_cnt = 0;
1507        clear_bus_cnt = 0;
1553                    clear_bus_cnt = 0;
1556                    clear_bus_cnt++;
 	                    if (clear_bus_cnt >= I2C_ACKERR_CNT_MAX) {
 	                        clear_bus_cnt = 0;
 	                    }
 1573            clear_bus_cnt = 0;

I have commented out all the lines, recompiled with no errors and the application has been running for months?
Am I missing something, does this variable and the related code actually do anything useful?

o-marshmallow, 2022-08-08:
Hi @ammaree ,

You are right, that's an issue that we've found recently. Indeed, in the `if (clear_bus_cnt >= I2C_ACKERR_CNT_MAX) ` statement, we **must** manually call the fsm reset function, else, the variable is useless.

An internal merge request is already on-going.

AxelLin, 2022-11-17:
fixed in master: 75ede16b3872b4989bad52728eee4d1dcfd6f46f
But v5.0 and v4.x still need fix.

AxelLin, 2022-12-17:
v5.0: 4ef2ba26f98f423805bbb63a3ca6a8eb6ab5deb4
v4.4: 3ec5041255bc6877713df88168955b5f4050f4a1

AxelLin, 2023-05-05:
@o-marshmallow 
Looks like v4.3 branch needs the backport fix. (Missing i2c_hw_fsm_reset() call)
https://github.com/espressif/esp-idf/blob/release/v4.3/components/driver/i2c.c#L1245-L1247

ammaree, 2024-02-14:
Have moved on to IDF v5.3 and the new master driver so guess this issue is not relevant any more

closed 2024-02-14 by ammaree