espressif/esp-idf #9502: INCORRECT data on UART TRANSMIT

Josh-bolt, 2022-08-05:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No|Yes (please give details)]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description
Hi,
Im working on uart transmission using esp32s3 microcontroller and RS485 ic module,
While transmitting the data buffer like {3,1,1} instead of getting the data i have sent, im getting data like {3 0 0 0 1 0 0 0 1} with 3 zeros appended in between each time. below i have attached the code, it will be helpful if someone explains why this is happening,
below have attached my code 
//Detailed problem description goes here.

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include <sys/time.h>
#include <errno.h>
#include "freertos/event_groups.h"
#include "freertos/queue.h"
#include "freertos/semphr.h"
#include <time.h>
#include "sdkconfig.h"
#include "driver/timer.h"
#include "driver/uart.h"
#include "driver/gpio.h"
#include "driver/adc.h"
#include "esp_adc_cal.h"
#include "esp_system.h"
#include "esp_log.h"
[... 200 lines cut ...]
    io_conf.intr_type = GPIO_INTR_DISABLE;
    //set as output mode
    io_conf.mode = GPIO_MODE_INPUT_OUTPUT;
    //bit mask of the pins that you want to set,e.g.GPIO18/19
    io_conf.pin_bit_mask = GPIO_OUTPUT_PIN_SEL;
    //disable pull-down mode
    io_conf.pull_down_en = 0;
    //disable pull-up mode
    io_conf.pull_up_en = 0;
    //configure GPIO with the given settings
    //gpio_set_level(GPIO_OUTPUT_IO_1, 0);
    printf("%d", gpio_get_level(GPIO_OUTPUT_IO_1));
    gpio_config(&io_conf);
    Tx_handle = xTimerCreate("Transmitter",pdMS_TO_TICKS(1000), pdTRUE,0,Transmit_data_callback);
    //xTaskCreate(uart_event_task, "uart_event_task", 4048, NULL, 12, &Rx_task);
    xTaskCreate(tx_task, "uart_tx_task", 2048, NULL, 11, &Tx_task);
}
```

output picture using the serial reader

[image]

picture of the waveform from the logic analyzer is below
[image]

Josh-bolt, 2022-08-05:
@espressif-bot

grzegorzniemirowski, 2022-08-06:
Hello @Josh-bolt There are several errors here:
1. How many elements does `data` array have? You specify the size as 2 but you initialize it with 3 elements. The specified size is more important hence the content is {3, 1}, not {3,1,1}.
2. Another length problem: why is `len` set to 5?
3. The most important thing: `uart_write_bytes()` takes, not surprisingly, bytes. But you pass ints. What is actually passed is 4 bytes of first element (int equal to 3, store in memory as bytes {3, 0, 0, 0, 0}) and 1 byte of second element (int equal to 1). So you get {3, 0, 0, 0, 1} as one int takes 4 bytes. The `data` array is 8 bytes in size and you take first 5 of them.
If you operate on bytes then use uint8_t, not int. And pay attention to how many bytes you wish to send. maybe `sizeof()` would be better instead of `len`?

Josh-bolt, 2022-08-08:
@grzegorzniemirowski THANKS!

closed 2022-08-08 by Josh-bolt