espressif/esp-idf #9505: BLE failed to set new device name after reconnection

jrHernandezSio, 2022-08-05:
**ENVIRONMENT:**

espressif32@3.3.2 -> framework-espidf @ 3.40300.0 (4.3.0)
BLE driver: Bluedroid (loaded dynamically)

**DESCRIPTION**
I need to change device name when advertising BLE and it is giving an error if the device was already advertising with other name. See full details below.

[image]

I have tried unregister `esp_ble_gatts_app_unregister(gl_profile_tab[PROFILE_A_APP_ID].gatts_if);` before registering again and no success

Disconection process before reconnection in same boot cylce:

```
    esp_ble_gap_stop_advertising();
    
    esp_bluedroid_disable();
    esp_bluedroid_deinit();

    esp_bt_controller_disable();
    esp_bt_controller_deinit();
 
    esp_ble_gatts_app_unregister(gl_profile_tab[PROFILE_A_APP_ID].gatts_if);
```

Instead of setting my new name, the "ESP32" name is set.

The name set is done inside handler:
[image]

Am I missing sth ?

Regards, Ramon.

esp-cjh, 2022-08-08:
The error code means that your argument is invalid, please check your argument not NULL and the length of argument no more than `CONFIG_BT_MAX_DEVICE_NAME_LEN`.

The device name in advertising is included in advertising data or scan response data. So if you want to change the device name when advertising, you don't need to reboot or stop advertising. Just follow the steps below:
```
/* set device name */
esp_ble_gap_set_device_name("device name");
/* if adv data include name */
esp_ble_gap_config_adv_data(&adv_data);
/* or if scan rsp data include name */
esp_ble_gap_config_adv_data(&scan_rsp_data);
```

jrHernandezSio, 2022-08-22:
Hi,

Thanks for your helpful answer.

You were right, I was passing a longer device name (37 bytes) due to a non-initialized buffer in my side.

Regards, Ramon.

closed 2022-08-22 by jrHernandezSio