espressif/esp-idf #9509: ESP-IDF v4.4.2 release not available, VSCode extension can't install ESP-IDF, Online installer fails

grzegorzniemirowski, 2022-08-06:
It seems that latest ESP-IDF v.4.4 version is 4.4.2. This information is here:
https://docs.espressif.com/projects/esp-idf/en/stable/esp32/
[image]

Unfortunately it can't be downloaded. The release is not available at
https://github.com/espressif/esp-idf/releases/tag/v4.4.2
There is just the source code. You can't also download v4.4.2 from
https://dl.espressif.com/dl/esp-idf/
There is an online installer which fails when v4.4.2 is selected.
[image]

The same applies to VSCode Extension. It fails when it tries to install ESP-IDF.
[image]

Please make release 4.4.2 available or don't pretend it's the latest version so the installer won't try to download something which doesn't exist.

tongclement, 2022-08-06:
Same issue here:
Error message:
```Waiting 8 seconds...
File download response error: Failed web connection with error code: 404
https://github.com/espressif/esp-idf/releases/download/v4.4.2/esp-idf-v4.4.2.zip
Failed download. Retrying...
Error: Error: HTTP/HTTPS Response Error
Waiting 16 seconds...
File download response error: Failed web connection with error code: 404
https://github.com/espressif/esp-idf/releases/download/v4.4.2/esp-idf-v4.4.2.zip
Failed download. Retrying...
Error: Error: HTTP/HTTPS Response Error
Downloaded release/v4.4 (release branch). Extracting..```

tongclement, 2022-08-06:
[https://drive.google.com/file/d/1K1mtP5ifr1B3jY-AOdW8TTc5_YkGKHT4/view?usp=sharing](url)

klandreth-cardinal, 2022-08-06:
They tagged a release without doing a release...

moefear85, 2022-08-07:
yep, i can confirm, esp-idf is broken again.

igrr, 2022-08-07:
Sorry for the issue everyone! The release is now available (https://github.com/espressif/esp-idf/releases/tag/v4.4.2) and the release archive will be available on the [mirror site](https://dl.espressif.com/github_assets/espressif/esp-idf/releases/download/v4.4.2/esp-idf-v4.4.2.zip) in a couple of hours.

We've taken a note not to update the release list used by VS Code extension until the release archive is published, in the future!

closed 2022-08-07 by igrr