espressif/esp-idf #9512: ESP venv prevents Python plugins from working

jonsmirl, 2022-08-06:
If I use 'source ./export.sh' Python plugins stop working (like gedit and eog). But instead if I go into the .espressif/.../python/bin directory run 'activate' the plugins stay working. It is annoying to have the editor plugins stop working if IDF environment is active. Also there is no way to deactivate the environment if started with 'source ./export.sh'

This is with 4.4.2 on Ubuntu Jaunty and Python3.  Example errors with 'source ./export.sh' active...
```
** (eog:11220): WARNING **: 11:44:54.811: Error initializing Python Plugin Loader: PyGObject initialization failed
ImportError: could not import gobject (error was: ModuleNotFoundError("No module named 'gi'"))

** (eog:11220): WARNING **: 11:44:54.811: Please check the installation of all the Python related packages required by libpeas and try again

(eog:11220): libpeas-WARNING **: 11:44:54.811: Loader 'python3' is not a valid PeasPluginLoader instance
```

dobairoland, 2022-08-06:
Hi @jonsmirl. Are you sourcing the script globally for the entirely system? The environment of ESP-IDF is not designed for that. One should open a new terminal, source it only there, run esp-idf commands.

jonsmirl, 2022-08-06:
I am in a conundrum, I have to source it globally to make the VSCode IDF plugin work. Which may be related to this:
https://github.com/espressif/vscode-esp-idf-extension/issues/775

dobairoland, 2022-08-07:
Please try to solve the root cause of setting up an environment with vscode. This workaround you've found is clearly not working and it is not supported. The purpose of our install & export scripts is  to be able to work in an isolated environment without effecting global settings. 

I've read your vscode report. Please provide more information so the vscode team could help you. The value of PATH set globally and from vscode terminal after the setup has been done would be a good start. The output of your `which python` is another useful information.

closed 2022-08-07 by dobairoland