espressif/esp-idf #9515: Add .component_hash to gitignore

dizcza, 2022-08-07:
I have the "espressif/esp_websocket_client" in the dependencies (managed components). The sources already contain the gitignore but `.component_hash` is missing there.

I cannot build the project on my two machines that have different versions of cmake.

igrr, 2022-08-08:
@dizcza just a question, do you have `managed_components` in your project's .gitignore file? I think adding that should cover both the components themselves and the .component_hash files created therein.

kumekay, 2022-08-08:
@dizcza What kind of issue do you have with different CMake versions? Content of `.component_hash` should not depend on CMake version.

We suggest to gitignore whole `managed_components` directory for projects and only store `dependencies.lock` in the repo.

dizcza, 2022-08-08:
>  Content of .component_hash should not depend on CMake version.

Perhaps, I misunderstood how the hash is generated.

> We suggest to gitignore whole managed_components directory for projects and only store dependencies.lock in the repo.

I see. All right, I'll add the directory to gitignore. Didn't find this in the docs.

closed 2022-08-08 by dizcza