espressif/esp-idf #9516: Increase the buffer size for esp_modem.

DarshanDodal, 2022-08-07:


Alvin1Zhang, 2022-08-08:
Thanks for your contribution.

david-cermak, 2022-08-19:
@DarshanDodal Could you please share more details to your request?

Are you talking about this component: https://github.com/espressif/esp-protocols/tree/master/components/esp_modem ? Which version and which IDF version you're using it with? Which buffer size you mean?

I believe that all buffers are configurable using the standard runtime config struct.

DarshanDodal, 2022-08-19:
Hello, 
Thanks for getting back.
I am using IDF v4.4.1 and yes I am speaking of "esp_modem".  Yes the buffer size is configurable, I was unaware of that earlier. But, I am facing an issue with the component. I am getting a warning " CMUX: Protocol mismatch: Missed leading SOF, recovering..." and the success rate of connection after this warning is 60%, 40% of times it does not connect. I have tried increasing the rx tx buffer, but still no luck.  And, even if connection happens, I get "Protocol Recovered" log very very rarely.

david-cermak, 2022-08-19:
Hi @DarshanDodal 

Could you please close this issue then and create a new one with a detailed description of the problem in https://github.com/espressif/esp-protocols/issues ?

I recall we had a similar issue, but it was resolved in https://github.com/espressif/esp-protocols/pull/53/commits/128c0a2d876e206e5e127df97090a9f2e828af43 (please double check that you included this commit before filing the issue)

closed 2022-08-20 by DarshanDodal

DarshanDodal, 2022-08-21:
Please check the issue here [#118](https://github.com/espressif/esp-protocols/issues/118)