espressif/esp-idf #9520: mcpwm_capture_channel_register_event_callbacks(322): install interrupt service for cap channel failed

dizcza, 2022-08-07:
## Environment
- Development Kit:  TTGO-T8-ESP32
- Kit version: v1.7
- Module or chip used:  ESP32
- IDF version: master branch
- Build System: idf.py
- xtensa-esp32-elf-gcc (crosstool-NG esp-2022r1-RC1) 11.2.0
- Operating System: Ubuntu 22.04
- Using an IDE?: No
- Power Supply: USB

## Problem Description

I'm migrating MCPWM capture mode from legacy to mcpwm_cap module using esp-idf master branch.
In my code, which I cannot show all, I
* Initialize WDT at startup
* mount an SD card
* install an I2C on any port
* initialize GPS uBlox and UART pattern detection similar to https://github.com/liebman/esp32-gps-ntp/blob/master/main/GPS.cpp
* initialize MCPWM capture

and the code fails with

```
E (00:00:02.317) mcpwm: mcpwm_capture_channel_register_event_callbacks(322): install interrupt service for cap channel failed
```

If I don't use WDT service OR don't mount an SD card OR don't initialize the I2C OR don't initialize GPS UART, `mcpwm_capture_channel_register_event_callbacks` works fine. As the return code suggests - `ESP_ERROR_CHECK`, - "No free interrupt found with the specified flags".

### Expected Behavior

There is at least one free interrupt with the MCPWM flags specified and `mcpwm_capture_channel_register_event_callbacks` succeeds.

### Actual Behavior

There are no free interrupts with the MCPWM flags specified and `mcpwm_capture_channel_register_event_callbacks` fails.

### Code to reproduce this issue

This is schematic code to perform GPS-RTC sync. It isn't meant to run.

```c

static void shutdown_callback() {
	ESP_LOGW(TAG, "SHUTDOWN");
}

static void eample_i2c_init()
{
    int intr_flag_disable = 0;
    int i2c_port = 0;

    /* I2C master doesn't need buffer */
    size_t i2c_master_rx_buf_disable = 0;
    size_t i2c_master_tx_buf_disable = 0;

    i2c_config_t conf = {
        .mode = I2C_MODE_MASTER,
        .sda_io_num = 33,
        .scl_io_num = 32,
        .sda_pullup_en = GPIO_PULLUP_DISABLE,
        .scl_pullup_en = GPIO_PULLUP_DISABLE,
        .master.clk_speed = 400000,  /*!< I2C master clock frequency */
    };

    ESP_ERROR_CHECK(i2c_param_config(i2c_port, &conf));

	ESP_ERROR_CHECK(
			i2c_driver_install(i2c_port, conf.mode,
					i2c_master_rx_buf_disable, i2c_master_tx_buf_disable,
[... 332 lines cut ...]

file: "./main/syncman.c" line 619
func: syncman_mcpwm_enable
expression: mcpwm_capture_channel_register_event_callbacks(shandle->channel_gps, &cbs, (void*) shandle)

abort() was called at PC 0x40088fcb on core 0
0x40088fcb: _esp_error_check_failed at /home/dizcza/tools/esp-idf/components/esp_system/esp_err.c:47

Backtrace: 0x40081fde:0x3ffc7f40 0x40088fd5:0x3ffc7f60 0x4008eade:0x3ffc7f80 0x40088fcb:0x3ffc7ff0 0x400d75d1:0x3ffc8020 0x400d7ae5:0x3ffc8060
0x40081fde: panic_abort at /home/dizcza/tools/esp-idf/components/esp_system/panic.c:412

0x40088fd5: esp_system_abort at /home/dizcza/tools/esp-idf/components/esp_system/esp_system.c:135

0x4008eade: abort at /home/dizcza/tools/esp-idf/components/newlib/abort.c:38

0x40088fcb: _esp_error_check_failed at /home/dizcza/tools/esp-idf/components/esp_system/esp_err.c:47

0x400d75d1: syncman_mcpwm_enable at /home/dizcza/Projects/Embedded/eclipse-workspace/esp32-sdpsensor/main/syncman.c:619

0x400d7ae5: syncman_task at /home/dizcza/Projects/Embedded/eclipse-workspace/esp32-sdpsensor/main/syncman.c:951
```

## Other items if possible

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/9277263/sdkconfig.txt)

suda-morris, 2022-08-08:
@dizcza Thanks for the detailed debug log, could you helo to check if the following fix can help?
```diff
diff --git a/components/driver/mcpwm/mcpwm_private.h b/components/driver/mcpwm/mcpwm_private.h
index b5881280f8..eb176fc29e 100644
--- a/components/driver/mcpwm/mcpwm_private.h
+++ b/components/driver/mcpwm/mcpwm_private.h
@@ -29,9 +29,9 @@ extern "C" {
 #endif

 #if CONFIG_MCPWM_ISR_IRAM_SAFE
-#define MCPWM_INTR_ALLOC_FLAG     (ESP_INTR_FLAG_SHARED | ESP_INTR_FLAG_INTRDISABLED | ESP_INTR_FLAG_IRAM)
+#define MCPWM_INTR_ALLOC_FLAG     (ESP_INTR_FLAG_LOWMED | ESP_INTR_FLAG_SHARED | ESP_INTR_FLAG_INTRDISABLED | ESP_INTR_FLAG_IRAM)
 #else
-#define MCPWM_INTR_ALLOC_FLAG     (ESP_INTR_FLAG_SHARED | ESP_INTR_FLAG_INTRDISABLED)
+#define MCPWM_INTR_ALLOC_FLAG     (ESP_INTR_FLAG_LOWMED | ESP_INTR_FLAG_SHARED | ESP_INTR_FLAG_INTRDISABLED)
 #endif

 #define MCPWM_PERIPH_CLOCK_PRE_SCALE (2)
```

dizcza, 2022-08-08:
> @dizcza Thanks for the detailed debug log, could you help to check if the following fix can help?
> 
> ```diff
> diff --git a/components/driver/mcpwm/mcpwm_private.h b/components/driver/mcpwm/mcpwm_private.h
> index b5881280f8..eb176fc29e 100644
> --- a/components/driver/mcpwm/mcpwm_private.h
> +++ b/components/driver/mcpwm/mcpwm_private.h
> @@ -29,9 +29,9 @@ extern "C" {
>  #endif
> 
>  #if CONFIG_MCPWM_ISR_IRAM_SAFE
> -#define MCPWM_INTR_ALLOC_FLAG     (ESP_INTR_FLAG_SHARED | ESP_INTR_FLAG_INTRDISABLED | ESP_INTR_FLAG_IRAM)
> +#define MCPWM_INTR_ALLOC_FLAG     (ESP_INTR_FLAG_LOWMED | ESP_INTR_FLAG_SHARED | ESP_INTR_FLAG_INTRDISABLED | ESP_INTR_FLAG_IRAM)
>  #else
> -#define MCPWM_INTR_ALLOC_FLAG     (ESP_INTR_FLAG_SHARED | ESP_INTR_FLAG_INTRDISABLED)
> +#define MCPWM_INTR_ALLOC_FLAG     (ESP_INTR_FLAG_LOWMED | ESP_INTR_FLAG_SHARED | ESP_INTR_FLAG_INTRDISABLED)
>  #endif
> 
>  #define MCPWM_PERIPH_CLOCK_PRE_SCALE (2)
> ```

Yes, the fix solves the issue. Checked for both options for the CONFIG_MCPWM_ISR_IRAM_SAFE.

closed 2022-08-16 by espressif-bot