espressif/esp-idf #9521: Calling esp_wifi_set_country_code("US", true) returns ESP_ERR_WIFI_NOT_STARTED

AxelLin, 2022-08-08:
## Environment

- Module or chip used: ESP32-WROOM-32D
- IDF version: v5.0-dev-4723-g30e8f19f5a
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2022r1-RC1) 11.2.0
- Operating System: Linux
- Power Supply: USB

## Problem Description

Exactly the same code works in v4.3 and v4.4.
But it returns ESP_ERR_WIFI_NOT_STARTED when compile with master tree.
So I think this is a regression.

davidzuhn, 2022-08-09:
I ran into this as well, and then changed the order of calls in my wifi setup, now calling esp_wifi_start first, and then esp_wifi_set_country_code, and it's now working for me.   It's definitely a change from 4.4, and I think at least warrants a note in the migration guide.   

But logically speaking, it seems like one ought to be able to set the country code before esp_wifi_start; that way there isn't a possible window where the (possibly incorrect) default country code configuration is in use.

MaxwellAlan, 2022-08-10:
Hi @AxelLin @davidzuhn 

Sorry for that, it's a breaking changes. This api now only can be used after wifi start.
We need to update related docs in migration guide.

Thanks.

AxelLin, 2022-08-10:
@MaxwellAlan 
It's a **BUG** rather than breaking change.
If you cannot set country code before esp_wifi_start, it means you have a small window that can violate regulatory before calling esp_wifi_set_country_code.

mhdong, 2022-08-17:
Hi @davidzuhn  @AxelLin The default country is "01" (world safe mode) , it will not violate regulatory before calling esp_wifi_set_country_code.

AxelLin, 2022-08-17:
> Hi @davidzuhn @AxelLin The default country is "01" (world safe mode) , it will not violate regulatory before calling esp_wifi_set_country_code.

Now you got another issue:
Calling esp_wifi_get_country_code() before esp_wifi_start always returns "US" rather than "01" in my test.

Besides, what's the reason to keep this BREAKAGE?
What's the problem to allow setting country code before esp_wifi_start?
Remember that people may compile the same application code with different IDF versions.
And you are forcing people to change application code.
IMHO, you should fix it if there is no good reason for the API breakage.

AxelLin, 2022-08-17:
> Hi @davidzuhn @AxelLin The default country is "01" (world safe mode) , it will not violate regulatory before calling esp_wifi_set_country_code.

On the second though, I think above comment is wrong.
If country code is "01", you cannot set channel 13/14(in JP).
So you must allow setting country code first before calling esp_wifi_start().

closed 2022-09-22 by AxelLin

AxelLin, 2022-09-28:
@MaxwellAlan @mhdong 

I did not hit error if calling esp_wifi_set_country_code() before esp_wifi_start() in v4.3 and v4.4.
But now I found there are backport fixes regarding country code for v4.3 and v4.4.
So what's exactly fixed for v4.x branches?