espressif/esp-idf #9522: gdb: unable to interrupt gdb on Windows

xobs, 2022-08-08:
On Windows, it is no longer possible to break into GDB. Ordinarily, `Ctrl-C` would do this, however this does not appear to work anymore. Note that console support in gdb appears to be slightly broken, because e.g. the backspace key adds a space rather than deleting the previous character.

## Steps to reproduce

1. Run openocd. I'm using an ESP32S3 with built-in JTAG:

```
$ openocd -f board/esp32s3-builtin.cfg
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: serial (F4:12:FA:47:27:CC)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32s3.cpu0: Detected debug stubs @ 3fca9770
Info : starting gdb server for esp32s3.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
```

2. Connect GDB, and continue execution:

```
$ > xtensa-esp32s3-elf-gdb.exe .\build\blackmagic.elf -ex 'tar ext :3333' -ex 'c'
GNU gdb (esp-gdb) 11.2_20220729
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-w64-mingw32 --target=xtensa-esp-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from .\build\blackmagic.elf...
Remote debugging using :3333
spinlock_acquire (timeout=<optimized out>, lock=0x3fced1bc) at C:/Users/Sean/esp/esp-idf/components/esp_hw_support/include/spinlock.h:123
123             lock->count++;  // Finally, we increment the lock count
Continuing.
```

3. Attempt to break, using any common key combination (e.g. Ctrl-C, Ctrl-[, etc.)

The only key combination that will work is Ctrl-Break, which terminates GDB.

This is happening both in Powershell, as well as in cmd.exe.

(Note separately, that when running xtensa-esp32s3-elf-gdb, a dialog box pops up saying "The application was unable to start correctly (0xc0000022), and it then starts up normally. I suspect this is a holdover from #9379)

xobs, 2022-08-08:
I normally use Windows Terminal, however I've verified that this also happens in the legacy `conhost.exe` console.

Lapshin, 2022-08-08:
@xobs Thank you for the issue report!

Please show your `winver` (run in terminal)

xobs, 2022-08-08:
I'm running:

Windows 21H2 (OS Build 22000.795)

Lapshin, 2022-08-09:
@xobs , the new GDB release is available https://github.com/espressif/binutils-gdb/releases/tag/esp-gdb-v11.2_20220808

IDF will be updated with it soon

xobs, 2022-08-11:
I can confirm that 20220808 fixes the "interrupt" issue.

Should I open a separate issue for the "Backspace does not work" issue, which I feel is somewhat related?

Lapshin, 2022-08-11:
> Should I open a separate issue for the "Backspace does not work" issue, which I feel is somewhat related?

Feel free to open a new issue

closed 2022-08-16 by espressif-bot

Lapshin, 2022-08-24:
@xobs , FYI, the issue for "Backspace does not work" is fixed in the [latest GDB release](https://github.com/espressif/binutils-gdb/releases/tag/esp-gdb-v11.2_20220823). The release will be added to IDF's (v5.0 and master) tools soon

xobs, 2022-08-24:
Fantastic, thanks!