espressif/esp-idf #9526: Add Bacnet Protocol Support Library For ESP-IDF

maialbe2823, 2022-08-08:
Would be awesome if we could get BACnet protocol working on ESP-IDF, as the bacnetstack doesnt work on ESP-IDF.

david-cermak, 2022-08-19:
@maialbe2823 Have you checked their esp32 port layer?

https://github.com/bacnet-stack/bacnet-stack/tree/master/ports/esp32

Any trouble building that or making it work? What exactly does not work?

maialbe2823, 2022-08-20:
It doesn’t start the BACnet protocol as it keeps stuck in the wifi return 5. However; it is connected to wifi.

david-cermak, 2022-08-22:
So if the issue is connecting to WiFi correctly, we can help with that. Could you please describe which API exactly failed and what have you tried?

> It doesn’t start the BACnet protocol as it keeps stuck in the wifi return 5

I can only guess that this line

https://github.com/bacnet-stack/bacnet-stack/blob/c8671d54f719bbb616a4a125f2324640822f44c7/ports/esp32/src/main.c#L102

never gets executed since the `event->id` is set to 5? i.e. `SYSTEM_EVENT_STA_DISCONNECTED` ? But you're saying:

> However; it is connected to wifi.

Could you please also share the version of IDF you're using?

The general idea is to look at their port [init code](https://github.com/bacnet-stack/bacnet-stack/blob/master/ports/esp32/src/main.c)
and our basic example of [WiFi station init](https://github.com/espressif/esp-idf/blob/release/v4.4/examples/wifi/getting_started/station/main/station_example_main.c) for your preferred IDF version (`v4.4` branch in the link) and come up with your specific initialization.

maialbe2823, 2022-08-22:
My IDF version is 4.4. Im trying to run this sample from the ESP-IDF, however I get the following error:
Executing action: all (aliases: build)
Running cmake in directory c:\espressif\frameworks\esp-idf-v4.4.1\examples\bacnet-stack\ports\esp32\src\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=1 c:\espressif\frameworks\esp-idf-v4.4.1\examples\bacnet-stack\ports\esp32\src"...
-- Found Git: C:/Espressif/tools/idf-git/2.34.2/cmd/git.exe (found version "2.34.1.windows.1")
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file C:/Espressif/frameworks/esp-idf-v4.4.1/examples/bacnet-stack/ports/esp32/src/sdkconfig
-- Found Python3: C:/Python310/python.exe (found version "3.10.2") found components: Interpreter
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "bacnet-stack" version: bacnet-stack-1.0.0-169-g319a1df
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.1/examples/bacnet-stack/ports/esp32/src/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.1/components/soc/esp32/ld/esp32.peripherals.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor cmock coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_lcd esp_local_ctrl esp_netif esp_phy esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap idf_test ieee802154 jsmn json libsodium log lwip mbedtls mdns mqtt newlib nghttp nvs_flash openssl openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/Espressif/frameworks/esp-idf-v4.4.1/components/app_trace C:/Espressif/frameworks/esp-idf-v4.4.1/components/app_update C:/Espressif/frameworks/esp-idf-v4.4.1/components/asio C:/Espressif/frameworks/esp-idf-v4.4.1/components/bootloader C:/Espressif/frameworks/esp-idf-v4.4.1/components/bootloader_support C:/Espressif/frameworks/esp-idf-v4.4.1/components/bt C:/Espressif/frameworks/esp-idf-v4.4.1/components/cbor C:/Espressif/frameworks/esp-idf-v4.4.1/components/cmock C:/Espressif/frameworks/esp-idf-v4.4.1/components/coap C:/Espressif/frameworks/esp-idf-v4.4.1/components/console C:/Espressif/frameworks/esp-idf-v4.4.1/components/cxx C:/Espressif/frameworks/esp-idf-v4.4.1/components/driver C:/Espressif/frameworks/esp-idf-v4.4.1/components/efuse C:/Espressif/frameworks/esp-idf-v4.4.1/components/esp-tls C:/Espressif/frameworks/esp-idf-v4.4.1/components/esp32 C:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_adc_cal C:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_common C:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_eth C:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_event C:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_gdbstub C:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_hid C:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_http_client C:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_http_server C:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_https_ota C:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_https_server C:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_hw_support C:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_ipc C:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_lcd C:/ [... cut ...]
-- Configuring incomplete, errors occurred!
See also "C:/Espressif/frameworks/esp-idf-v4.4.1/examples/bacnet-stack/ports/esp32/src/build/CMakeFiles/CMakeOutput.log".
See also "C:/Espressif/frameworks/esp-idf-v4.4.1/examples/bacnet-stack/ports/esp32/src/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1

I have try lots of ways of getting this bacnet sample to run on ESP32 but no luck. Either I get this error or through platform.io I get the Wifi stuck event which makes no sense since ESP32 does get connected to my wifi.

david-cermak, 2022-09-19:
@maialbe2823 I've created a simple example based on their port layer: https://github.com/david-cermak/esp-network-examples/tree/main/backnet-sever
It's not tested as I don't have any HW that supports it, but builds okay with IDF 4.x and 5.x.

david-cermak, 2022-10-21:
@maialbe2823 Any update? Have you checked the component above?

I'm afraid we cannot maintain a separate Bacnet port and library in IDF, but we can help adapting the layers and connecting to network, if someone wanted to maintain this as a component, for example in https://components.espressif.com/components

maialbe2823, 2022-11-13:
Hi David, the build works. However, I got one question, where does  line 225 if (Binary_Output_Present_Value(0) == BINARY_ACTIVE) comes from? What is the instance number of that binary output and why if I add more bacnet analog input or outputs it just doesnt read them. Thank you.

david-cermak, 2022-11-14:
Hi @maialbe2823 

I'm sorry, I have no knowledge on the protocol itself. This line is a plain copy-paste from https://github.com/bacnet-stack/bacnet-stack/blob/e3fab4a96c5bec179051c2166c6bf1758830ad44/bacnet-stack/ports/esp32/src/main.c. You can check the differences by looking at [this](https://github.com/david-cermak/esp-network-examples/commit/ec25e6b24e86cf23d3d47f16c4348eb81ad7da85) commit. As you can see, I've only updated the initialization of esp-wifi, esp-event and netif components to comply with IDF v4.1 and later.

As for the analogue inputs and outputs, my guess is that they are defined in the form of an object dictionary and it is specific to the device. The definition of this project directly includes the original demo application of Bacnet port layer: [device.h](https://github.com/bacnet-stack/bacnet-stack/blob/e3fab4a96c5bec179051c2166c6bf1758830ad44/bacnet-stack/ports/esp32/src/device.h)

maialbe2823, 2022-11-14:
Ok, I'll check it out. One more thing is it possible to change the esp-wifi, events and netif to be compatible with ethernet. Cuz I have tried to but no luck on it.

david-cermak, 2022-11-15:
Added in https://github.com/david-cermak/esp-network-examples/commit/b784ebd134bc0fcd92f07bf1c5d83ca6a2622702 (see the implementation in [eth_init.c](https://github.com/david-cermak/esp-network-examples/blob/main/backnet-sever/main/eth_init.c), you can simply switch between Ethernet and WiFi station in the project makefile.

Please note that it uses IP101 (standard Epsressif's Ethernet Kit) with ESP32. Also the initialization is a little different in `v4.x` (this works for `v5.0` and later). For more details and support for other devices please check:
* https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_eth.html#ethernet
* https://github.com/espressif/esp-idf/tree/769ec42899/examples/ethernet/basic

maialbe2823, 2022-11-15:
Thanks so much, will try it out.

maialbe2823, 2022-12-16:
Hi David, iI tried it out in the ESP_IDF V5.0 but I get an error with the CONNECTED BIT constant something that doesnt happen in v4.4.
Here is the log:
C:\Espressif\frameworks\esp-idf-v5.0\examples\BACnet_ETH_V5>idf.py build
Executing action: all (aliases: build)
Running ninja in directory c:\espressif\frameworks\esp-idf-v5.0\examples\bacnet_eth_v5\build
Executing "ninja all"...
[1/1] cmd.exe /C "cd /D C:\Espressif\frameworks\esp-idf-v5.0\examples\BACnet_ETH_V5\build\bootloader\esp-idf\esptool_...0x8000 bootloader 0x1000 C:/Espressif/frameworks/esp-idf-v5.0/examples/BACnet_ETH_V5/build/bootloader/bootloader.bin"Bootloader binary size 0x6710 bytes. 0x8f0 bytes (8%) free.
[3/13] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/port/ai.c.objFAILED: esp-idf/main/CMakeFiles/__idf_main.dir/port/ai.c.obj
ccache C:\Espressif\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -IC:/Espressif/frameworks/esp-idf-v5.0/examples/BACnet_ETH_V5/build/config -IC:/Espressif/frameworks/esp-idf-v5.0/examples/BACnet_ETH_V5/main/include -IC:/Espressif/frameworks/esp-idf-v5.0/examples/BACnet_ETH_V5/main/port/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/newlib/platform_include -IC:/Espressif/frameworks/esp-idf-v5.0/components/freertos/FreeRTOS-Kernel/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/freertos/esp_additions/include/freertos -IC:/Espressif/frameworks/esp-idf-v5.0/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/freertos/esp_additions/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_hw_support/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_hw_support/include/soc -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_hw_support/include/soc/esp32 -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_hw_support/port/esp32/. -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_hw_support/port/esp32/private_include -IC:/Espressif/frameworks/esp-idf-v5.0/components/heap/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/log/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/soc/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/soc/esp32/. -IC:/Espressif/frameworks/esp-idf-v5.0/components/soc/esp32/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/hal/esp32/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/hal/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/hal/platform_port/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_rom/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_rom/include/esp32 -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_rom/esp32 -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_common/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_system/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_system/port/soc -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_system/port/include/private -IC:/Espressif/frameworks/esp-idf-v5.0/components/xtensa/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/xtensa/esp32/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/lwip/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/lwip/include/apps -IC:/Espressif/frameworks/esp-idf-v5.0/components/lwip/include/apps/sntp -IC:/Espressif/frameworks/esp-idf-v5.0/components/lwip/lwip/src/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/lwip/port/esp32/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/lwip/port/esp32/include/arch -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_ringbuf/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/efuse/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/efuse/esp32/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/driver/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/driver/deprecated -IC:/Espressif/frameworks/esp-idf-v5.0/components/driver/esp32/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_pm/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/mbedtls/port/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/mbedtls/mbedtls/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/mbedtls/mbedtls/library -IC:/Espressif/frameworks/esp-idf-v5.0/components/mbedtls/esp_crt_bundle/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_app_format/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/bootloader_support/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/bootloader_support/bootloader_flash/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_partition/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/app_update/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/spi_flash/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/pthread/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_timer/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/app_trace/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_event/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/nvs_flash/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_phy/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_phy/esp32/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/vfs/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_netif/include -IC:/Espressif/frameworks/esp-id [... cut ...]
HINT: The issue is better to resolve by replacing format specifiers to 'PRI'-family macros (include <inttypes.h> header file).
ninja failed with exit code 1, output of the command is in the c:\espressif\frameworks\esp-idf-v5.0\examples\bacnet_eth_v5\build\log\idf_py_stderr_output_26716 and c:\espressif\frameworks\esp-idf-v5.0\examples\bacnet_eth_v5\build\log\idf_py_stdout_output_26716

I cannt seem to fix it as it thinks it is undeclared something that is not correct as indeed he is declared in network_init.h.
Please let me know how can I fix it. Thanks once again.

david-cermak, 2022-12-17:
Hi @maialbe2823 

The first issue (warning treated as error) is formatting, you can easily fix/ignore it with
```cmake
target_compile_options(${COMPONENT_LIB} PRIVATE "-Wno-format")
```
(or by following the steps in the *hints* provided by `idf.py`)

The second issue is related to `BIT0` definition, so you can try to comment the macro below?

https://github.com/david-cermak/esp-network-examples/blob/b784ebd134bc0fcd92f07bf1c5d83ca6a2622702/backnet-sever/main/net.h#L3-L4 

This was added, since IDF defines the same macros in `components/esp_common/include/esp_bit_defs.h`

Interestingly, I don't see any of the errors above when I try to compile the project against latest `release/v5.0`.
Which IDF commit are you exactly on? Haven't you modified the example? Are you building with IDF build system or via some framework? (arduino, platformio, ?)

maialbe2823, 2022-12-18:
I use ESP-IDF 5.0. I'll try this changes and see if it works. Thanks

david-cermak, 2023-01-02:
You can try to increase number of Tx buffers in [CONFIG_ETH_DMA_TX_BUFFER_NUM](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-eth-dma-tx-buffer-num)](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-eth-dma-tx-buffer-num), from 10 to say 16?

This error appears from time to time if the traffic is too heavy, if the Ethernet driver cannot handle sending packets accumulated in lwip. Adjusting task priorities might help as well, sometimes?

Alvin1Zhang, 2023-02-16:
@maialbe2823 Thanks for reporting, would you please help share if any further updates for the issue? Thanks.

maialbe2823, 2023-02-16:
@Alvin1Zhang @david-cermak I Have it working completely and it is stable and it works fast something I thought it could be an problem.
However, I am now trying to make it MS/TP portable but ESP32 is giving a hard time with the RS485 communication and max485. Does someone has any guide on how to tackle it?
I have follow every examples for RS485 communication on esp-idf but none work. I think the ESP32 is not getting or seeing my max485.
Thanks.

david-cermak, 2023-02-16:
@maialbe2823 Could you please close this issue then? And perhaps open another one, related to RS485?

RS485 (AFAICT) communication is being continuously tested on all supported chips and should work without any problem. But if you're not able to get a simple IDF example working, please ask on forum or create an issue, so the IDF support engineers responsible for RS485 will get back to you with help.
(Just note that the ESP32 won't *detect* an MAX485 or similar, it's a simple tranceiver and you just have to drive it)

closed 2023-02-16 by maialbe2823

lukedukeus, 2023-02-28:
I am trying the example here: https://github.com/david-cermak/esp-network-examples

I don't have an esp with a network jack, but trying to get it to work over wifi. It connects to the network ok, but I cannot find it with yabe. I am running IDF v5.0. 

@maialbe2823 were you able to get this working?

maialbe2823, 2023-02-28:
I was able to make it work with Wifi, but I’m currently trying to make it work with ETH plus with RS485. The latest is giving a hard time to send data through uart though.Any advices when it comes to using the esp rs85 communication?Once again thanks.

lukedukeus, 2023-02-28:
Can you publish the code that you have working over wifi?

lukedukeus, 2023-02-28:
For using bacnet with an max485, see this example: https://github.com/KaiDroste/bacnet-stack-esp32/tree/master/ports/esp32_mstp

It builds ok, but I do not have a 485 network to test it

maialbe2823, 2023-02-28:
Yeah, I’ll publish it.That mstp example I have tried it but it just doesn’t work. As it just gives an error that there is not data available. I get to see data now but it doesn’t receive it the right way. Plus the transmitting part just doesn’t work. That’s where I am at now.

lukedukeus, 2023-03-01:
@maialbe2823 link?

lukedukeus, 2023-03-04:
@maialbe2823 Can you send me your source code, or at least your implementation of bip.c?

maialbe2823, 2023-03-07:
@lukedukeus Check this link out: https://github.com/maialbe2823/BACnet-Bip/blob/main/bip_init_old_wifi.c

lukedukeus, 2023-03-07:
@maialbe2823 I wasn't able to get that working, can you post the full project?

Thanks!

maialbe2823, 2023-03-07:
@lukedukeus check this one out: https://github.com/b1b2b3b4b5b6/esp32-bacnet

lukedukeus, 2023-03-07:
@maialbe2823 
I tried that, but it doesnt show up on my network. Were you able to use it as is, or did you have to make any changes?

maialbe2823, 2023-03-07:
Which explorer are you using? Cuz I’m using Yabe and it is working great.

lukedukeus, 2023-03-07:
@maialbe2823 I am using yabe as well, what version of esp-idf are you using?

maialbe2823, 2023-03-07:
ESP-IDF 4.4

lukedukeus, 2023-03-07:
@maialbe2823  Attempting to compile it with v4.4 on windows for esp32, I get the following error. Did you get this as well?

FAILED: app-template.elf
cmd.exe /C "cd . && C:\Expressif\tools\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe -mlongcalls -Wno-frame-address  @CMakeFiles\app-template.elf.rsp -o app-template.elf  && cd ."
c:/expressif/tools/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/freertos/libfreertos.a(port_common.c.obj):(.literal.main_task+0x1c): undefined reference to `app_main'      
c:/expressif/tools/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/freertos/libfreertos.a(port_common.c.obj): in function `main_task':
C:/Expressif/esp-idf/components/freertos/port/port_common.c:135: undefined reference to `app_main'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

lukedukeus, 2023-03-08:
@maialbe2823 I got it working! I took the code, and ported it to idf v5.0, and made some small changes with the networking, and it works. Thank you so much.

Link: https://github.com/lukedukeus/esp32-bacnet-master-v5.01

maialbe2823, 2023-03-08:
Glad it work. I’m now working on getting mstp however esp32 user is giving a hard time.

maialbe2823, 2023-03-08:
@lukedukeus Do you have any advice or documents for the esp32 UARTs? I am trying to make that work.

lukedukeus, 2023-03-08:
@maialbe2823 Not much more than what is listed here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/uart.html

If you have an osiciliscope or digital analyser, you could hook it up to your 485 network, then create a minimal example to read data from the uart, and confirm that it is the same on the wire. This way you could isolate the problem to software by knowing that your transiever can speak at the correct baud rate, and there are no buffering errors. It might help to try different hardware configurations if you think your software is working correctly.

There are also rs485 specific uart options: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/uart.html

lukedukeus, 2023-03-08:
@maialbe2823 Also, curious to know how you are taking care of collision detection. You have to always be listening for data, but you cant when you want to send data back out. Can you share the code you are working on, and Ill try it on my hardware?

maialbe2823, 2023-03-08:
@lukedukeus This is what I am using https://github.com/KaiDroste/bacnet-stack-esp32/tree/mstp-developement/ports/esp32_mstp
Refer to dlmstp.c and rs485.c