espressif/esp-idf #9527: Can't build sample project

grzegorzniemirowski, 2022-08-08:
## Environment

- Module or chip used:  ESP32-C3
- IDF version: 4.4.2
- Build System: CMake
- Operating System: Windows
- Using an IDE: VSCode with IDF Plugin

## Problem Description

I'm trying to start development for ESP32-C3. To make it happen I have installed VSCode, the IDF plugin and latest ESP-IDF (4.4.2) using the plugin. Then I selected sample project (blink) and set the target to ESP32-C3 built in. Then I try to build the project but I'm getting strange error:

`cannot open file:///c%3A/Users/Grzegorz/Documents/src/ESP32/blink/esp32c3-SDK-main/mdk-main/tools/msys2_path.txt. Detail: Unable to read file 'c:\Users\Grzegorz\Documents\src\ESP32\blink\esp32c3-SDK-main\mdk-main\tools\msys2_path.txt' (Error: Unable to resolve nonexistent file 'c:\Users\Grzegorz\Documents\src\ESP32\blink\esp32c3-SDK-main\mdk-main\tools\msys2_path.txt')`

Why the build system looks for `esp32c3-SDK-main`? It's not present in sample project folder. How can I solve it? I tried reinstalling ESP-IDF and Tools many times and generating the sample project. What am I doing wrong? I jutr try to build simplest example and I don't configure anything manually. Is this IDF Plugin bug? There is also some CMake error.

[image]

grzegorzniemirowski, 2022-08-08:
Removing CMake extension makes CMake error go away but missing file error is still there. This error is shown just at the start of building process.

brianignacio5, 2022-08-08:
Yes as far as I can see the user is trying to build with the CMake extension instead of the ESP-IDF extension build command.

Please take a look at the tutorials for ESP-IDF extension in [here](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/toc.md) and [available commands](https://github.com/espressif/vscode-esp-idf-extension#available-commands)

grzegorzniemirowski, 2022-08-08:
Thanks @brianignacio5 . I thought that clicking Build in the blue toolbar at bottom of the VSCode window launches ESP-IDF extension build. It's misleading then.

brianignacio5, 2022-08-09:
It's understandable @grzegorzniemirowski . The thing is that Build in the blue toolbar is contributed by the CMake extension. Our extension has a cylinder icon which shows `ESP-IDF Build Project` on hover.

closed 2022-08-09 by grzegorzniemirowski