espressif/esp-idf #9528: sourcing export.sh doesn't override CMake path

Sponge5, 2022-08-08:
Expected behavior:

```
$ source esp-idf/export.sh
$ which cmake
<home>/.espressif/tools/cmake/3.16.4/bin/cmake
```

Actual behavior:

```
$ which cmake
/usr/bin/cmake
```

And when I remove my CMake installation, the binary isn't found.

Tested with `release/v4.1`, `release/v4.4` and `master`.

I believe this isn't correct behavior, since `ninja` is correctly sourced?

Sponge5, 2022-08-08:
Thanks for the speedy reply! Yes, I'm on Linux. What's weird is that I haven't touched any configuration, I'm building esp-idf from source and it installed both CMake and Ninja in my home directory (``.espressif/tools``), but sourced only ``ninja``. So the bug would be the other way around.

I don't know which version did that, I just tested by removing .espressif folder and it didn't happen after clean ``./install.sh`` for ``v4.4``. I'd call this issue resolved then.

But before you close it, could you please point the next person to the specific part of the documentation, where it says CMake is optional for Linux, I can't find it..

igrr, 2022-08-08:
The recommendation to install cmake and ninja via the OS package manager is reflected in the getting started guide: https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/get-started/linux-setup.html#install-prerequisites

The status (optional/required) of cmake tool can be found here: https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-guides/tools/idf-tools.html#cmake

and ninja here: https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-guides/tools/idf-tools.html#ninja

igrr, 2022-08-08:
> What's weird is that I haven't touched any configuration, I'm building esp-idf from source and it installed both CMake and Ninja in my home directory (.espressif/tools), but sourced only ninja. So the bug would be the other way around.

If you happen to reproduce this again, please attach the output of `idf_tools.py list`, the contents of `~/.espressif/idf-env.json` and the output of `tree -L 2 ~/.espressif/tools`. We'll try to figure out why this happened.

closed 2022-08-08 by igrr