espressif/esp-idf #9532: M1 max芯片 MacBook Pro 无法正常编译项目

Phantom2000X, 2022-08-08:
## Environment

- Development Kit:      [ESP32-S3-WROOM-1]
- Kit version (for WroverKit/PicoKit/DevKitC): [none]
- Module or chip used:  [ESP32-S3-WROOM-1]
- IDF version (run ``git describe --tags`` to find it):
    // v4.4.2
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    //xtensa-esp32s3-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     [macOS]
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

已经参照官方文档完成环境配置https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32s3/get-started/linux-macos-setup.html，master分支和4.4.2都试过，编译hello_world样例的时候会出现以下错误：

`FAILED: esp-idf/mbedtls/x509_crt_bundle /Users/phantomx/esp/hello_world/build/esp-idf/mbedtls/x509_crt_bundle
cd /Users/phantomx/esp/hello_world/build/esp-idf/mbedtls && /Users/phantomx/.espressif/python_env/idf4.4_py3.10_env/bin/python /Users/phantomx/esp/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py --input /Users/phantomx/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem /Users/phantomx/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_local.pem -q
Traceback (most recent call last):
  File "/Users/phantomx/esp/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py", line 224, in <module>
    main()
  File "/Users/phantomx/esp/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py", line 208, in main
    bundle.add_from_file(path)
  File "/Users/phantomx/esp/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py", line 86, in add_from_file
    self.add_from_pem(crt_str)
  File "/Users/phantomx/esp/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py", line 116, in add_from_pem
    self.certificates.append(x509.load_pem_x509_certificate(crt.encode(), default_backend()))
  File "/Users/phantomx/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/cryptography/hazmat/backends/__init__.py", line 8, in default_backend
    from cryptography.hazmat.backends.openssl.backend import backend
  File "/Users/phantomx/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/cryptography/hazmat/backends/openssl/__init__.py", line 6, in <module>
    from cryptography.hazmat.backends.openssl.backend import backend
  File "/Users/phantomx/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/cryptography/hazmat/backends/openssl/backend.py", line 63, in <module>
    from cryptography.hazmat.bindings.openssl import binding
  File "/Users/phantomx/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/cryptography/hazmat/bindings/openssl/binding.py", line 14, in <module>
    from cryptography.hazmat.bindings._openssl import ffi, lib
ImportError: dlopen(/Users/phantomx/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/_cffi_backend.cpython-310-darwin.so, 0x0002): tried: '/Users/phantomx/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/_cffi_backend.cpython-310-darwin.so' (mach-o file, but is an incompatible architecture (have (arm64), need (x86_64)))
ninja: build stopped: subcommand failed.
ninja failed with exit code 1`

### Expected Behavior

### Actual Behavior

### Steps to reproduce

full according to the official guide:https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32s3/get-started/linux-macos-setup.html#

kumekay, 2022-08-08:
@Phantom2000X how did you install python 3.10 on your MacBook?

Could you please try to run `python` after `. ./export.sh` and then:

```python

import platform
platform.platform()

```

Phantom2000X, 2022-08-09:
> @Phantom2000X how did you install python 3.10 on your MacBook?
> 
> Could you please try to run `python` after `. ./export.sh` and then:
> 
> ```python
> import platform
> platform.platform()
> ```

I have got these after run the comment you mention:
`❯ . ./export.sh
Setting IDF_PATH to '/Users/phantomx/esp/esp-idf'
Detecting the Python interpreter
Checking "python" ...
/Users/phantomx/esp/esp-idf/tools/detect_python.sh:16: command not found: python
Checking "python3" ...
Python 3.10.4
"python3" has been detected
Adding ESP-IDF tools to PATH...
Using Python interpreter in /Users/phantomx/.espressif/python_env/idf4.4_py3.10_env/bin/python
Checking if Python packages are up to date...
Python requirements from /Users/phantomx/esp/esp-idf/requirements.txt are satisfied.
Updated PATH variable:
  /Users/phantomx/esp/esp-idf/components/esptool_py/esptool:/Users/phantomx/esp/esp-idf/components/espcoredump:/Users/phantomx/esp/esp-idf/components/partition_table:/Users/phantomx/esp/esp-idf/components/app_update:/Users/phantomx/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin:/Users/phantomx/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin:/Users/phantomx/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/Users/phantomx/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220411/openocd-esp32/bin:/Users/phantomx/.espressif/python_env/idf4.4_py3.10_env/bin:/Users/phantomx/esp/esp-idf/tools:/Users/phantomx/.rvm/gems/ruby-2.3.0/bin:/Users/phantomx/.rvm/gems/ruby-2.3.0@global/bin:/Users/phantomx/.rvm/rubies/ruby-2.3.0/bin:/Users/phantomx/.poetry/bin:/Users/phantomx/flutter/bin:/Library/Frameworks/Python.framework/Versions/3.10/bin:/Library/Frameworks/Python.framework/Versions/3.9/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Applications/VMware Fusion.app/Contents/Public:/Library/Apple/usr/bin:/Users/phantomx/.hikit/bin:/Users/phantomx/.local/bin:/Users/phantomx/.rvm/bin
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

❯ python
Python 3.10.4 (v3.10.4:9d38120e33, Mar 23 2022, 17:29:05) [Clang 13.0.0 (clang-1300.0.29.30)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import platform
>>> platform.platform()
'macOS-12.5-arm64-arm-64bit'
>>>`

Phantom2000X, 2022-08-09:
> 

> @Phantom2000X how did you install python 3.10 on your MacBook?
> 
> Could you please try to run `python` after `. ./export.sh` and then:
> 
> ```python
> import platform
> platform.platform()
> ```

I install python3.10 by python official installer，https://www.python.org/downloads/release/python-3106/

kumekay, 2022-08-09:
@Phantom2000X, thank you for the details, I tried to reproduce your issue on a MacBook with M1 pro, but unfortunately, I wasn't able to reproduce your issue. 

It looks like somehow you got `cffi` package for x64 instead of arm64,
could you please try to 

run `pip uninstall  cffi cryptography -y  && pip install cryptography==36.0.2` in your idf environment. If it install's `cffi-1.15.1-cp310-cp310-macosx_11_0_arm64.whl`, then it worth retrying the build.

Phantom2000X, 2022-08-09:
> @Phantom2000X, thank you for the details, I tried to reproduce your issue on a MacBook with M1 pro, but unfortunately, I wasn't able to reproduce your issue.
>
> It looks like somehow you got `cffi` [...]

I have try this way,but it install's`cffi-1.15.1-cp310-cp310-macosx_11_0_arm64.whl`,but also build with same fail.
In the log it show that  (have (arm64), need (x86_64)) may means python have run in right arm64 but other's run in x86?
How can I know other module run in x86 or arm64?

dobairoland, 2022-08-09:
Maybe the following would be also worth to try:
```sh
pip uninstall cffi cryptography -y && pip install cryptography==36.0.2 cffi==1.15.0
```

Phantom2000X, 2022-08-09:
> Maybe the following would be also worth to try:
> 
> ```shell
> pip uninstall cffi cryptography -y && pip install cryptography==36.0.2 cffi==1.15.0
> ```

I have try these but it also fail.

pip uninstall cffi cryptography -y && pip install cryptography==36.0.2 cffi==1.15.0
Found existing installation: cffi 1.15.1
Uninstalling cffi-1.15.1:
  Successfully uninstalled cffi-1.15.1
Found existing installation: cryptography 37.0.4
Uninstalling cryptography-37.0.4:
  Successfully uninstalled cryptography-37.0.4
Collecting cryptography==36.0.2
  Using cached cryptography-36.0.2-cp36-abi3-macosx_10_10_universal2.whl (4.7 MB)
Collecting cffi==1.15.0
  Downloading cffi-1.15.0-cp310-cp310-macosx_11_0_arm64.whl (173 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 174.0/174.0 kB 1.4 MB/s eta 0:00:00
Requirement already satisfied: pycparser in /Users/phantomx/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages (from cffi==1.15.0) (2.21)
Installing collected packages: cffi, cryptography
Successfully installed cffi-1.15.0 cryptography-36.0.2

[notice] A new release of pip available: 22.2.1 -> 22.2.2
[notice] To update, run: pip install --upgrade pip
❯ idf.py build
Executing action: all (aliases: build)
Running ninja in directory /Users/phantomx/esp/hello_world/build
Executing "ninja all"...
[1/380] Performing build step for 'bootloader'
[1/1] cd /Users/phantomx/esp/hello_world/build/bootloader/esp-idf/esptool_py && /Users/phantomx/.espressif/python_env/idf4.4_py3.10_env/bin/python /Users/phantomx/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 /Users/phantomx/esp/hello_world/build/bootloader/bootloader.bin
Bootloader binary size 0x51f0 bytes. 0x2e10 bytes (36%) free.
[2/378] Generating x509_crt_bundle
FAILED: esp-idf/mbedtls/x509_crt_bundle /Users/phantomx/esp/hello_world/build/esp-idf/mbedtls/x509_crt_bundle
cd /Users/phantomx/esp/hello_world/build/esp-idf/mbedtls && /Users/phantomx/.espressif/python_env/idf4.4_py3.10_env/bin/python /Users/phantomx/esp/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py --input /Users/phantomx/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem /Users/phantomx/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_local.pem -q
Traceback (most recent call last):
  File "/Users/phantomx/esp/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py", line 224, in <module>
    main()
  File "/Users/phantomx/esp/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py", line 208, in main
    bundle.add_from_file(path)
  File "/Users/phantomx/esp/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py", line 86, in add_from_file
    self.add_from_pem(crt_str)
  File "/Users/phantomx/esp/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py", line 116, in add_from_pem
    self.certificates.append(x509.load_pem_x509_certificate(crt.encode(), default_backend()))
  File "/Users/phantomx/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/cryptography/hazmat/backends/__init__.py", line 9, in default_backend
    from cryptography.hazmat.backends.openssl.backend import backend
  File "/Users/phantomx/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/cryptography/hazmat/backends/openssl/__init__.py", line 6, in <module>
    from cryptography.hazmat.backends.openssl.backend import backend
  File "/Users/phantomx/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/cryptography/hazmat/backends/openssl/backend.py", line 64, in <module>
    from cryptography.hazmat.bindings.openssl import binding
  File "/Users/phantomx/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/cryptography/hazmat/bindings/openssl/binding.py", line 14, in <module>
    from cryptography.hazmat.bindings._openssl import ffi, lib
ImportError: dlopen(/Users/phantomx/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/_cffi_backend.cpython-310-darwin.so, 0x0002): tried: '/Users/phantomx/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/_cffi_backend.cpython-310-darwin.so' (mach-o file, but is an incompatible architecture (have (arm64), need (x86_64)))
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

kumekay, 2022-08-10:
@Phantom2000X It still looks like you are running python for the wrong architecture. What it the terminal app you are using?
What is output of `arch` command in your terminal?
If it's `i386` 
 Please make sure that `Open using Rosetta` is unchecked in the terminal app info

[image]

Phantom2000X, 2022-08-10:
> @Phantom2000X It still looks like you are running python for the wrong architecture. What it the terminal app you are using? What is output of `arch` command in your terminal? If it's `i386` Please make sure that `Open using Rosetta` is unchecked in the terminal app info
> 
> [image]

it output arm64,and I try to using a fish shell for build.And It success...It may be something wrong with my zsh

kumekay, 2022-08-11:
@Phantom2000X Thank you for checking, and it's nice that you found some clues. Did you try to reinstall `zsh` or run it forcefully on arm64?

I made a few more attempts to reproduce it locally but didn't have success. If you can provide any hints on how we can replicate it, we can add some notes to the installation docs. 

Please let us know if you still think it's something we can handle on the ESP-IDF side. Otherwise, consider closing this issue.

dobairoland, 2022-08-15:
I'm closing this but can be reopened if needed.

closed 2022-08-15 by dobairoland